armbian/build #5293: Do we want to cache files on the root partition?

Ekmansoft, 2023-06-04:
### What happened?

When making a few different Armbian builds for my boards last week, I noticed a failure message in the console due to it ran out of disk space. 

I thought this seemed strange since there is no problem on my user partition, but did not have time to look into it further. 

However, a couple of days later when trying to reboot Ubuntu, Ubuntu was no longer booting....

After some investigation a couple of days ago I realized that the root partition (maybe a bit smallish at 19 GB) was full due to a few GB found under a .cache/ccache directory. After booting into recovery mode and deleting these cached files, Ubuntu nicely booted again...

My understanding / suspicion is that the Armbian build system added these cached files, but I am not certain of this...

Is this caused by Armbian build? 
If so is the behavior known / expected / wanted?

Regards, Kenneth

### How to reproduce?

Run the armbian build on Ubuntu 22.xx.
Build for a few different boards / a few different kernel versions.
Your root partition is not huge, maybe less than 20 GB.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-06-04:
Related to this? https://github.com/armbian/build/pull/5287

Ekmansoft, 2023-06-04:
Yes definitely seems so! Thanks!

EvilOlaf, 2023-06-04:
Awesome. If possible let us know if this truly fixes your issue :)

Ekmansoft, 2023-06-04:
I increased the size of my root partition yesterday, but after building a few images, I do not see the bit increase in the .cache directory that I found on Friday!

closed 2023-06-04 by Ekmansoft