armbian/build #5330: Ubuntu Jammy 22.04.2 LTS aarch64 build failed with "chroot: failed to run command ‘/bin/bash’: Exec format error"

GrayLand119, 2023-06-14:
### What happened?

Machine:
Mac Book Pro M2 Max
Parallels Desktop

Build failed on aarch64 Ubuntu 22.04.2 LTS

Here are the ENV and error info:
```shell
lgl@server2204:~$ uname -a;lsb_release -a
Linux server2204 5.15.0-72-generic #79-Ubuntu SMP Tue Apr 18 16:53:43 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy

lgl@server2204:~/armbianbuild/build$ ./compile.sh \
BOARD=orangepizero \
BRANCH=current \
RELEASE=focal \
BUILD_MINIMAL=yes \
BUILD_DESKTOP=no \
KERNEL_ONLY=no \
KERNEL_CONFIGURE=no

[🔨]   '/usr/bin/qemu-arm-static' -> '/home/lgl/os/armbianbuild/.tmp/rootfs-1a9a371b-4727-4b8e-9014-0703574c8a35/usr/bin/qemu-arm-static'
[🌱] Installing base system [ Stage 2/2 ]
[🔨]   chroot: failed to run command ‘/bin/bash’: Exec format error
[💥] Error 126 occurred in main shell [ at /home/lgl/os/armbianbuild/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                  chroot_sdcard() --> lib/functions/logging/runners.sh:122
   create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:129
        create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
   artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:82
    artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
       obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:254
       build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:311
   get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:189
                           main() --> compile.sh:50
 ]
[💥] Error context msg [ Debootstrap second stage failed focal   yes ]

```

log file:
```
(125) COMMAND: cp -pv /usr/bin/qemu-arm-static /home/lgl/os/armbianbuild/.tmp/rootfs-73e9b59c-1f34-482b-a7a8-cab797dd5a18/>
   '/usr/bin/qemu-arm-static' -> '/home/lgl/os/armbianbuild/.tmp/rootfs-73e9b59c-1f34-482b-a7a8-cab797dd5a18/usr/bin/qemu-arm->
--> (125) INFO: Installing base system [ Stage 2/2 ]
--> (125) COMMAND: LC_ALL=C LANG=C /debootstrap/debootstrap --second-stage
   chroot: failed to run command ‘/bin/bash’: Exec format error
--> (125) ERROR: Error 126 occurred in main shell [ at /home/lgl/os/armbianbuild/lib/functions/logging/runners.sh:211
       run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                     chroot_sdcard() --> lib/functions/logging/runners.sh:122
      create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:110
           create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:69
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
      artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:82
       artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
          obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:254
          build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:311
      get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:83
            build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50
    ]
   -->--> command failed with error code 126 after 0 seconds
```

and other info:
```
lgl@lglserver:~/os/armbianbuild$ file /usr/bin/qemu-arm-static
/usr/bin/qemu-arm-static: ELF 64-bit LSB executable, ARM aarch64, version 1 (GNU/Linux), statically linked, BuildID[sha1]=5c2a87cff764802fe47f3414e294e0dd42bb96fa, for GNU/Linux 3.7.0, stripped
```

### How to reproduce?

```
./compile.sh build BOARD=orangepizero BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=yes CLEAN_LEVEL=images KERNEL_CONFIGURE=no RELEASE=focal SHARE_LOG=yes
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/owahokodat

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-06-14:
Hey @GrayLand119 a few things so we can try to address this

1) Can you build with `DEBUG=yes SHARE_LOG=yes` and share the debug log?
2) Can you try with `RELEASE=jammy` (eg, not focal -- that is a leftover and not really tested for a long time)
3) Can you do build with clean sources? You've modifications there that I can't tell what are doing.

That said, this looks like a problem with `binfmt`, but again, that shouldn't be necessary. If you're on Apple M1, using an aarch64 VM, armhf should run natively and no emulation should be necessary. Let's debug it further...

GrayLand119, 2023-06-15:
> Hey @GrayLand119 a few things so we can try to address this
>
> 1. Can you build with `DEBUG=yes SHARE_LOG=yes` and share the debug log?
> 2. Can you try with `RELEASE=jammy` (eg, not focal -- that is a [...]

Thanks for your reply. In response to your questions, I would like to add:
1. I will build later using the DEBUG=yes SHARE_LOG=yes RELEASE=jammy parameter
2. Although git shows changes, I used a pure project to build, and the modified part only added log code after encountering problems to view the information of the executable file
3. Apple Silicon CPU + aarch64 VM(Parallels Desktop)

GrayLand119, 2023-06-15:
> Hey @GrayLand119 a few things so we can try to address this
>
> 1. Can you build with `DEBUG=yes SHARE_LOG=yes` and share the debug log?
> 2. Can you try with `RELEASE=jammy` (eg, not focal -- that is a [...]

I had build with:
```
./compile.sh build BOARD=orangepizero BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=yes CLEAN_LEVEL=images DEBUG=yes KERNEL_CONFIGURE=no RELEASE=jammy SHARE_LOG=yes
```

Here is the log:
[https://paste.next.armbian.com/ojuvewuzag](https://paste.next.armbian.com/ojuvewuzag)

Here is the changed in git status, It's the code all commented,
git diff `lib/functions/cli/entrypoint.sh`:
```
lgl@lglserver:~/os/armbianbuild$ git diff lib/functions/cli/entrypoint.sh
diff --git a/lib/functions/cli/entrypoint.sh b/lib/functions/cli/entrypoint.sh
index efb48c6..d7a4e57 100644
--- a/lib/functions/cli/entrypoint.sh
+++ b/lib/functions/cli/entrypoint.sh
@@ -117,6 +117,19 @@ function cli_entrypoint() {
        declare -g -r MOUNT="${WORKDIR_BASE_TMP}/mount-${ARMBIAN_BUILD_UUID}"   # MOUNT ("mounted on the loop") is the mounted root on final image (via loop). "image" stage
        declare -g -r DESTIMG="${WORKDIR_BASE_TMP}/image-${ARMBIAN_BUILD_UUID}" # DESTIMG is where the backing image (raw, huge, sparse file) is kept (not the final destination)
 
+       # mkdir -p $SDCARD/bin
+       # mkdir -p $SDCARD/usr/bin
+       # mkdir -p $SDCARD/lib/aarch64-linux-gnu
+
+       # cp /bin/bash $SDCARD/bin/
+       # cp /lib/aarch64-linux-gnu/libtinfo.so.6 $SDCARD/lib/aarch64-linux-gnu/libtinfo.so.6
+       # cp /lib/aarch64-linux-gnu/libc.so.6 $SDCARD/lib/aarch64-linux-gnu/libc.so.6
+       # cp /lib/ld-linux-aarch64.so.1 $SDCARD/lib/ld-linux-aarch64.so.1
+
+       # echo "Copy /bin/bash to $SDCARD..."
+
+       # cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+
        # Make sure ARMBIAN_LOG_CLI_ID is set, and unique, and readonly.
        # Pre-runs might change it before this, but if not set, default to ARMBIAN_COMMAND.
        declare -r -g ARMBIAN_LOG_CLI_ID="${ARMBIAN_LOG_CLI_ID:-${ARMBIAN_COMMAND}}"
```

diff `lib/functions/rootfs/rootfs-create.sh`:
```
lgl@lglserver:~/os/armbianbuild$ git diff lib/functions/rootfs/rootfs-create.sh
diff --git a/lib/functions/rootfs/rootfs-create.sh b/lib/functions/rootfs/rootfs-create.sh
index 167d89b..051c513 100644
--- a/lib/functions/rootfs/rootfs-create.sh
+++ b/lib/functions/rootfs/rootfs-create.sh
@@ -106,6 +106,25 @@ function create_new_rootfs_cache_via_debootstrap() {
        deploy_qemu_binary_to_chroot "${SDCARD}" # this is cleaned-up later by post_debootstrap_tweaks() @TODO: which is too late for a cache
 
        display_alert "Installing base system" "Stage 2/2" "info"
+       # cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+       # mkdir -p $SDCARD/bin
+       # mkdir -p $SDCARD/usr/bin
+       # mkdir -p $SDCARD/lib/aarch64-linux-gnu
+
+       # cp /bin/bash $SDCARD/bin/
+       # cp /lib/aarch64-linux-gnu/libtinfo.so.6 $SDCARD/lib/aarch64-linux-gnu/libtinfo.so.6
+       # cp /lib/aarch64-linux-gnu/libc.so.6 $SDCARD/lib/aarch64-linux-gnu/libc.so.6
+       # cp /lib/ld-linux-aarch64.so.1 $SDCARD/lib/ld-linux-aarch64.so.1
+       # display_alert "Copy /bin/bash to $SDCARD..." "" "info"
+
+       # cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+       # display_alert "Copy qemu-arm-static..." "" "info"
+
+       # echo `ls $SDCARD`
+       # echo `file ${SDCARD}/bin/bash`
+       # echo `file ${SDCARD}/usr/bin/qemu-arm-static`
+
+
        declare -g if_error_detail_message="Debootstrap second stage failed ${RELEASE} ${DESKTOP_APPGROUPS_SELECTED} ${DESKTOP_ENVIRONMENT} ${BUILD_MINIMAL}"
        chroot_sdcard LC_ALL=C LANG=C /debootstrap/debootstrap --second-stage
        [[ ! -f "${SDCARD}/bin/bash" ]] && exit_with_error "Debootstrap first stage did not produce /bin/bash"
```

rpardini, 2023-06-15:
@GrayLand119 

1) You've no trouble with binfmt. This is looking like some problem in Parallels, Parallels VM configuration, or such.
1) You've a different error now, `Couldn't download packages: libgssapi-krb5-2 adduser` which is a network issue.
2) Please run with a clean git clone, without any modifications, from branch `main`

GrayLand119, 2023-06-15:
> @GrayLand119
> 
> 1. You've no trouble with binfmt. This is looking like some problem in Parallels, Parallels VM configuration, or such.
> 2. You've a different error now, `Couldn't download packages: libgssapi-krb5-2 adduser` which is a network issue.
> 3. Please run with a clean git clone, without any modifications, from branch `main`

I rebuild with a clean git clone, and here is my log uploaded manually:
[https://paste.next.armbian.com/uluridogoh](https://paste.next.armbian.com/uluridogoh)

Error with "chroot: failed to run command ‘/bin/bash’: Exec format error":
```
[🌱] Adding Armbian repository and authentication key [ image-early :: /etc/apt/sources.list.d/armbian.list ]
chroot: failed to run command ‘/bin/bash’: Exec format error
[💥] Error  126 occurred in SUBSHELL [ SUBSHELL at /home/lgl/os/armbianbuild/lib/functions/rootfs/distro-specific.sh:185 ]
[💥] Error 126 occurred in main shell [ at /home/lgl/os/armbianbuild/lib/functions/rootfs/distro-specific.sh:185
   create_sources_list_and_deploy_repo_key() --> lib/functions/rootfs/distro-specific.sh:185
        extract_rootfs_artifact() --> lib/functions/rootfs/create-cache.sh:129
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
   get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:89
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
```

rpardini, 2023-06-15:
Ok. Let's debug this outside Armbian. I've run your exact same build without problems under native arm64 iron without any problems; same under a qemu/kvm aarch64 VM on Linux. 

 Run those and include their output please.

```bash
apt -y update
apt -y install arch-test
arch-test
```

Check the output of `arch-test`, does it include `armhf`? 

 Run those and include their output please.

```bash
ls -la /proc/sys/fs/binfmt_misc/
update-binfmts --enable qemu-arm
```

GrayLand119, 2023-06-16:
@rpardini 
It seems to be caused by Apple Silicon not supporting 32-bit programs.

```
sudo apt-get remove qemu-user-static
sudo apt-get install qemu-user-static
sudo reboot
...

lgl@lglserver:~/os/env$ qemu-arm-static --version
qemu-arm version 6.2.0 (Debian 1:6.2+dfsg-2ubuntu6.10)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers

lgl@lglserver:~/os/env$ arch-test
alpha
amd64
arm64
hppa
i386
m68k
mips
mips64
mips64el
mipsel
mipsn32
mipsn32el
powerpc
ppc64
ppc64el
riscv64
s390x
sh4
sparc
sparc64
```

```
lgl@lglserver:~/os/env$ ls -la /proc/sys/fs/binfmt_misc/
total 0
drwxr-xr-x 2 root root 0 Jun 16 05:19 .
dr-xr-xr-x 1 root root 0 Jun 16 05:10 ..
-rw-r--r-- 1 root root 0 Jun 16 05:19 python2.7
-rw-r--r-- 1 root root 0 Jun 16 05:19 python3.10
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-alpha
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-armeb
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-cris
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-hexagon
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-hppa
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-i386
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-m68k
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-microblaze
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mips
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mips64
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mips64el
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mipsel
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mipsn32
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-mipsn32el
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-ppc
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-ppc64
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-ppc64le
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-riscv32
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-riscv64
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-s390x
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-sh4
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-sh4eb
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-sparc
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-sparc32plus
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-sparc64
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-x86_64
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-xtensa
-rw-r--r-- 1 root root 0 Jun 16 05:19 qemu-xtensaeb
--w------- 1 root root 0 Jun 16 05:19 register
-rw-r--r-- 1 root root 0 Jun 16 06:50 status
```

```
lgl@lglserver:~/os/env$ sudo update-binfmts --enable qemu-arm
update-binfmts: warning: qemu-arm not in database of installed binary formats.
update-binfmts: exiting due to previous errors
```

GrayLand119, 2023-06-19:
It doesn't seem to be armbian build causing this problem.

closed 2023-06-19 by GrayLand119

rpardini, 2023-06-23:
Hey @GrayLand119 -- although it is not Armbian's "fault", we could maybe work-around it. Did you find any solution?

GrayLand119, 2023-06-26:
> Hey @GrayLand119 -- although it is not Armbian's "fault", we could maybe work-around it. Did you find any solution?

It has not been solved yet, I tried to compile qemu from the source code, and failed to compile with `--static` and `--disable-system` parameters.

Maybe try again later, currently I have successfully compiled it using another Linux host (amd64 base).

jclink, 2023-10-24:
I encountered the runners.sh build error as well.  For me, it was a mismatch in versioning for the chromium-browser package.  I was able to build by editing:
armbian/build/config/desktop/jammy/appgroups/browsers/packages

Commenting out chromium-browser allowed for a build, and changing it to firefox did also.  For the firefox change, I also edited out the chromium entries in file 'create_desktop_package.sh'.
I think I may have been able to accomplish the same thing during the build by simply leaving browsers deselected in the 'config' menu, but I have not tried it.
After searching online for my chroot issue, I noticed a comment by Nico at Armbian suggesting a chromium downgrade.  I think that has a good chance of working, but did not need it so I can't say for sure.  

If the issue for others is the same, package versioning, then a possible workaround could be found in the build scripts.  Maybe not fail on error, or allow user to decide during build.   Another possible solution is to look up the distro packages and install the debs.  I pull a couple things from 'Mint packages' as it is.  Here's a link to the Ubuntu packages if anyone needs it:
[Ubuntu Package Search](https://packages.ubuntu.com/)

lgrunenberg, 2024-07-05:
Just leaving this here in case it is useful for anyone, I managed to finally successfully build Armbian on Ubuntu 22.04 on my MacBook M1. The steps involved were:

1. Enable binfmt emulation for arm binaries
1.1 Create symlink to interpreter:
```
cd /usr/libexec/qemu-binfmt/ && sudo ln -s ../../bin/qemu-arm-static arm-binfmt-P
```
1.2 Add binfmt support:
```
sudo sh -c "cat<<EOF >> /usr/share/binfmts/qemu-sdfgsdf
package qemu-user-static
interpreter /usr/libexec/qemu-binfmt/arm-binfmt-P
magic \x7f\x45\x4c\x46\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00
offset 0
mask \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
credentials yes
fix_binary no
preserve yes
EOF"
```
1.3 Enable binfmt:
```
sudo update-binfmts --enable qemu-arm
```
1.4 Verify that arm emulation is working:
Output of arch-test should look like this:

> alpha
amd64
arm
arm64
armel
armhf
hppa
i386
m68k
mips
mips64
mips64el
mipsel
mipsn32
mipsn32el
powerpc
ppc64
ppc64el
riscv64
s390x
sh4
sparc
sparc64

2. Apply following changes to Armbian build:
```
`diff` --git a/lib/functions/cli/entrypoint.sh b/lib/functions/cli/entrypoint.sh
index a1ca193..a4e7f6e 100644
--- a/lib/functions/cli/entrypoint.sh
+++ b/lib/functions/cli/entrypoint.sh
@@ -117,6 +117,12 @@ function cli_entrypoint() {
        declare -g -r MOUNT="${WORKDIR_BASE_TMP}/mount-${ARMBIAN_BUILD_UUID}"   # MOUNT ("mounted on the loop") is the mounted root on final image (via loop). "image" stage
        declare -g -r DESTIMG="${WORKDIR_BASE_TMP}/image-${ARMBIAN_BUILD_UUID}" # DESTIMG is where the backing image (raw, huge, sparse file) is kept (not the final destination)
 
+        mkdir -p $SDCARD/bin
+        mkdir -p $SDCARD/usr/bin
+
+        cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+        cp -r /usr/libexec/qemu-binfmt $SDCARD/usr/libexec
+
        # Make sure ARMBIAN_LOG_CLI_ID is set, and unique, and readonly.
        # Pre-runs might change it before this, but if not set, default to ARMBIAN_COMMAND.
        declare -r -g ARMBIAN_LOG_CLI_ID="${ARMBIAN_LOG_CLI_ID:-${ARMBIAN_COMMAND}}"
diff --git a/lib/functions/rootfs/rootfs-create.sh b/lib/functions/rootfs/rootfs-create.sh
index efc8550..4e0cfb2 100644
--- a/lib/functions/rootfs/rootfs-create.sh
+++ b/lib/functions/rootfs/rootfs-create.sh
@@ -124,6 +124,13 @@ function create_new_rootfs_cache_via_debootstrap() {
        deploy_qemu_binary_to_chroot "${SDCARD}" # this is cleaned-up later by post_debootstrap_tweaks() @TODO: which is too late for a cache
 
        display_alert "Installing base system" "Stage 2/2" "info"
+        cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+        mkdir -p $SDCARD/bin
+        mkdir -p $SDCARD/usr/bin
+
+        cp /usr/bin/qemu-arm-static $SDCARD/usr/bin/
+        cp -r /usr/libexec/qemu-binfmt $SDCARD/usr/libexec
+
        declare -g -a if_error_find_files_sdcard=("debootstrap.log") # if command fails, go look for this file and show it's contents during error processing
        declare -g if_error_detail_message="Debootstrap second stage failed ${RELEASE} ${DESKTOP_APPGROUPS_SELECTED} ${DESKTOP_ENVIRONMENT} ${BUILD_MINIMAL}"
        chroot_sdcard LC_ALL=C LANG=C /debootstrap/debootstrap --second-stage
```

After that you should be able to build Armbian for arm targets, arm64 and others not checked.

rpardini, 2024-12-28:
Thanks @lgrunenberg - I've finally some Apple Silicon to test with, and I adapted incorporated the general idea into https://github.com/armbian/build/pull/7621

igorpecovnik, 2025-01-01:
Probably related:

https://forum.armbian.com/topic/49062-problem-building-from-source-code

muratdemirtas, 2025-02-21:
i have same problem too.. ubuntu 24.04 amd64 ubuntu native pc.  i'll downgrade ubuntu version to 22.04 jammy. only solution....
 
[🔨]   I: Extracting mawk...
[🔨]   I: Extracting mount...
[🔨]   I: Extracting ncurses-base...
[🔨]   I: Extracting ncurses-bin...
[🔨]   I: Extracting passwd...
[🔨]   I: Extracting perl-base...
[🔨]   I: Extracting sed...
[🔨]   I: Extracting sysvinit-utils...
[🔨]   I: Extracting tar...
[🔨]   I: Extracting tzdata...
[🔨]   I: Extracting usr-is-merged...
[🔨]   I: Extracting util-linux...
[🔨]   I: Extracting util-linux-extra...
[🔨]   I: Extracting zlib1g...
[🔨]   '/usr/bin/qemu-aarch64-static' -> '/home/oem/Desktop/armbian/build/.tmp/rootfs-50db3f76-6005-478d-ace2-c960e78db81a/usr/bin/qemu-aarch64-static'
[🌱] Installing base system [ Stage 2/2 ]
[🔨]   chroot: failed to run command ‘/usr/bin/env’: Exec format error
[💥] Error 126 occurred in main shell [ at /home/oem/Desktop/armbian/build/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                  chroot_sdcard() --> lib/functions/logging/runners.sh:122
   create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:129
        create_new_rootfs_cache() --> lib/functions`.   

`58  git clone --depth=1 --branch=v24.11 https://github.com/armbian/build
   59  cd build/
   60  ls
   61  sudo ./compile.sh 
   62  ./compile.sh 
   63  sudo apt -y update
   64  sudo apt -y install arch-test
   65  sudo arch-test
   66  ./compile.sh 
   67  history
`

``oem@oem-BOHK-WAX9X:~/Desktop/armbian/build$ uname -a
Linux oem-BOHK-WAX9X 6.8.0-41-generic #41-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug  2 20:41:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
oem@oem-BOHK-WAX9X:~/Desktop/armbian/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
oem@oem-BOHK-WAX9X:~/Desktop/armbian/build$