armbian/build #5332: `Exit Code 100` and `Error 1 in main shell` when building Armbian Bullseye

lasermind, 2023-06-14:
### What happened?

After a whole day of red errors, I need to file an issue here. I have no idea what's wrong. As follows:

I'm trying to build an Armbian from source. I set up everything the guide said: https://docs.armbian.com/Developer-Guide_Build-Preparation/

1. Created a VM with the recommended Ubuntu 22.04.2 Jammy. I'm on Windows 10 and I have tried this with VirtualBox and meanwhile with VMware Player.
2. Logged in and escalated to root with `sudo su -`
3. I ran these commands, as described:
```
apt-get -y -qq install git  
git clone --depth=1 --branch=main https://github.com/armbian/build  
cd build  
./compile.sh
```
4. I selected: `Do not change the kernel configuration` → `rockpi4b` but also tried `rockpro64` and others → `current` but also tried `legacy` now → `bullseye` → `server` → `minimal image`.

Then it starts compiling. It's looking good at first. After some time, there is a first suspicious message:

[image]

I have no idea whether this causes the final error some minutes later:

[image]

As you might see, there are some red errors. Actually two of them:

```
(154) COMMAND: DEBIAN_FRONTEND=noninteractive ARMBIAN_IMAGE_BUILD_BOOTFS_TYPE=unset apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/armbian-bsp-cli-rockpi-4b-legacy_23.08.0-trunk--1-PC5d49-Vee82-Hcb18-B9c45_arm64.deb
   E: Unable to correct problems, you have held broken packages.
   -->--> command failed with error code 100 after 0 seconds
--> (154) WARNING: stacktrace for failed command [ exit code 100:chroot /root/build/.tmp/rootfs-bf6cf66b-b1a2-4e91-9afb-5541511e7b2e /bin/bash -e -o pipefail -c DEBIAN_FRONTEND=noninteractive ARMBIAN_IMAGE_BUILD_BOOTFS_TYPE=unset apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/armbian-bsp-cli-rockpi-4b-legacy_23.08.0-trunk--1-PC5d49-Vee82-Hcb18-B9c45_arm64.deb
                     chroot_sdcard() --> lib/functions/logging/runners.sh:122
             chroot_sdcard_apt_get() --> lib/functions/logging/runners.sh:107
                install_deb_chroot() --> lib/functions/rootfs/apt-install.sh:106
      install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:366
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
            build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> ./compile.sh:50 ]

--> (154) ERROR: Error context msg [ Installation of /root/armbian-bsp-cli-rockpi-4b-legacy_23.08.0-trunk--1-PC5d49-Vee82-Hcb18-B9c45_arm64.deb failed rockpi-4b bullseye no rockchip64 ]
--> (154) COMMAND: umount /root/build/.tmp/rootfs-bf6cf66b-b1a2-4e91-9afb-5541511e7b2e/var/cache/apt
--> (154) COMMAND: umount /root/build/.tmp/rootfs-bf6cf66b-b1a2-4e91-9afb-5541511e7b2e/var/lib/apt/lists
--> (154) ERROR: Error 1 occurred in main shell [ at /root/build/lib/functions/rootfs/apt-install.sh:106
                install_deb_chroot() --> lib/functions/rootfs/apt-install.sh:106
      install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:366
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
            build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> ./compile.sh:50
    ]
--> (154) ERROR: Cleaning up [ please wait for cleanups to finish ]
```

I provided the whole log file as a Gist, please see link below.

Do you have any clue of what's causing the script to not build and produce these errors? Did I make any mistakes?  I'm pulling my hair out over this …

### How to reproduce?

See above steps ☝, they should be sufficient to reproduce this behaviour.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://gist.github.com/lasermind/4a8050898bda6172b63e792418844dc4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-06-15:
Please check your network. This is the DNS issue.

rpardini, 2023-06-15:
Confirmed. Closing

closed 2023-06-15 by rpardini