armbian/build #5348: FriendlyElec vendor "driver" patch in mainline

Tonymac32, 2023-06-18:
https://github.com/armbian/build/blob/efa936525c04f4a087489e6e0a2a6e6172b5a20f/patch/kernel/archive/rockchip64-6.3/add-maker-friendlyarm.patch

Why does this exist?  Can this functionality not be done in user space?

rpardini, 2023-06-23:
I've stumbled into that before and wondered exactly the same. What purpose does that serve....?

Tonymac32, 2023-06-23:
I *think* this is how they sort out the CF that is their "nanopin-rev-xx" situation.  We don't need it that I can see.

Tonymac32, 2023-10-03:
For any @armbian/boards-rockchip maintainers I plan on deleting this patch before next release, it serves no purpose.  If you need it, some with a *very* convincing argument

igorpecovnik, 2023-10-03:
the only relation i can come up with is their patched wiringpi scripting compatibility. i support removing.

Tonymac32, 2023-10-06:
https://github.com/armbian/build/pull/5792

closed 2023-10-06 by igorpecovnik