armbian/build #5370: Rock-5B images with edge kernel are not bootable

omikron88, 2023-06-22:
### What happened?

When I build Rock-5B image with edge kernel, it doesnt boot on board. Both green and blue LEDs are on permanently.
Images with other kernel options works as expected.

### How to reproduce?

./compile.sh
and choose Rock-5B board and edge kernel.
other options doesnt matter

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-06-22:
> Images with other kernel options works as expected.

Which proves this is not a build framework bug. Try forums https://forum.armbian.com/ Closing.

closed 2023-06-22 by igorpecovnik

Tonymac32, 2023-06-22:
Some background https://github.com/armbian/build/pull/4606

Tonymac32, 2023-06-23:
@amazingfate in case you missed this before it got auto-closed by Igor.  I can't get one of these edge images to boot either (admittedly my local copy is a little different due to adding prelim T6 stuff, but it shouldn't cause u-boot to fail to load)

amazingfate, 2023-06-23:
I haven't tried mainline kernel on rk3588 yet. I will take a look into it when I have time.

Tonymac32, 2023-06-28:
I will take a closer look at this tonight, as you probably noticed there are no edge patches at the moment, I need to pull in some patches and set up the new device tree/overlay structure.