armbian/build #5383: Problem found when building with rk3399-legacy kernel

Damima3369, 2023-06-25:
### What happened?

🐳|🌱] User patches directory for kernel [ /armbian/userpatches/kernel/rk3399-legacy ]
[🐳|🌱] Using kernel config file [ config/kernel/linux-rk3399-legacy.config ]
sha256sum: config/kernel/linux-rk3399-legacy.config: No such file or directory
[🐳|💥] Error  1 occurred in SUBSHELL [ SUBSHELL at /armbian/lib/functions/general/hash-files.sh:74 ]

### How to reproduce?

Ask script make system with editing legacy kernel parameters for orangepi4 or another one who has rk3399

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/ihixajiluw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-06-25:
https://github.com/armbian/linux-rockchip/pull/57
support for legacy kernel 4.4 was dropped as its not maintained anymore. switching rockchip RK3399 legacy to 5.10.y is just a wish ... no plans.

Tonymac32, 2023-10-03:
I think we log this as a "will not fix" as we've cleaned up/removed almost all the legacy stuff for RK3399

closed 2023-10-03 by igorpecovnik

rpardini, 2023-10-05:
Hmm, it seems we still have a few boards with rockchip64-legacy (ayufan 4.4.y) enabled:

- `rk3318-box`		Rockchip RK3318 quad core 2/16Gb 1xETH USB3 WiFi
- `rock64`			Rockchip RK3328 quad core 1GB-4GB GBE eMMC USB3 SPI
- `rockpi-e`		Rockchip RK3328 quad core 256MB-2GB 1xGBE 1xETH USB3 [WiFi]
- `renegade`		Rockchip RK3328 quad core 2GB/4GB GBE eMMC USB3
- `rockpi-4a`		Rockchip RK3399 hexa core 1-4GB SoC GBe eMMC USB3
- `rockpi-4b`		Rockchip RK3399 hexa core 1-4GB SoC GBe eMMC USB3 WiFi/BT PoE
- `rockpi-4c`		Rockchip RK3399 hexa core 1-4GB SoC GBe eMMC USB3 WiFi/BT PoE miniDP USB host
- `rockpro64`		Rockchip RK3399 hexa core 2G/4GB SoC GBe eMMC USB3 WiFi
- `pinebook-pro`	Rockchip RK3399 hexa core 2G/4GB SoC Laptop eMMC USB3 WiFi
- `clockworkpi-a06`	Rockchip RK3399 hexa core 4GB RAM SoC WiFi/BT

igorpecovnik, 2023-10-05:
Made a ticket https://armbian.atlassian.net/browse/AR-1882