armbian/build #5384: Broken Chromium

cantalupo555, 2023-06-25:
### What happened?

After installing the drivers the GPU does not recognize Chromium.

```
sudo add-apt-repository ppa:liujianfeng1994/panfork-mesa
sudo add-apt-repository ppa:liujianfeng1994/rockchip-multimedia
sudo apt update
sudo apt dist-upgrade
sudo apt install mali-g610-firmware rockchip-multimedia-config
```

### How to reproduce?

Orange Pi 5

```
o@orangepi5:~$ chromium-browser 
[3400:3400:0625/161224.783903:ERROR:gpu_init.cc(523)] Passthrough is not supported, GL is egl, ANGLE is 
[3400:3400:0625/161224.788330:ERROR:gl_display.cc(504)] EGL Driver message (Error) eglQueryDeviceAttribEXT: eglQueryDeviceAttribEXT
[3400:3400:0625/161224.788524:ERROR:gl_angle_util_vulkan.cc(189)] Failed to retrieve vkGetInstanceProcAddr
[3400:3400:0625/161224.788670:ERROR:vulkan_instance.cco@orangepi5:~$ chromium-browser 
[3400:3400:0625/161224.783903:ERROR:gpu_init.cc(523)] Passthrough is not supported, GL is egl, ANGLE is 
[3400:3400:0625/161224.788330:ERROR:gl_display.cc(504)] EGL Driver message (Error) eglQueryDeviceAttribEXT: eglQueryDeviceAttribEXT
[3400:3400:0625/161224.788524:ERROR:gl_angle_util_vulkan.cc(189)] Failed to retrieve vkGetInstanceProcAddr
[3400:3400:0625/161224.788670:ERROR:vulkan_instance.cc(91)] Failed to get vkGetInstanceProcAddr pointer from ANGLE.
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device(91)] Failed to get vkGetInstanceProcAddr pointer from ANGLE.
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2023-06-26:
What's the issue? What do you want?

cantalupo555, 2023-06-26:
> What's the issue? What do you want?

I don't know, I just installed the recommended repository on the official download page and the bug happens.

I'm no longer using Armbian, but I'm here to report what happened.

cantalupo555, 2023-06-26:
Problem encountered when switching to X11, chromium has constant crashes, very slow.
Any open video becomes stop motion.

amazingfate, 2023-06-26:
Known issue under x11, Wayland is recommended. Or you can try some other chromium args.

igorpecovnik, 2023-06-26:
> I'm no longer using Armbian

The same problem will almost certainly manifest elsewhere too.

> but I'm here to report what happened.

On the download pages, where you found this information, there is a tab "Report bugs". It leads you to this page:
https://www.armbian.com/bugs/

This issue tracker belongs to build system and end user issue with Chromium has nothing to do with this. Closing.

closed 2023-06-26 by igorpecovnik