armbian/build #5439: `armbian-desktop` package is severely inconsistent across arch/appgroups/config_base

rpardini, 2023-07-05:
### What happened?

Ok so here's an old impasse. Hold on to your hat.
`config/optional/architectures/amd64/_config/desktop/_all_distributions/appgroups/browsers/packages`
This adds `google-chrome-stable` (with an extra apt-source) to `AGGREGATED_PACKAGES_DESKTOP_COMMA`.
But only if:
- ARCH=amd64 *and*
- APPGROUPS includes "browsers"
So it is _not_ included for any arm64, ever, nor if APPGROUPS does not include "browsers".
All good.
_Unfortunately_, the `armbian-desktop` deb is then inconsistent:
- Resulting package name would be `armbian-jammy-desktop-gnome`
- package is `Architecture: all`
- Recommends: `AGGREGATED_PACKAGES_DESKTOP_COMMA` (only for the amd64+browsers case, that includes `google-chrome-stable`)

So it's actually impossible to conciliate desktop aggregation (release+arch+appgroups+config_base+...) with a "single package", arch-independent, "armbian-desktop" package that is only per-`RELEASE` + `DESKTOP_ENVIROMENT`.
It gets worse if we consider `AGGREGATED_DESKTOP_POSTINST` and `AGGREGATED_DESKTOP_CREATE_DESKTOP_PACKAGE` which are _also_ included in that package.

Some possibilities:
- Make the package arch-specific (solves some cases, but not the config_base nor appgroups scenarios.)
- Twist aggregation so we produce some kind of appgroup/config_base-independent `AGGREGATED_PACKAGES_DESKTOP_SANS_APPGROUPS_SANS_CONFIG_BASE_COMMA` and use that in the package Recommends.

### How to reproduce?

"Simple" way to reproduce:

- `./compile.sh 'BETA=yes' 'BOARD=uefi-arm64' 'BRANCH=current' 'BUILD_DESKTOP=yes' 'BUILD_MINIMAL=no' 'DESKTOP_APPGROUPS_SELECTED=browsers' 'DESKTOP_ENVIRONMENT=gnome' 'DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base' 'RELEASE=jammy' ARTIFACT_IGNORE_CACHE=yes`

- `./compile.sh 'BETA=yes' 'BOARD=uefi-amd64' 'BRANCH=current' 'BUILD_DESKTOP=yes' 'BUILD_MINIMAL=no' 'DESKTOP_APPGROUPS_SELECTED=browsers' 'DESKTOP_ENVIRONMENT=gnome' 'DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base' 'RELEASE=jammy' ARTIFACT_IGNORE_CACHE=yes`

- `./compile.sh 'BETA=yes' 'BOARD=uefi-amd64' 'BRANCH=current' 'BUILD_DESKTOP=yes' 'BUILD_MINIMAL=no' 'DESKTOP_APPGROUPS_SELECTED=none' 'DESKTOP_ENVIRONMENT=gnome' 'DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base' 'RELEASE=jammy' ARTIFACT_IGNORE_CACHE=yes`

Each of these 3 builds produces a different `armbian-jammy-desktop-gnome`.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-07-05:
For completeness: `rootfs-amd64-jammy-gnome-desktop` (rootfs artifact) has exactly the same problem, since it does not include appgroups or config_base in the name (it does include ARCH, though).
But, since rootfs artifacts are **not published to an apt repo**, this is a non-problem; the rootfs versions will vary and will be found correctly in OCI, and life goes on.

`armbian-bsp-desktop` also has the same problem.

igorpecovnik, 2023-07-05:
> it's actually impossible to conciliate desktop aggregation

Yes. 

"armbian-desktop-$DE-$ARCH" should be either with all groups or with no groups, so something that is fixed. Idea is that this package, when installed from minimal image, provides desktop.

Also regarding configs, if we don't use them, perhaps removing them?

rpardini, 2023-07-05:
Yes, we could remove appgroups and configs, but still, ARCH differences would still occur... 

I'm testing making this package ARCH-specific, since nothing practical would change (package name is the same), and would allow us to "get by" as long as the targets.yaml used for repo-publishing does not have different appgroups/configs across all targets.

igorpecovnik, 2023-07-05:
AFAIK configs are always the same - at least for our internal needs. If someone else uses this, can't tell. Lets discuss this problem later on meeting.

rpardini, 2023-07-05:
Yes. Configs are always the same. (`config_base`).

Unfortunately, the `armbian-desktop` package is also responsible for handling:
- `debian/postinst` config files, in `AGGREGATED_DESKTOP_POSTINST` (goes into the package postinst)
- `armbian/create_desktop_package.sh` in `AGGREGATED_DESKTOP_CREATE_DESKTOP_PACKAGE` (is eval()'ed!) during package creation

So if we have a "fixed"/common armbian-desktop package (same for all appgroups) we'd be in trouble, we'd miss at least those:

- `config/desktop/focal/appgroups/editors/debian/postinst`
- `config/desktop/focal/appgroups/browsers/debian/postinst`
- `config/desktop/focal/appgroups/browsers/armbian/create_desktop_package.sh`
- `config/desktop/buster/appgroups/browsers/debian/postinst`
- `config/desktop/buster/appgroups/browsers/armbian/create_desktop_package.sh`
- `config/desktop/bullseye/appgroups/browsers/debian/postinst`
- `config/desktop/bullseye/appgroups/browsers/armbian/create_desktop_package.sh`
- `config/desktop/sid/appgroups/browsers/debian/postinst`
- `config/desktop/sid/appgroups/browsers/armbian/create_desktop_package.sh`

So we'd need to move those to image creation or something.

Similarly, the `desktop-bsp`  package also handles

- `debian/armbian-bsp-desktop/postinst` in `AGGREGATED_DESKTOP_BSP_POSTINST` (no occurences of this in appgroups, thankfully)
- `debian/armbian-bsp-desktop/prepare.sh` in `AGGREGATED_DESKTOP_BSP_PREPARE` (same, no occurences for appgroups)