armbian/build #5449: When build img with CRYPTROOT_ENABLE=yes，build broken！

sputnik2019, 2023-07-08:
### What happened?

I tried to compile the armbian system, and after adding CRYPTROOT_ENABLE=yes parameters, I compiled all board images with this error。I'm guessing that the device-mapper layer doesn't support docker containers。
when build log output
### install_distribution_agnostic.log 
--> (927) INFO: Installing rootfs encryption related packages [ cryptsetup ]
--> (927) COMMAND: mkdir -pv /home/rll/rpi4b/build/.tmp/rootfs-3c3e189f-a771-4c45-abbb-0dfc9b2fc48c/var/cache/apt /home/rll/rpi4b/build/.tmp/rootfs-3c3e189f-a771-4c45-abbb-0dfc9b2fc48c/var/lib/apt/lists
--> (927) COMMAND: mount --bind /home/rll/rpi4b/build/cache/aptcache/jammy-arm64 /home/rll/rpi4b/build/.tmp/rootfs-3c3e189f-a771-4c45-abbb-0dfc9b2fc48c/var/cache/apt
--> (927) COMMAND: mount --bind /home/rll/rpi4b/build/cache/aptcache/lists/jammy-arm64 /home/rll/rpi4b/build/.tmp/rootfs-3c3e189f-a771-4c45-abbb-0dfc9b2fc48c/var/lib/apt/lists
--> (927) COMMAND: DEBIAN_FRONTEND=noninteractive apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 --no-install-recommends install cryptsetup
   E: Package 'cryptsetup' has no installation candidate
   -->--> command failed with error code 100 after 0 seconds

### How to reproduce?

./compile.sh build BOARD=rpi4b BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=1234 KERNEL_CONFIGURE=no PREFER_DOCKER=no RELEASE=jammy

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-07-08:
You read this, right? https://docs.armbian.com/Developer-Guide_Build-Options/

sputnik2019, 2023-07-08:
I added DOCKER_FLAGS+=(--privileged) to the file userpatches/config-docker
But when compiling, it still prompts an error
cli：./compile.sh build BOARD=rpi4b BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=1234 KERNEL_CONFIGURE=no RELEASE=jammy

log output：
### install_distribution_agnostic.log 
--> (19) INFO: Installing rootfs encryption related packages [ cryptsetup ]
--> (19) COMMAND: mkdir -pv /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/cache/apt /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/lib/apt/lists
--> (19) COMMAND: mount --bind /home/rll/rpi4b/build/cache/aptcache/jammy-arm64 /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/cache/apt
--> (19) COMMAND: mount --bind /home/rll/rpi4b/build/cache/aptcache/lists/jammy-arm64 /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/lib/apt/lists
--> (19) COMMAND: DEBIAN_FRONTEND=noninteractive apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 --no-install-recommends install cryptsetup
   E: Package 'cryptsetup' has no installation candidate
   -->--> command failed with error code 100 after 0 seconds
--> (19) WARNING: stacktrace for failed command [ exit code 100:chroot /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07 /bin/bash -e -o pipefail -c DEBIAN_FRONTEND=noninteractive apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 --no-install-recommends install cryptsetup
                     chroot_sdcard() --> lib/functions/logging/runners.sh:122
             chroot_sdcard_apt_get() --> lib/functions/logging/runners.sh:107
      chroot_sdcard_apt_get_install() --> lib/functions/logging/runners.sh:15
      install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:17
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
            build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50 ]
--> (19) COMMAND: umount /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/cache/apt
--> (19) COMMAND: umount /home/rll/rpi4b/build/.tmp/rootfs-f91d49f7-a064-4ddd-b430-337731d3ed07/var/lib/apt/lists
--> (19) ERROR: Error 1 occurred in main shell [ at /home/rll/rpi4b/build/lib/functions/logging/runners.sh:15
      chroot_sdcard_apt_get_install() --> lib/functions/logging/runners.sh:15
      install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:17
                   do_with_logging() --> lib/functions/logging/section-logging.sh:81
            build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50
    ]

sputnik2019, 2023-07-08:
full log here
[log-build-f91d49f7-a064-4ddd-b430-337731d3ed07.log](https://github.com/armbian/build/files/11990497/log-build-f91d49f7-a064-4ddd-b430-337731d3ed07.log)

viraniac, 2023-07-08:
Try creating a build with `ARTIFACT_IGNORE_CACHE=yes`

viraniac, 2023-07-21:
The Encrypted rootfs functionality is now fixed in [PR#5496](https://github.com/armbian/build/pull/5496) and the fixes are merged, hence closing this issue.

closed 2023-07-21 by viraniac

maxwowpow, 2023-07-23:
the build works.

raspberry pi 3 starts until dropbear.
dropbear starts and drops right away into initramfs.
the root volume is mountable manually, after mounting and exec switch_root system hangs in `Started: File System Check Daemon to report state`
in the wait for root error there's a mention of "armbi_root".

bananapi2ultra initramfs cannot start the dropbear.
manual unlock with given passcode doesn't work.

https://paste.next.armbian.com/rinodofulu.shell
```
[🔨]   mode of '/etc/kernel/postinst.d/initramfs-tools' retained as 0644 (rw-r--r--)
[🔨]   mode of '/etc/initramfs/post-update.d/flash-kernel' changed from 0755 (rwxr-xr-x) to 0644 (rw-r--r--)
[🌿] Updating flash-kernel initramfs... [ update-initramfs -c -k all ]
[🔨]   update-initramfs: Generating /boot/initrd.img-6.1.39-bcm2711
**[🔨]   cryptsetup: ERROR: Couldn't find sysfs directory for 253:0
[🔨]   dropbear: WARNING: Invalid authorized_keys file, SSH login to initramfs won't work!**
[🌱] Configuring flash-kernel DB... [ bcm2711 ]
[🌱] flash-kernel [ Raspberry Pi 4 Model B ]
```

viraniac, 2023-07-24:
@maxwowpow 
Could you please give us some information about your build environment? From the above posted log, I can see you are not running the build within docker. What was your host OS and version?

maxwowpow, 2023-07-30:
hello, it was the gist linked on the official build webpage:
https://docs.armbian.com/Developer-Guide_Build-Preparation/
https://gist.github.com/atomic77/7633fcdbf99dca80f31fd6d64bfd0565

Removed the multipass and enabled wsl2.

- Wsl2 uses debian bookworm. It's what you get if you `wsl --install -d Debian`
- Built successfully w/o warnings bananapi2ultra and rpi4b.

- `bananapim2ultra` has no `authorized_keys`, but after manual unlock (which works) and config of dropbear - everything runs OK.

- `rpi4b` doesn't work. Starts but no dropbear and then drops into initramfs after some minutes giving up on waiting 'armbi_root' same as with multipass (I think it was jammy as described in a build doc that it only tested/recommended for jammy).