armbian/build #5502: During build process:   cp: error writing '/home/user/build/.tmp/work-XXXXXXXX: No space left on device

dtischler, 2023-07-23:
### What happened?

On a freshly formatted and installed Ubuntu Jammy x86_64 host, attempting to build `uefi_arm64` target results in a "No space left on device" failure.  This host has a 480gb SSD, Ubuntu was just installed (no other OS'es on the drive), so roughly 460gb free when starting the build process.  Default kernel configuration is used, no changes made.  Performed a `git clone`, kicked off a build with `./compile.sh`.

The build begins, but fails during the process here:

<snip>

```
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8150-sony-xperia-kumano-griffin.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8150-sony-xperia-kumano-bahamut.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-sony-xperia-edo-pdx206.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-sony-xperia-edo-pdx203.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-xiaomi-elish-csot.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8250-xiaomi-elish-boe.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8350-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8350-microsoft-surface-duo2.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8350-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8350-sony-xperia-sagami-pdx214.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8350-sony-xperia-sagami-pdx215.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8450-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8450-qrd.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8450-sony-xperia-nagara-pdx223.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8450-sony-xperia-nagara-pdx224.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8550-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/dtbs/qcom/sm8550-qrd.dtb
[🔨]   make: Nothing to be done for 'install'.
[🌱] Kernel built in [ 10155 seconds - 6.4.5-arm64 ]
[🌱] Packaging kernel [ arm64 linux-uefi-arm64-edge ]
[🌱] Kernel .deb package version [ 23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-C0836Hfe66-HK01ba-Vc222-B3b41 ]
[🌱] Packaging linux-image [ arm64 linux-uefi-arm64-edge ]
[🔨]   [ 32M]  /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/image/boot
[🔨]   ├── [291K]  config-6.4.5-arm64
[🔨]   ├── [4.6M]  System.map-6.4.5-arm64
[🔨]   └── [ 28M]  vmlinuz-6.4.5-arm64
[🔨]   
[🔨]     32M used in 0 directories, 3 files
[🌱] Kernel-built image filetype [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 23 09:31 /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/image/boot/vmlinuz-6.4.5-arm64
[🔨]   
[🔨]   /home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.K2mwXxAZMZ/image/boot:
[🔨]   total 33152
[🔨]   drwxrwxr-x 2 root root      100 Jul 23 09:31 .
[🔨]   drwxrwxr-x 3 root root       60 Jul 23 06:42 ..
[🔨]   -rw-rw-r-- 1 root root   297721 Jul 23 09:31 config-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root  4782384 Jul 23 09:31 System.map-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 23 09:31 vmlinuz-6.4.5-arm64
[🌱] Kernel image filetype after pre_package_kernel_image [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192ee/rtl8192ee.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_core.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822b.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822be.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822c.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8822ce.ko': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/kernel/drivers/net/wireless/realtek/rtw88/rtw88_8723d.ko': No space left on device
```

</snip>

...and continues on through all the modules, until exiting out:

<snip>

```
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/modules.dep.bin': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/modules.alias': No space left on device
[🔨]   cp: error writing '/home/david/build/.tmp/work-f5ac6642-9fa6-460c-abc1-e771b87c2145/tmp.LrzkWvOis2/lib/modules/6.4.5-arm64/modules.alias.bin': No space left on device
[... 38 lines cut ...]
### How to reproduce?

Freshly install Jammy x86_64 on a host, then:

```
david@david-HP:~/build$ history
    1  sudo apt update
    2  sudo apt install git htop
    3  sudo apt install build-essential
    4  git clone --depth=1 --branch=main https://github.com/armbian/build
    5  cd build
    6  ./compile.sh
```    

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-07-23:
> any obvious

How much memory your host have?

dtischler, 2023-07-23:
Aha - good thought.  Only 8gb RAM, although I don't recall it being an issue, as I had `htop` running as well to keep an eye on it.

rpardini, 2023-07-23:
This is definitely on tmpfs, so RAM size is important.
uefi-arm64 kernel is huge (full of modules) so might consume quite a bit of tmpfs (RAM) space during kernel packaging.
try adding swap?
that said, this is a recurring problem with bigger kernels (or debug kernels)

dtischler, 2023-07-23:
Added 32gb of Swap, but receive the same error during the process.  The build does consume up to the physical RAM limit, and then begin bleeding over into the Swap, but only maxes out at about 1 GB into the Swap.  I have another host, with 16gb RAM, let me try building there to see if that helps.

dtischler, 2023-07-24:
It gets a bit further on this 16gb RAM machine, but still fails, in a slightly different way but the symptom is the same:

...snip...

```
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8250-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8250-sony-xperia-edo-pdx203.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8250-sony-xperia-edo-pdx206.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8250-xiaomi-elish-boe.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8250-xiaomi-elish-csot.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8350-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8350-microsoft-surface-duo2.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8350-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8350-sony-xperia-sagami-pdx214.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8350-sony-xperia-sagami-pdx215.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8450-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8450-qrd.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8450-sony-xperia-nagara-pdx223.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8450-sony-xperia-nagara-pdx224.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8550-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/dtbs/qcom/sm8550-qrd.dtb
[🔨]   make: Nothing to be done for 'install'.
[🌱] Kernel built in [ 34 seconds - 6.4.5-arm64 ]
[🌱] Packaging kernel [ arm64 linux-uefi-arm64-edge ]
[🌱] Kernel .deb package version [ 23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-Cbd6cHfe66-HK01ba-Vc222-B3b41 ]
[🌱] Packaging linux-image [ arm64 linux-uefi-arm64-edge ]
[🔨]   [ 32M]  /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/image/boot
[🔨]   ├── [290K]  config-6.4.5-arm64
[🔨]   ├── [4.6M]  System.map-6.4.5-arm64
[🔨]   └── [ 28M]  vmlinuz-6.4.5-arm64
[🔨]   
[🔨]     32M used in 0 directories, 3 files
[🌱] Kernel-built image filetype [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 23 21:03 /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/image/boot/vmlinuz-6.4.5-arm64
[🔨]   
[🔨]   /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uSRgeI9lM7/image/boot:
[🔨]   total 33152
[🔨]   drwxrwxr-x 2 root root      100 Jul 23 21:03 .
[🔨]   drwxrwxr-x 3 root root       60 Jul 23 21:03 ..
[🔨]   -rw-rw-r-- 1 root root   297450 Jul 23 21:03 config-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root  4782212 Jul 23 21:03 System.map-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 23 21:03 vmlinuz-6.4.5-arm64
[🌱] Kernel image filetype after pre_package_kernel_image [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   removed '/home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.NDb68MxF5s/lib/modules/6.4.5-arm64/build'
[🔨]   removed '/home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.NDb68MxF5s/lib/modules/6.4.5-arm64/source'
[🔨]   [6.7K]  /home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.NDb68MxF5s/lib/modules/6.4.5-arm64/kernel
[🔨]   ├── [  60]  arch
[🔨]   ├── [ 100]  block
[🔨]   ├── [1.5K]  crypto
[🔨]   ├── [1.7K]  drivers
[🔨]   ├── [1.2K]  fs
[🔨]   ├── [ 540]  lib
[🔨]   ├── [ 100]  mm
[🔨]   ├── [1.1K]  net
[🔨]   └── [ 240]  sound
[🔨]   
[🔨]    6.7K used in 9 directories
[🌱] shellcheck found no problems in package scripts [ shellchecked 4 scripts in 'linux-image-edge-arm64' ]
[🔨]   dpkg-deb: building package 'linux-image-edge-arm64' in '/home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uFBojZi8Ix//linux-image-edge-arm64_23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-Cbd6cHfe66-HK01ba-Vc222-B3b41_arm64.deb'.
[🔨]   dpkg-deb: error: cannot append ar member file (data.tar) to '/home/david/build/.tmp/work-12d0fa78-1d1f-4372-8395-642d4a4dc196/tmp.uFBojZi8Ix//linux-image-edge-arm64_23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-Cbd6cHfe66-HK01ba-Vc222-B3b41_arm64.deb': failed to write (No space left on device)
[💥] Error 2 occurred in main shell [ at /home/david/build/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        fakeroot_dpkg_deb_build() --> lib/functions/compilation/packages/utils-dpkgdeb.sh:78
              create_kernel_deb() --> lib/functions/compilation/kernel-debs.sh:143
   prepare_kernel_packaging_debs() --> lib/functions/compilation/kernel-debs.sh:72
                 kernel_package() --> lib/functions/compilation/kernel.sh:177
[... cut ...]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-12d0fa78-1d1f-4372-8395-642d4a4dc196.log.ans https://paste.next.armbian.com/log ]
```

Interestingly this machine has more cores / threads, so I hit the RAM limit at 16gb (and start to once again use swap, but that doesn't seem to help...it really only seems to care about physical RAM). Though that makes me wonder if I can find the exact location in the build script that is spawning this process, and try to back off the number of cores with a `make -j2` or similar.  Even if it does take much longer, hopefully that gets it to complete.

rpardini, 2023-07-24:
This is quite extreme. 16gb RAM on an otherwise unloaded system should be enough.
During a-n, the testcase was building the `meson64` `edge` kernel on a 4gb system, and that worked back then.
I'm spinning up a 8gb RAM VM to check, I've been building on 64gb+ systems for a few months now.
I've a patch to completely disable tmpfs usage (using regular disk instead) you can try:

```patch
diff --git a/lib/functions/host/tmpfs-utils.sh b/lib/functions/host/tmpfs-utils.sh
--- a/lib/functions/host/tmpfs-utils.sh	(revision c62c639a04cad1f7b150bb7dbbb344c4ce5efd40)
+++ b/lib/functions/host/tmpfs-utils.sh	(date 1690192638507)
@@ -39,6 +39,9 @@
 	elif [[ "${ARMBIAN_INSIDE_DOCKERFILE_BUILD}" == "yes" ]]; then
 		display_alert "prepare_tmpfs_for: inside Dockerfile build, skipping" "${tmpfs_name}" "cleanup"
 		cleanup_params+=("no_umount_tmpfs")
+	elif [[ "${USE_TMPFS}" == "no" ]]; then
+		display_alert "prepare_tmpfs_for:" "USE_TMPFS=no, skipping: '${tmpfs_name}'" "warning"
+		cleanup_params+=("no_umount_tmpfs")
 	else
 		display_alert "prepare_tmpfs_for: on Linux and root, MOUNTING TMPFS" "${tmpfs_name}" "cleanup"
 		# mount tmpfs on it

```

Apply this & build with `USE_TMPFS=no`?

rpardini, 2023-07-24:
> Interestingly this machine has more cores / threads, so I hit the RAM limit at 16gb (and start to once again use swap, but that doesn't seem to help...it really only seems to care about physical [...]

This (make -jX) should be unrelated.
The kernel is originally built on regular disk (`cache/sources/...`).
It is only during _packaging_ that we copy the results to tmpfs, since we're gonna heavily work it/clean it specially during headers packaging.
Since we're using standard `tmpfs`, that is indeed limited by the total virtual memory space, but still would only ever use the physical RAM, possibly forcing other stuff to swap.

rpardini, 2023-07-24:
@dtischler 
cannot reproduce.
I've successfully built the uefi-arm64 kernel on a 7.5gb RAM virtual machine.

```
./compile.sh BOARD=uefi-arm64 BRANCH=edge kernel ARTIFACT_IGNORE_CACHE=yes
```

The tmpfs usage is quite modest, around 1.5gb, at its peak:
```
Filesystem     1M-blocks  Used Available Use% Mounted on
tmpfs               7379     3      7376   1% /armbian/.tmp/logs-889A1691-2E4C-42FB-8EEA-C9F4B240F007
tmpfs               7379  1427      5952  20% /armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007
```

Distributed the following way, at its peak:

````
2.8M	/armbian/.tmp/logs-889A1691-2E4C-42FB-8EEA-C9F4B240F007
212K	/armbian/.tmp/extensions-889A1691-2E4C-42FB-8EEA-C9F4B240F007
0		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR/etc
376M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR/lib
42M		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR/boot
34M		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR/usr
752K	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR/DEBIAN
452M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.kesRBcFAiR
112M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.9264VtN5iY
376M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.XC4gRcolZv/modules
42M		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.XC4gRcolZv/image
34M		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.XC4gRcolZv/dtbs
452M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/tmp.XC4gRcolZv
0		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/ccache_tmp
0		/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007/bin
1015M	/armbian/.tmp/work-889A1691-2E4C-42FB-8EEA-C9F4B240F007
1018M	/armbian/.tmp/
```

That said, make sure you actually got _free_ RAM at your host machine when building.

dtischler, 2023-07-24:
That `USE_TMPFS=no` patch worked perfect @rpardini !

I can now proceed through properly, and never even hit 1gb of RAM consumed, wow.

Odd that I received it on 2 separate machines, but oh well.

With `USE_TMPFS=no` in place I now receive:

```
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8350-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8350-microsoft-surface-duo2.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8250-xiaomi-elish-csot.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8350-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8350-sony-xperia-sagami-pdx214.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8350-sony-xperia-sagami-pdx215.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8450-hdk.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8450-qrd.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8450-sony-xperia-nagara-pdx223.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8450-sony-xperia-nagara-pdx224.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8550-mtp.dtb
[🔨]     INSTALL /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs/qcom/sm8550-qrd.dtb
[🔨]   make: Nothing to be done for 'install'.
[🌱] Kernel built in [ 41 seconds - 6.4.5-arm64 ]
[🌱] Packaging kernel [ arm64 linux-uefi-arm64-edge ]
[🌱] Kernel .deb package version [ 23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-Cbd6cHfe66-HK01ba-Vc222-B3b41 ]
[🌱] Packaging linux-image [ arm64 linux-uefi-arm64-edge ]
[🔨]   [ 32M]  /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/image/boot
[🔨]   ├── [290K]  config-6.4.5-arm64
[🔨]   ├── [4.6M]  System.map-6.4.5-arm64
[🔨]   └── [ 28M]  vmlinuz-6.4.5-arm64
[🔨]   
[🔨]     32M used in 0 directories, 3 files
[🌱] Kernel-built image filetype [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 24 05:56 /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/image/boot/vmlinuz-6.4.5-arm64
[🔨]   
[🔨]   /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/image/boot:
[🔨]   total 33160
[🔨]   drwxrwxr-x 2 root root     4096 Jul 24 05:57 .
[🔨]   drwxrwxr-x 3 root root     4096 Jul 24 05:56 ..
[🔨]   -rw-rw-r-- 1 root root   297450 Jul 24 05:56 config-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root  4782212 Jul 24 05:56 System.map-6.4.5-arm64
[🔨]   -rw-rw-r-- 1 root root 28860928 Jul 24 05:56 vmlinuz-6.4.5-arm64
[🌱] Kernel image filetype after pre_package_kernel_image [ vmlinuz-6.4.5-arm64: Linux kernel ARM64 boot executable Image, little-endian, 4K pages ]
[🔨]   removed '/home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.BSFqU2ZJSd/lib/modules/6.4.5-arm64/build'
[🔨]   removed '/home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.BSFqU2ZJSd/lib/modules/6.4.5-arm64/source'
[🔨]   [ 40K]  /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.BSFqU2ZJSd/lib/modules/6.4.5-arm64/kernel
[🔨]   ├── [4.0K]  arch
[🔨]   ├── [4.0K]  block
[🔨]   ├── [4.0K]  crypto
[🔨]   ├── [4.0K]  drivers
[🔨]   ├── [4.0K]  fs
[🔨]   ├── [4.0K]  lib
[🔨]   ├── [4.0K]  mm
[🔨]   ├── [4.0K]  net
[🔨]   └── [4.0K]  sound
[🔨]   
[🔨]     40K used in 9 directories
[🌱] shellcheck found no problems in package scripts [ shellchecked 4 scripts in 'linux-image-edge-arm64' ]
[🔨]   dpkg-deb: building package 'linux-image-edge-arm64' in '/home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.ceeCeW1fV0//linux-image-edge-arm64_23.08.0-trunk--6.4.5-Sbcec-D710f-P720e-Cbd6cHfe66-HK01ba-Vc222-B3b41_arm64.deb'.
[🌱] Packaging linux-dtb [ arm64 linux-uefi-arm64-edge ]
[🔨]   [ 31M]  /home/david/build/.tmp/work-99f6bccb-afde-48ad-acce-e57c303ee6d3/tmp.YzMrmp3iW1/dtbs
[🔨]   ├── [1.0M]  allwinner
[🔨]   │   ├── [10.0K]  sun50i-a100-allwinner-perf1.dtb
[🔨]   │   ├── [ 29K]  sun50i-a64-amarula-relic.dtb
[🔨]   │   ├── [ 29K]  sun50i-a64-bananapi-m64.dtb
[🔨]   │   ├── [ 28K]  sun50i-a64-nanopi-a64.dtb
[🔨]   │   ├── [ 27K]  sun50i-a64-oceanic-5205-5inmfd.dtb
[🔨]   │   ├── [ 29K]  sun50i-a64-olinuxino.dtb
[🔨]   │   ├── [ 29K]  sun50i-a64-olinuxino-emmc.dtb
[🔨]   │   ├── [ 29K]  sun50i-a64-orangepi-win.dtb
[... 41 lines cut ...]
[🌱] Getting sources from Git [ armbian-firmware-git master ]
[🌿] Fetching updates from remote repository [ armbian-firmware-git master ]
[🔨]   From https://github.com/armbian/firmware
[🔨]    * branch            master     -> FETCH_HEAD
[🌱] git: Fetch from remote completed, rev-parsing... [ 'armbian-firmware-git' 'master' 'FETCH_HEAD' ]
[🌱] Skipping checkout [ armbian-firmware-git master  ]
[🌱] Getting sources from Git [ linux-firmware-git main ]
[🌿] Fetching updates from remote repository [ linux-firmware-git main ]
^C[🚸] Build interrupted [ Build interrupted by SIGINT ]
[... cut ...]
[🚸] improved_git try 2 failed, retrying in 10 seconds [ git fetch --recurse-submodules=no --no-tags https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git main ]
^C[🚸] Build interrupted [ Build interrupted by SIGINT ]
```

So I am dying at trying to download `linux-firmware`:

```
david@david-HP:~$ git fetch --recurse-submodules=no --no-tags https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git main
fatal: not a git repository (or any of the parent directories): .git
```

But that is certainly unrelated and this issue can be closed out.

Thanks!

closed 2023-07-24 by igorpecovnik