armbian/build #5567: gzip: stdout: No space left on device

dtischler, 2023-08-08:
### What happened?

During a build of Rock5B, using Collabora as the kernel source, and flipping on a few GPU related options for Radeon, Nouveau, and AMDGPU (I fully realize they will not likely work due to BAR and GIC issues), build fails with the following log related to "gzip No space left on device":  https://paste.next.armbian.com/dofoheqaji

124gb of space available on the host, and I'm using the new `USE_TMPFS=no` to keep my memory usage down.    :-)

As usual, any ideas on what I could have done wrong?  Thanks!

### How to reproduce?

./compile.sh, then used the menu to select the Rock5B, chose server image, chose Collabora as kernel source, and enabled AMDGPU, Radeon, and Nouveau during kernel configuration (everything else left as-is).

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/dofoheqaji

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2023-08-08:
I think you ran out of space on boot partition of the image. Can you try building after adding `BOOTSIZE="512"` in `config/boards/rock-5b.conf`

dtischler, 2023-08-08:
Just added that to my config and starting a new build now.  Will find out if that fixes it in a bit, and report back.  Thanks @viraniac.

dtischler, 2023-08-08:
No luck, thought the failure is ever so slightly different:

```
mkimage: Write only 201781184/256621662 bytes, probably no space left on the device
   run-parts: /etc/initramfs/post-update.d//99-uboot exited with return code 1
```

Full log at:  https://paste.next.armbian.com/botigecago

(I accidentally chose Desktop image this time around as opposed to Server image on my initial attempt, so the packages do vary in the logs...but ultimately still fails).

viraniac, 2023-08-08:
interesting. you have a initramfs file thats more than 256MB in size. Looks like you enabled a lot  of things. You can further bumped the size if you want, and go for something like BOOTSIZE=1024 for example, but with initrd that big, I am not sure the system will boot correctly without modifying bootscripts. You can still give it a try though.

One possible solution can be to modify initramfs configuration to use a list of required modules instead of bundling all modules into initrd. I can share a patch for that tomorrow. Need to sleep now

dtischler, 2023-08-08:
No problem, no rush, and many thanks!

dtischler, 2023-08-08:
Side note:   The kernel configuration is whatever Collabora has set as default, with only Radeon, AMDGPU, and Nouveau being directly modified / enabled by me.  (Granted I imagine those are large).

dtischler, 2023-08-08:
Increasing to `BOOTSIZE=1024` did indeed allow the build to complete, and the board did successfully boot!   Thanks for the assistance.  :-)

closed 2023-08-08 by dtischler