armbian/build #5638: Package armbian-esp-cli-espressobin-current does not generate a FIT supporting EMMC boot variants

kainz, 2023-08-23:
### What happened?

The package armbian-esp-cli-espressobin-current is missing fixes that were implemented in PR #4521 . This causes any initramfs hook call to produce a DTB that will fail to boot a EMMC espressobin board.

### How to reproduce?

See https://github.com/armbian/build/blob/main/packages/bsp/mvebu64/initramfs/99-uboot-fit#L55 , the dtb 'board configurations' block and its references, vs https://github.com/armbian/build/blob/main/config/its/marvell/a37xx/espressobin.its#L55 .

Incidentally, the image build blob doesn't actually set stuff up correctly either, the `configurations` block at its end points to the wrong dtb inclusion block, so trying to boot a v7 emmc board with current images wouldn't work anyway if you are trying to use the onboard eMMC.

Previous work (PR#4521) references support work and dtb details in the thread at https://forum.armbian.com/topic/12393-espressobin-install-to-emmc/.

After fixing the embedded dts in *both* of these locations (to produce build images that can work correctly on an eMMC board, and to produce an initramfs hook that will not brick the system), booted images can see the eMMC mmcblk device, and armbian-install produces expected results.

This doesn't give you an easy way to install to eMMC espressobins out-of-the-box, but it does get you 99% of the way there.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kainz, 2023-08-23:
http://paste.armbian.com/nagetepufu.sh shows a 99-uboot-fit hook including the correct embedded DTS configuration to load the DTBs correctly. Do also note that https://www.armbian.com/espressobin/ doesn't document the (current?) FIT image booting method, or the impact of `/boot/armbianEnv.txt`'s `board_version` variable.

igorpecovnik, 2025-01-19:
We have dropped this board, closing.

closed 2025-01-19 by igorpecovnik