armbian/build #5675: Problem found ... on compile.sh

advancel, 2023-09-04:
### What happened?

A bug happened!
Err log is :
Err:5 http://ppa.launchpad.net/ntadej/tano/ubuntu focal Release
[🔨]     404  Not Found [IP: 185.125.190.52 80]
[🔨]   Get:6 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
[🔨]   Hit:7 http://security.ubuntu.com/ubuntu focal-security InRelease
[🔨]   Reading package lists...
[🔨]   E: The repository 'http://ppa.launchpad.net/ntadej/tano/ubuntu focal Release' does not have a Release file.

### How to reproduce?

Command you gave to the framework!

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

advancel, 2023-09-04:
@igorpecovnik  How can i solve this? I can't compile right now. 
I am trying to compile for Allwinner A13 by the if it helps.

igorpecovnik, 2023-09-04:
I don't know. Not our bug means your input is wrong and that we can't afford to investigate.

closed 2023-09-04 by igorpecovnik

viraniac, 2023-09-04:
@advancel There is next to no information in your log that you shared when you opened the issue. Whenever opening an issue, its best to share the complete log rather than just a piece of error message.

Still there are couple of things I can tell you from that error. You are running this on ubuntu 20.04, which is no longer a supported environment for running armbian build framework. You also seems to have some third party ppa configured on the system that gave the error which is not used by us.

Hence its not a issue of our framework, rather the issue is with your build environment

advancel, 2023-09-04:
> @advancel There is next to no information in your log that you shared when you opened the issue. Whenever opening an issue, its best to share the complete log rather than just a piece of error [...]

Oh, thank you a lot. 
I though checking repositories that belongs armbian's script, not mines.
Thank you again for explaining well. Unlike the others...