armbian/build #5682: When I first started configuring Armbian, I found that there was a dispute over the country where the configuration time was set

Zhonghui-Gao, 2023-09-06:
### What happened?

@clee @tmm1 @joekhoobyar @steev @zx2c4 
[image]
[image]
This is China Taiwan, China Hong Kong, not a country. This is the one China principle. Please add China before Hong Kong and Taiwan
Thx！

### How to reproduce?

Command you gave to the framework!

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

steev, 2023-09-06:
Why are you tagging me? If you want to see that done, do the work and submit a merge request.

igorpecovnik, 2023-09-06:
Even we are not a maintainer of locale/ language related packages, a [hack](https://github.com/armbian/build/commit/27df94e9240c8971a5a70332490c2df6b92b18f2) was applied as it was already **politely** asked. If this is not good enough and you don't know how to fix this further, **hire someone** and send a PR. This is how things works here - open source developers are not anyone slaves. I respect and understand your situation, but please stop abusing non profit organisations around the world that has nothing to do with this. Open source software is work of tens of thousands of people. As we (10-20 people that are maintaining Armbian) are not responsible for technical functioning of shipped software in any way, we are not responsible for what that software throws out (like in this case). This is written in a contract that exists between you and us. [Read it](https://www.gnu.org/licenses/old-licenses/gpl-2.0.html)!

Since this is not our issue, I have to close it down. If you want to change it, you know how: hire person, pay for the work, send a PR and make a donation to our project. And show some respect. We owe you nothing.

Thank you!

closed 2023-09-06 by igorpecovnik

Zhonghui-Gao, 2023-09-06:
> 

> Even we are not a maintainer of locale/ language related packages, a [hack](https://github.com/armbian/build/commit/27df94e9240c8971a5a70332490c2df6b92b18f2) was applied as it was already [...]

ok fine my problem sry

Zhonghui-Gao, 2023-09-06:
> Even we are not a maintainer of locale/ language related packages, a [hack](https://github.com/armbian/build/commit/27df94e9240c8971a5a70332490c2df6b92b18f2) was applied as it was already [...]

 I'm very sorry for wasting the administrator's time on this issue，please delete this issue

igorpecovnik, 2023-09-06:
> I'm very sorry for wasting the administrator's time on this issue

https://www.armbian.com/donate

> please delete this issue

We don't do that. I always try to reply impersonal, in case someone comes up with the same idea - answer is already here. You are second person asking for this and as you can see we tried to help. Obviously fix was not good enough, but more we can't afford to get involved.

Zhonghui-Gao, 2023-09-06:
> https://www.armbian.com/donate
>
>
> We don't do that. I always try to reply impersonal, in case someone comes up with the same idea - answer is already here. You are second person asking for this and [...]

Okay, thank you very much for replying to this question！