armbian/build #5719: the patch problem

Zhonghui-Gao, 2023-09-17:
### What happened?

[image]

### How to reproduce?

how I solve this？ thx！

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2023-09-17:
In order to answer how you can solve the error, We first need to know how to reproduce the error. Would it be possible for you to upload the complete build log somewhere. From the error report we can't tell what board you are trying to compile this for and for what kernel version. If its for some allwinner board with 6.2+ kernel, then we are already aware of the issue and working on it.

Zhonghui-Gao, 2023-09-18:
> 为了回答如何解决错误，我们首先需要知道如何重现错误。您是否可以将完整的构建日志上传到某处。从错误报告中，我们无法判断您尝试编译它的主板以及内核版本。

sry the **share_log** cant  share by url

Zhonghui-Gao, 2023-09-18:
> In order to answer how you can solve the error, We first need to know how to reproduce the error. Would it be possible for you to upload the complete build log somewhere. From the error report we can't tell what board you are trying to compile this for and for what kernel version.

sry, the **share_log** cant share by url

Zhonghui-Gao, 2023-09-18:
# Armbian ANSI build logs for 007c2e8d-72c8-4349-81af-dbaa3a4f9f85 - use "less -SR" to view
# Armbian build at Mon Sep 18 09:05:16 CST 2023 on NULL
------------------------------------------------------------------------------------------------------------
# Repeat build: ./compile.sh build BOARD=hm3568 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=jammy SHARE_LOG=yes
# ARGs: 'PREFER_DOCKER=no' 'BUILD_MINIMAL=no' 'BUILD_DESKTOP=no' 'ARMBIAN_RELAUNCHED=yes' 'BRANCH=edge' 'BOARD=hm3568' 'SET_OWNER_TO_UID=1000' 'RELEASE=jammy' 'KERNEL_CONFIGURE=no' 'SHARE_LOG=yes' 'build'
------------------------------------------------------------------------------------------------------------
# GIT revision
   commit fee5cd03b1a89320bc3332478578caa0c017aac6 (HEAD -> hm3568, origin/hm3568)
   Merge: dfabbb96e a190d11d7
   Author: Wenguo Liu <liuwenguo@gmail.com>

       Merge branch 'main' into hm3568
------------------------------------------------------------------------------------------------------------
# GIT status
   On branch hm3568
   Your branch is up to date with 'origin/hm3568'.

   Untracked files:
     (use "git add <file>..." to include in what will be committed)
        userpatches/customize-image.sh
        userpatches/kernel/rockchip64-edge/hm3568-dts.patch
        userpatches/kernel/rockchip64-edge/led.patch
        userpatches/kernel/rockchip64-edge/wol.patch

   nothing added to commit but untracked files present (use "git add" to track)
------------------------------------------------------------------------------------------------------------
### config_early_init.log
--> (0) INFO: Starting single build process [ hm3568 ]
--> (0) INFO: Checking [ basic host setup ]
--> (0) INFO: Build host OS release [ jammy ]
--> (0) INFO: Build host architecture [ amd64 ]
------------------------------------------------------------------------------------------------------------
### config_source_board_file.log
--> (0) INFO: Sourcing board configuration [ /home/gao/armbian-hm3568/config/boards/hm3568.conf ]
------------------------------------------------------------------------------------------------------------
### do_main_configuration.log
--> (0) INFO: Starting main configuration
--> (0) INFO: Using revision from [ main VERSION file: '23.08.0-trunk' ]
--> (0) INFO: Sourcing family configuration [ /home/gao/armbian-hm3568/config/sources/families/rk35xx.conf ]
--> (0) INFO: Enabling extension [ rkbin-tools ]
--> (0) INFO: Determined BOOT_SOC from BOOTCONFIG [ BOOT_SOC: 'rk3568'; BOOTCONFIG: 'hm-rk3568_defconfig' ]
--> (0) INFO: Sourcing arch configuration [ arm64.conf ]
--> (0) INFO: Extension manager [ processed 4 Extension Methods calls and 4 Extension Method implementations ]
------------------------------------------------------------------------------------------------------------
### config_post_main.log
--> (1) INFO: Configuration prepared for BOARD build [ hm3568.conf ]
--> (1) EXT: Repeat Build Options (early) [ ./compile.sh build BOARD=hm3568 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=jammy SHARE_LOG=yes ]
------------------------------------------------------------------------------------------------------------
### check_dir_for_mount_options.log
--> (1) INFO: Checked directory OK for mount options [ /home/gao/armbian-hm3568/.tmp ('main temporary dir') ]
------------------------------------------------------------------------------------------------------------
### prepare_host_noninteractive.log
--> (1) INFO: Preparing [ host ]
--> (1) COMMAND: update-ccache-symlinks
--> (1) INFO: Syncing clock [ host ]
--> (1) COMMAND: ntpdate pool.ntp.org
   18 Sep 09:04:52 ntpdate[1933675]: adjust time server 202.118.1.81 offset -0.050308 sec
--> (8) INFO: Ignoring toolchains [ SKIP_EXTERNAL_TOOLCHAINS: yes ]
------------------------------------------------------------------------------------------------------------
### aggregate_packages.log
--> (8) INFO: Aggregating packages [ rootfs ]
--> (8) INFO: Python3 version [ 3.10.12 - 'Python 3.10.12' ]
--> (9) INFO: pip3 version [ 22.0.2: 'pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)' ]
--> (9) INFO: Using cached pip packages for Python tools [ 85fcd277f85b8937fcd206c83809af0ac0d675d366d013cacc9851ab5dda88f3 ]
--> (9) COMMAND: [...shortened...] /usr/bin/python3 /home/gao/armbian-hm3568/lib/tools/aggregation.py
   Summary: Aggregation: debootstrap: 5; rootfs: 114; image: 1; desktop: 0; apt-sources: 0; hash: 79c2bd05b5dd31ed
--> (9) COMMAND: rm -f /home/gao/armbian-hm3568/.tmp/work-007c2e8d-72c8-4349-81af-dbaa3a4f9f85/tmp.t8ielaBIKA
------------------------------------------------------------------------------------------------------------
### clean_deprecated_mountpoints.log
--> (9) INFO: artifact [ uboot :: uboot() ]
[... 158 lines cut ...]
--> (31) ERROR: * applying patch/misc/wireless-driver-for-uwe5622-park-link-v6.1-post.patch [ failed ]
--> (31) ERROR: error! [ Patching error, exiting.  ]
--> (31) ERROR: Exiting with error 43 [ at /home/gao/armbian-hm3568/lib/functions/logging/traps.sh:1
                   exit_with_error() --> lib/functions/logging/traps.sh:1
                process_patch_file() --> lib/functions/compilation/patch/patching.sh:103
          driver_uwe5622_allwinner() --> lib/functions/compilation/patch/drivers_network.sh:644
      kernel_drivers_prepare_harness() --> lib/functions/compilation/patch/drivers-harness.sh:144
[... cut ...]
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:31
             do_with_default_build() --> lib/functions/main/default-build.sh:42
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                              main() --> compile.sh:50
    ]
--> (31) ERROR: Cleaning up [ please wait for cleanups to finish ]

viraniac, 2023-09-18:
@Zhonghui-Gao i believe your problem can simply be solved by running `git pull --rebase` as I think you are either missing [this change](https://github.com/armbian/build/pull/5704/files) or need to revert it. 

We are no longer using 6.4 kernels and this is not a device supported by us. Also Issues section is for reporting bugs in build framework, but this is more like you need help for your own fork or device that is not supported by Armbian. Hence I am going to close this issue.

Please feel free to ask for help on our discord channel or on forums though. And please describe things in details next time. Just posting a screenshot doesn't help to understand what might have gone wrong there.

closed 2023-09-18 by viraniac

Zhonghui-Gao, 2023-09-18:
> @Zhonghui-Gao i believe your problem can simply be solved by running `git pull --rebase` as I think you are either missing [this change](https://github.com/armbian/build/pull/5704/files) or need to [...]

thx!