armbian/build #5720: allwinner h6 ETH

ZYQsteve, 2023-09-18:
### What happened?

how to enable allwinner h6 eth 

### How to reproduce?

Command you gave to the framework!

### Branch

other

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ZYQsteve, 2023-09-18:
> Jira ticket: [AR-1869](https://armbian.atlassian.net/browse/AR-1869)

there isn't the tag nameed ar-1869

viraniac, 2023-09-18:
@ZYQsteve You haven't explained what device you are facing the issue with, but I am going to take a guess and say its Tanix TX6 based on your previous issue you have created. Issues section is for reporting bugs in build framework. 

Tanix tx6 is not a supported device in Armbian right now. And hence this doesn't qualifies as a bug/issue. Hence I am going to close this issue.

Still I was trying to fix AC200 based network for H6 devices, so could you please try [this Image](https://drive.google.com/drive/folders/1T2u9Cdn2QeQgnS3FkiaWxD5_GK3Rygeq?usp=sharing) and let me know if ethernet works for you with the shared image.

viraniac, 2023-09-18:
Also lets move rest of the discussion to either armbian forums or to armbian-allwinner discord channel

closed 2023-09-18 by viraniac

ZYQsteve, 2023-09-18:
> Also lets move rest of the discussion to either armbian forums or to armbian-allwinner discord channel
in many months ago i find a branch named tanix-tx6 ,I used it to compile an image that can make my tanix-tx6 ETH work，but now i can't find that branch

ZYQsteve, 2023-09-18:
There is an aw-h6-box.tvb file in this branch called ar-1487, and I compiled it to run the kernel error 'can‘t’ register mdio bus'

viraniac, 2023-09-18:
> There is an aw-h6-box.tvb file in this branch called ar-1487, and I compiled it to run the kernel error 'can‘t’ register mdio bus'

Apart from main branch, other branches probably can have unfinished changes. So can't say if that branch is going to work or not. How did it go with the images I shared? Were you able to get ethernet working with the same?

ZYQsteve, 2023-09-18:
> Apart from main branch, other branches probably can have unfinished changes. So can't say if that branch is going to work or not. How did it go with the images I shared? Were you able to get [...]

I encountered a kernel panic on my 3GB board

viraniac, 2023-09-18:
The helpful part of that error was going to be about 50 more lines after that message that you have posted above. As that part is not provided, hard to tell whats causing the issue.

ZYQsteve, 2023-09-18:
> The helpful part of that error was going to be about 50 more lines after that message that you have posted above. As that part is not provided, hard to tell whats causing the issue.

`[ 1392.205808] Unable to handle kernel paging request at virtual address 000000000000a548
[ 1392.208421] Unable to handle kernel paging request at virtual address 0000000000005784
[ 1392.209628] Unable to handle kernel paging request at virtual address 00060000430463e0
[ 1392.209646] Mem abort info:
[ 1392.209648]   ESR = 0x0000000096000004
[ 1392.209650]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 1392.209654]   SET = 0, FnV = 0
[ 1392.209656]   EA = 0, S1PTW = 0
[ 1392.209658]   FSC = 0x04: level 0 translation fault
[ 1392.209661] Data abort info:
[ 1392.209662]   ISV = 0, ISS = 0x00000004
[ 1392.209665]   CM = 0, WnR = 0
[ 1392.209667] [00060000430463e0] address between user and kernel address ranges
[ 1392.209673] Internal error: Oops: 0000000096000004 [#1] SMP
[ 1392.209678] Modules linked in: cfg80211 r8188eu(C) libarc4 uas sunrpc lz4hc lz4 zram binfmt_misc hci_uart btqca btrtl btbcm btintel bluetooth ir_nec_decoder polyval_ce ecdh_generic sunxi_cedrus(C) polyval_generic snd_soc_hdmi_codec r
c_tanix_tx5max v4l2_mem2mem snd_soc_simple_card sunxi_cir snd_soc_spdif_tx videobuf2_dma_contig snd_soc_simple_card_utils rfkill rc_core ecc videobuf2_memops sun4i_spdif videobuf2_v4l2 videobuf2_common videodev mc panfrost gpu_sched drm
_shmem_helper dump_reg cpufreq_dt fuse dm_mod ac200_ephy_ctl ac200_phy dw_hdmi_cec dw_hdmi_i2s_audio dwmac_sun8i mdio_mux ac200 pwm_sun4i display_connector i2c_gpio
[ 1392.209790] CPU: 1 PID: 2591 Comm: systemd Tainted: G         C         6.1.53-current-sunxi64 #1
[ 1392.209796] Hardware name: Tanix TX6 (DT)
[ 1392.209799] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1392.209805] pc : __d_lookup+0x64/0x144
[ 1392.209818] lr : d_lookup+0x44/0x6c
[ 1392.209822] sp : ffff80000b7a39b0
[ 1392.209824] x29: ffff80000b7a39b0 x28: ffff80000948f944 x27: ffff80000b7a3b18
[ 1392.209832] x26: 0000000000000000 x25: ffff000006717840 x24: 00000000ffffffec
[ 1392.209839] x23: 00000000cf4a36dd x22: 0000000000000000 x21: 0000000000000001
[ 1392.209846] x20: ffff8000094585c0 x19: 00060000430463c8 x18: 0000000000000000
[ 1392.209853] x17: 0000000000000000 x16: 0000000000000000 x15: ffff00000856c680
[ 1392.209860] x14: 0000000000000000 x13: ffff80000946e810 x12: ffff80000946e788
[ 1392.209867] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
[ 1392.209874] x8 : 000000000065726f x7 : d0d0d0d0b5a2bfb3 x6 : 0000000000000005
[ 1392.209881] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffff80000b7a3b18
[ 1392.209888] x2 : ffff80000945b000 x1 : ffff0000b4400000 x0 : 0000000000067a51
[ 1392.209896] Call trace:
[ 1392.209899]  __d_lookup+0x64/0x144
[ 1392.209904]  d_lookup+0x44/0x6c
[ 1392.209908]  lookup_dcache+0x24/0x8c
[ 1392.209913]  __lookup_hash+0x24/0xd0
[ 1392.209917]  do_unlinkat+0x13c/0x284
[ 1392.209924]  do_coredump+0x3d0/0x13c0
[ 1392.209932]  get_signal+0x650/0x870
[ 1392.209940]  do_notify_resume+0x178/0x12b0
[ 1392.209948]  el0_svc+0x74/0x84
[ 1392.209957]  el0t_64_sync_handler+0xbc/0x140
[ 1392.209963]  el0t_64_sync+0x18c/0x190
[ 1392.209973] Code: 52800035 14000003 f9400273 b4000293 (b9401a63) 
[ 1392.209977] ---[ end trace 0000000000000000 ]---
[ 1392.213809] Mem abort info:
[ 1392.221742] Mem abort info:
[ 1392.221753]   ESR = 0x0000000096000004
[ 1392.221757]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 1392.221762]   SET = 0, FnV = 0
[ 1392.221764]   EA = 0, S1PTW = 0
[ 1392.221767]   FSC = 0x04: level 0 translation fault
[ 1392.221771] Data abort info:
[ 1392.221772]   ISV = 0, ISS = 0x00000004
[ 1392.221774]   CM = 0, WnR = 0
[ 1392.221777] user pgtable: 4k pages, 48-bit VAs, pgdp=0000000045c90000
[ 1392.221783] [0000000000005784] pgd=0000000000000000, p4d=0000000000000000
[ 1392.221797] Internal error: Oops: 0000000096000004 [#2] SMP
[ 1392.221802] Modules linked in: cfg80211 r8188eu(C) libarc4 uas sunrpc lz4hc lz4 zram binfmt_misc hci_uart btqca btrtl btbcm btintel bluetooth ir_nec_decoder polyval_ce ecdh_generic sunxi_cedrus(C) polyval_generic snd_soc_hdmi_codec r
c_tanix_tx5max v4l2_mem2mem snd_soc_simple_card sunxi_cir snd_soc_spdif_tx videobuf2_dma_contig snd_soc_simple_card_utils rfkill rc_core ecc videobuf2_memops sun4i_spdif videobuf2_v4l2 videobuf2_common videodev mc panfrost gpu_sched drm
_shmem_helper dump_reg cpufreq_dt fuse dm_mod ac200_ephy_ctl ac200_phy dw_hdmi_cec dw_hdmi_i2s_audio dwmac_sun8i mdio_mux ac200 pwm_sun4i display_connector i2c_gpio
[ 1392.221917] CPU: 3 PID: 618 Comm: systemd-journal Tainted: G      D  C         6.1.53-current-sunxi64 #1
[ 1392.221927] Hardware name: Tanix TX6 (DT)
[ 1392.221930] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1392.221937] pc : __d_lookup_rcu+0x4c/0xf4
[ 1392.221957]   ESR = 0x0000000096000004
[... 21 lines cut ...]
[ 1392.234926]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 1392.234931]   SET = 0, FnV = 0
[ 1392.234933]   EA = 0, S1PTW = 0
[ 1392.234935]   FSC = 0x06: level 2 translation fault
[ 1392.234938] Data abort info:
[ 1392.234939]   ISV = 0, ISS = 0x00000006
[ 1392.234942]   CM = 0, WnR = 0
[ 1392.234944] swapper pgtable: 4k pages, 48-bit VAs, pgdp=000000004139c000
[ 1392.234948] [fffffbffff000348] pgd=000000004175b003, p4d=000000004175b003, pud=000000004175c003, pmd=0000000000000000
[ 1392.236581]   SET = 0, FnV = 0
[... cut ...]
[ 1392.244948]   FSC = 0x04: level 0 translation fault
[ 1392.244953] Data abort info:
[ 1392.248087]  __arm64_sys_unlinkat+0x40/0x8c
[ 1392.248098]  invoke_syscall+0x48/0x114
[ 1392.248107]  el0_svc_common.constprop.0+0xd4/0xfc
[ 1392.248113]  do_el0_svc+0x30/0xd0
[ 1392.248124]   ISV = 0, ISS = 0x00000004
[ 1392.252995]  el0_svc+0x2c/0x84
[ 1392.253010]  el0t_64_sync_handler+0xbc/0x140
[ 1392.253018]  el0t_64_sync+0x18c/0x190
[ 1392.253030] Code: 9280000e 14000003 f9400021 b4000441 (b85fc02a) 
[ 1392.253053]   CM = 0, WnR = 0`

ZYQsteve, 2023-09-18:
When I set mem=2048M on bootars, the kernel panic disappeared

viraniac, 2023-09-18:
Thanks for sharing this. I will try to find out the cause later. Meanwhile, one another user reported that 6.5.3 edge image works fine on his 3GB model. That image was also present on the link that I shared. Could you please give that a try as well?

EvilOlaf, 2023-09-18:
This issue tracker is for problems with _building_ Armbian. For problems while _running_ Armbian use community forums/chat.