armbian/build #5729: NixOS + HomeManager used to manage `git` causes build failure

Kreyren, 2023-09-21:
### What happened?

```
╰───❯ ./compile.sh BOARD=olimex-teres-a64 BRANCH=edge RELEASE=bookworm
error: could not lock config file /home/raptor/.config/git/config: Read-only file system
[💥] Error 255 occurred in main shell [ at REDACTED/armbian-build/lib/functions/general/git.sh:60
      git_ensure_safe_directory() --> lib/functions/general/git.sh:60
                           main() --> ./compile.sh:47
 ]
 ```
 
 > error: could not lock config file /home/raptor/.config/git/config: Read-only file system

### How to reproduce?

in home.nix:
`programs.git.enable = true;`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ManoftheSea, 2023-09-21:
As mentioned, this bug will apply any time the global git configuration is locked down such that the armbian build system can't change the global state.  Armbian should really not be accepting a security risk for the user's entire system like this.

Kreyren, 2023-09-21:
Self-assigning

Kreyren, 2023-09-21:
@igorpecovnik Heads-up: Critical bug fix: https://github.com/armbian/build/pull/5730

igorpecovnik, 2023-09-21:
This was necessary hack at that time. This needs to be explored if its still need. Do not panic and if bug exists only Nix, its ignored. I told you that I don't want to hear anything Nix related here.

Reopen when you understand why this hack was made.

igorpecovnik, 2023-09-21:
> This is not nix-only issue

NiX is not suitable for dealing with custom hardware and this is not our problem.

> It's still a not a good idea, the CVE mentions windows, but this can affects linux as well it can be used to make operations outside the git directory.

Until you don't research why this was introduced, you don't need to start working on a fix. You have to understand the background ... but its enough that issue is reported. It will be addressed when it will be. Case closed until then.

Kreyren, 2023-09-28:
@igorpecovnik fwiw i handled this scenario on my end by removing the function and providing instructions for the users on how to configure `safe.directory` namely adding:

```
[safe]
  directory = /path/to/armbian-build
  directory = /path/to/armbian-build/cache/*
  directory = /path/to/armbian-build/cache/sources/*
  directory = /armbian
  directory = /armbian/cache/*
  directory = /armbian/cache/sources/*
  ```

closed 2024-01-26 by igorpecovnik