armbian/build #5744: rock 5b build with  CRYPTROOT_ENABLE=yes causes Linux failed to start

sputnik2019, 2023-09-26:
### What happened?

rock 5b build with  CRYPTROOT_ENABLE=yes is ok，but boot armbian  is fail

build：
./compile.sh build BOARD=rock-5b BRANCH=legacy BUILD_DESKTOP=no BUILD_MINIMAL=yes CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=1234 KERNEL_CONFIGURE=no RELEASE=jammy 

### How to reproduce?

./compile.sh build BOARD=rock-5b BRANCH=legacy BUILD_DESKTOP=no BUILD_MINIMAL=yes CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=1234 KERNEL_CONFIGURE=no RELEASE=jammy 

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/ukekikuyev.yaml

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2023-09-26:
@sputnik2019 I don't see any error messages in the shared log. It looks more like its might be waiting for some input. Do you have a display connected to the device? If yes then you should have a password prompt there to enter your cryptfs password. If you want to run it headless with password being entered via serial console, I will recommend adding DEFAULT_CONSOLE="serial" to the build command so that password prompt will be visible over the serial console.

sputnik2019, 2023-09-26:
yes it ok!ths

viraniac, 2023-09-26:
Closing as it is confirmed to work fine.

closed 2023-09-26 by viraniac