armbian/build #5764: fetch from http://apt.armbian.com/dists fails; compile halts.

wwortel, 2023-09-30:
### What happened?

Towards the end of compiling Jammy/edge/no-desktop for rpi4b this error occurs:
E: Failed to fetch http://apt.armbian.com/dists/jammy/jammy-desktop/Contents-arm64  File has unexpected size (1003301 != 1004871). Mirror sync in progress? [IP: 93.103.15.56 80]

Already for more than 24 hours. Found old thread on setting alternative source, but listing alternatives it turns out there are none.
Not sure why it does desktop stuff to start with when I want an image without. In any case cannot proceed now.

### How to reproduce?

compile.sh

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-10-01:
To avoid usage of Armbian's repos during image build, set `SKIP_ARMBIAN_REPO=yes`

wwortel, 2023-10-01:
@rpardini 
Thanks! Your suggestion works and could complete the build.

igorpecovnik, 2023-10-03:
This could be classified as temporally network issue, and this issue more as a support req. Anyway, solved, closing.

closed 2023-10-03 by igorpecovnik