armbian/build #5786: RPi: add the ability to USB boot

zieglerjm, 2023-10-04:
### Which feature would you like to have?

The current RPi build process must be using old fat32 partition files.  To get "/"(root) to boot from USB SSD I have to use the fat partition on a SD card currently.  The current RPi native image doesn't have this problem so I am assuming it is old files in the fat partition.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2023-10-04:
Untrue. If you have an updated RPi4b firmware (eeprom), and write the Armbian image _properly_ to an USB disk, it boots without any problem.

zieglerjm, 2023-10-04:
Then it is a 3B fat partition problem then.  Just notices with the SD card inserted, "/"(root) does map to the USB drive, but it is still using something from the SD card at boot.  I don't have this problem with the RPi OS image & firmware is latest.