armbian/build #5796: RockPro64: Building with a specific kernel version, but applying patches fails

lasermind, 2023-10-06:
### What happened?

Since you are a bit quick with closing, without hearing back from an issue opener, I need to now open a new issue.

First of all: thank you for your effort in issue https://github.com/armbian/build/issues/5788 and thanks for pointing out the `KERNEL_MAJOR_MINOR` option. Where should I have seen this documented beforehand? I only found [this](https://forum.armbian.com/topic/5180-lock-to-a-particular-kernel-version/) and [that](https://forum.armbian.com/topic/7798-how-to-build-specific-kernel/) concerning this topic. (And of course [this](https://github.com/armbian/build/blob/main/config/boards/README.md) and [that](https://docs.armbian.com/Developer-Guide_Build-Options/), before you ask.)

—

**So, what happened:**

I'm trying to build Armbian for the RockPro64 with a specific kernel version: 5.15.93. This has worked before, but now suddenly, it does not work any more and breaks. Also with your suggested option `KERNEL_MAJOR_MINOR` set to "5.15" it does **not** compile. Now suddenly there are even more patches that want to be applied and this fails. See my new log pasted below.

Are there any suggestion to remedy these new errors, shown in the log?

**Thanks again!**

—

Btw: other than you noted in https://github.com/armbian/build/issues/5788, the option `BRANCH` becomes not at all meaningless when using `KERNELBRANCH` and `KERNEL_MAJOR_MINOR`. `BRANCH`, in that regard, still determines a repo from where the files are pulled from. This repo can be different for `legacy`, `current` and `edge`. Also I do need to have one of these three selected; the `BRANCH` option does not become obsolete or optional while using `KERNELBRANCH`.

### How to reproduce?

```
compile.sh \
  KERNEL_CONFIGURE=no \
  BRANCH=current \
  RELEASE=bullseye \
  BUILD_DESKTOP=no \
  BUILD_MINIMAL=no \
  BOARD=rockpro64
```
~/build/userpatches/**lib.config:**
```
KERNELBRANCH="tag:v5.15.93"
KERNEL_MAJOR_MINOR="5.15"
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/uhibowimad

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-10-06:
I don't understand why you are re-opening ticket. Everything is in the logs - patches are not compatible with the version you have chose.

Here you can co-sponsor development of feature which will mitigate this problem in the future:
https://github.com/sponsors/armbian

Currently, if you want to stick to some specific old version, you need to find patches for that version or adjust them manually. We don't have any mechanism to make it easier yet.

viraniac, 2023-10-06:
@lasermind I forgot two things when closing the issue. I am sorry for that.

1. you also need to set `KERNELPATCHDIR="archive/${LINUXFAMILY}-${KERNEL_MAJOR_MINOR}"`
2. As this is not a bug in the system, please move this to forum if you continue facing the problem

> Also I do need to have one of these three selected; the `BRANCH` option does not become obsolete or optional while using `KERNELBRANCH`

I know that its still needed, what I meant by obsolete is that whatever customizations that developers have done based on the branch (and hence the kernel version associated with it) in board or family configuration code, might now not be the best option anymore as you are trying to override what is tested with your combination.

If you still face issues with compilation or functionality, please move to forums instead of opening issues on forum. If you are doing this because you are using that configuration in a commercial product, I would suggest to book a consultation call - https://calendly.com/armbian/consultation. This will make us aware that you need this for some commercial product and we can then possibly become partners and keep you supported and your configuration tested for the same.

lasermind, 2023-10-06:
So basically what you're saying is: "It is normal for us to constantly break things and if you complain about it, you are invited to pay us money." Wow, ok …

**My suggestion to you is:** take a month off, don't break stuff and think about how to manage and <b>document</b> this project better. I believe, this would avoid 80% of "support tickets" you're stuggling with and save a lot of time on **all** sides (yes, yours included). Maybe people don't like to donate and contribute to a mess – but to a stable and reliable project, where the maintainers themselves are not constantly overstrained.

(And I don't mean this 👆 in a rude way, not at all, but because Armbian is important and should be self-aware of the fact that it is.)

—

@viraniac: Thank you, I'll try my luck with `KERNELPATCHDIR`. Eventually, is it possible for you to point me to a documentation of how I can exclude seemingly non-essential patches for my purpose or "find working ones", like Igor suggested?

If not and if you don't have any more tricks to pull off the sleeve, you can close here. Thanks so far!

igorpecovnik, 2023-10-06:
>  you are invited to pay us money

Perhaps then you will start to respect efforts we are putting into you?

>  manage and document this project better

Useless. You don't read.

closed 2023-10-06 by igorpecovnik

igorpecovnik, 2023-10-06:
> is it possible for you to point me to a documentation

Most of things you are asking are coming with a package of "basics of handling with source code". It is pretty obvious that you lack those skills. Which is fine. Stick to unchanged build framework, listen and stay away from suggesting. 

Open source world is a big mess, a chaos, which few people can't put into order. But what we are doing with this framework is masking this complexity, so you are not scared away. From open source, not from Armbian per se.

We work hard to make it stable, so you can use it. If you step out of this path, you should be aware that you are on your own. You are on your own in any case as this is community open source project, which we finance with our own cash up to around 95%, probably higher. Software is released under certain terms ... It's not a commercial product, you pay nothing for its use, but if you need support it is expected that you pay. Why would you steal from us in first place? What did we do so bad to you?

If you don't have skills, you will not survive (Armbian only helps you) and worse things you can do is shouting on us. If you don't listen, but you expect we will drag you around and educate, you need to pay at least with complete obedience and respect towards our time. Or do things you are capable of doing. Anywhere.

This tooling is not a magic stick. It has limits. It still need to be driven with a person that knows what he is doing. And worst of all, we can't possible finance this with donation. What we get from you in one year, we lost more in one day, that its possible to collect with donations.

We can only finance dev with our own cash or by selling you professional support. Nobody is forcing you to pay. Invest your own time and learn what you are missing. When you will do that, you will be very sorry that you didn't rather pay us to deal with your shit. We are not responsible for lack of your knowledge or lack of money you have put into your education. That is not our problem. Be fair.

We are happy that you use our build framework, but not at any cost.

viraniac, 2023-10-06:
> So basically what you're saying is: "It is normal for us to constantly break things and if you complain about it, you are invited to pay us money." Wow, ok …
> 
> **My suggestion to you is:** take a month off, don't break stuff and think about how to manage and document this project better. I believe, this would avoid 80% of "support tickets" you're stuggling with and save a lot of time on **all** sides (yes, yours included). Maybe people don't like to donate and contribute to a mess – but to a stable and reliable project, where the maintainers themselves are not constantly overstrained.
> 
> (And I don't mean this 👆 in a rude way, not at all, but because Armbian is important and should be self-aware of the fact that it is.)

@lasermind We are hardly 5-10 people trying to maintain an open source software with almost non-existent funding. We give our precious hours to move the project forward and somehow you expect that we spend time testing maintaining patches for all existent kernels, testing all possible configurations just because 1 user thinks its bad to upgrade. If you really want to stick to older kernel, either get old image from archive and only upgrade userspace packages or go back in git history to the point where 5.15 kernel was supported and fork the project from there. Spend a month maintaining that fork and maintaining your single kernel and you might know a fraction of things we do. Maybe now after that you can tell us what was so bad for us to ask for some payment in return of using our software in a commercial project.