armbian/build #5798: Broken apt sources make build fail

mjtimblin, 2023-10-06:
### What happened?

All builds that I've tried from main fail due to broken apt sources. I get the following output:

```
[🐳|🔨]   E: Failed to fetch http://security.debian.org/dists/bookworm-security/main/binary-arm64/Packages  Could not change permissions for temporary file /var/lib/apt/lists/partial/security.debian.org_dists_bookworm-security_main_binary-arm64_Packages.3nTJ0x - fchmod (1: Operation not permitted)
[🐳|🔨]   E: Failed to fetch http://deb.debian.org/debian/dists/bookworm/main/i18n/Translation-en  404  Not Found [IP: 199.232.30.132 80]
[🐳|🔨]   E: Failed to fetch http://deb.debian.org/debian/dists/bookworm-updates/main/binary-arm64/Packages  Could not change permissions for temporary file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_bookworm-updates_main_binary-arm64_Packages.KIhczL - fchmod (1: Operation not permitted)
[🐳|🔨]   E: Failed to fetch http://deb.debian.org/debian/dists/bookworm-backports/main/binary-arm64/Packages  Could not change permissions for temporary file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_bookworm-backports_main_binary-arm64_Packages.LyFuuY - fchmod (1: Operation not permitted)
[🐳|🔨]   E: Some index files failed to download. They have been ignored, or old ones used instead.
[🐳|🚸] Command failed, retrying in 5s [ chroot_sdcard_apt_get_update ]
[🐳|🚸] Command failed 3 times, giving up [ chroot_sdcard_apt_get_update ]
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/rootfs/distro-agnostic.sh:226
```

### How to reproduce?

I'm running the command from within WSL2 on Windows 11 with Docker Desktop running. I ran the following build command:
```
./compile.sh \
  BOARD=orangepi5 \
  BRANCH=legacy \
  RELEASE=bookworm \
  BUILD_MINIMAL=yes \
  BUILD_DESKTOP=no \
  KERNEL_CONFIGURE=no \
  SHARE_LOG=yes
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log URL

https://paste.next.armbian.com/iquculijin

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-10-09:
> with Docker Desktop running

Don't know if this was ever tested. We tested on several different machines with default WSL2 installation. It worked even on ARM based Windows 11. Can this be disabled somehow?

rpardini, 2023-10-09:
This scenario is untested, I was not even aware Docker Desktop would infiltrate wsl2.
Please post the `.docker.` launcher logs as well, they'll be in `output/logs` folder.

Try to avoid using docker:
`PREFER_DOCKER=no ./compile.sh ....` (`PREFER_DOCKER`=no has to be an env, not a cmdline param, thus needs to come first)

rpardini, 2023-10-10:
My guess is this Docker Desktop WSL2 backend is messing up the Docker volumes.
Try cleaning all volumes? (`docker system prune -a` or some such)
Try running with `DEBUG=yes` and see if that gives you any hints in the log-docker.

Otherwise yeah use WSL2 jammy sans-docker.

dieter-l-git, 2023-10-12:
I did not use Docker Desktop, I'm using WSL2 Ubuntu 22.04 under Win10 and installing docker on Ubuntu:

sudo apt-get install docker-ce docker-ce-cli containerd.io docker-buildx-plugin docker-compose-plugin

With this Build is working

rpardini, 2023-10-13:
WSL2 jammy is the only supported build host on Windows.
Using any kind of Docker is unsupported, might or not work.

closed 2023-10-13 by rpardini

xfl12345, 2024-02-22:
Just
`sudo hwclock --hctosys`

Solution from [https://askubuntu.com/questions/1096930/sudo-apt-update-error-release-file-is-not-yet-valid](https://askubuntu.com/questions/1096930/sudo-apt-update-error-release-file-is-not-yet-valid)