armbian/build #5853: KDE build seems to be missing features,

philippe44, 2023-10-23:
### What happened?

When building for recent distribution (Bullseye, Bookworm) and selecting KDE for a uefi-x86 platformn a fews application are missing and for example "settingsmanager" is not there, although can be installed manually via apt-get. Whern trying to edit the menu, nothing happens. I've tried 23.08 and latest main. branches. It's all good using jammy. 

Is this known ? Shall I try to investigate? I'm new to building Armbian, but I'm from the Slackware generation 😄.

Congratulation for Armbian work anyway it is amazing!

### How to reproduce?

Command you gave to the framework!just build

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-10-23:
> Is this known ?

No. We don't have anyone dealing with KDE so its unsupported / in unknown state. We have some resources behind Gnome / XFCE / Cinnamon. General guidance for desktops is to have some minimal base and bulk of applications in app groups. So we can have choose between minimal or full desktop. 

> but I'm from the Slackware generation

I could say I was raised there too ;) [This](http://www.slackware.com/announce/1.0.php) was my 1st ever Linux install.

> Congratulation for Armbian work anyway it is amazing!

Thank you!

philippe44, 2023-10-23:
Ahah :-) I used to have pizza parties back in the days where the mission of the existing group members was to bring at least 2 copies of Linux to newbies, so we had to build 2 sets of 20 x 1.44MB floppies!

I'll try to look at this KDE issue (FYI, the size of the build (same features checked) is ~12GB with Jammy and less then 9GB with Bullseye, so obviously something is missing, unless software size is shrinking which would be something new...

philippe44, 2023-10-24:
So ... 

- ~/build/config/desktop/bullseye/environments/kde-plasma/config_base/packages is missing "systemsettings" in packages list. 
- "plasma-discover" is also missing and makes the packager manager missing, BUT, when added to create a bullseye version, this causes the user creation on firstrun to misbehave. The /home/<userdir> is still owned by root:root and when launching desktop after first run, it does not end well...
- funny enough, that issue does not happen with bookworm. I have no idea what is causing what I'd assume the chmod on /home/<user> to fail. I'm not familiar enough with the build mechanism of Armbian to figure out.
- in the same "packages" file, the "gnome-orca" application cannot be installed when building bookworm (fine with bullseyes) and "fonts-ubuntu-font-family-console" must be removed as well

philippe44, 2023-10-25:
See https://github.com/armbian/build/pull/5863

igorpecovnik, 2023-10-25:
> obviously something is missing

Nothing unusual.

> BUT, when added while creating a bullseye version, this causes the user creation on firstrun to misbehave. The /home/ is still owned by root:root and when launching desktop after first run for that , it does not end well...

Odd. Armbian first run user creation is [fairly straight](https://github.com/armbian/build/blob/main/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L255-L308). And it works as expected at desktops we use primarily. Not much is going on here. Some dependency must be missing, so script fails at some point?

igorpecovnik, 2023-10-25:
> The /home/ is still owned by root:root and when launching desktop after first run for that , it does not end well...

This was reported elsewhere too. It seems related to build host environment ...

closed 2023-11-02 by philippe44