armbian/build #5864: Building a system using main and for current/bullseyse/KDE generate a "ghost" user in image

philippe44, 2023-10-25:
### What happened?

This is an interesting one that cost me a few brain cells. I wanted to build for KDE and I realized that there was some missing packages in the config, see https://github.com/armbian/build/pull/5863. After these changes, building for bullseye and bookworm worked but when booting the resulting image in VirtualBox for bullseye and creating a user at first run, then the desktop would not start for that user and I realized that `/home><user>` was owned by root:root. So desktop could not create anything under that home directory. That did **not** happen with bookworm. 

Took me a long time to realize that the problem is not in the first run script, but in the created image *itself* that already contains "user" and a `/home><user>` with the wrong permission. It's difficult to believe, but see below for (some) explanations.

### How to reproduce?

- get latest uefi_x86/bookworm/cinnamon in VirtualBox as the build machine
- create user "bob" in that machine, log under "bob"
- clone the github main branch
- run ./compile.sh EXPERT="yes"
- select current kernel, uefi_x86, bullseye, kde
- build

The resulting image has a "/home/bob" directory that belongs to root:root so when starting that image and creating a user "bob", things go wrong... The created image has also been mounted *before* a first boot inside the build machine to confirm that "/home/bob" is indeed there.

I have no idea why this happens and why the build process creates a home dire of the name of the username that does the build. This doe snot happen when building bookworm, the /home is clear of anything.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2023-10-25:
Same here for gnome jammy. user's home directory is owned by root.

rpardini, 2023-10-25:
I bet there is some leakage of `$HOME` or `$USER` or such in the desktop scripts....

rpardini, 2023-10-25:
Hmm, couldn't reproduce.
Command line: `PREFER_DOCKER=no ./compile.sh build BOARD=uefi-x86 BRANCH=edge BUILD_DESKTOP=yes DESKTOP_APPGROUPS_SELECTED=browsers DESKTOP_ENVIRONMENT=kde-plasma DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base EXPERT=yes KERNEL_CONFIGURE=no RELEASE=bullseye`

Resulting image has empty `/home`.
This was done under x86 jammy, with a `human` regular user, no Docker. What am I missing @philippe44 ?

amazingfate, 2023-10-25:
I reproduces this issue on rock5b jammy to build arm64 image for xiaomi-elish. Here is the log: https://paste.next.armbian.com/nusafoqape

amazingfate, 2023-10-25:
I have one ppa added and here is what in the ghost home directory:
```
$ sudo ls -l home/jfliu/.launchpadlib/api.launchpad.net/cache/
总计 2228
-rw------- 1 root root    2891 10月 25 21:21  api.launchpad.net,devel,-application,json,fc4e1e3a03117146fb1e9d492ab76690
-rw------- 1 root root 2268952 10月 25 21:21  api.launchpad.net,devel,-application,vnd.sun.wadl+xml,2f09acb494bdefdbf8ef0d1396a05e86
-rw------- 1 root root    2500 10月 25 21:21 'api.launchpad.net,devel,~liujianfeng1994,+archive,ubuntu,qcom-mainline,ws.op=getSigningKeyData-application,jso,713e13196fbfb4fbf67f42cbdc348ca5'
-rw------- 1 root root    2307 10月 25 21:21 'api.launchpad.net,devel,~liujianfeng1994,name=%22qcom-mainline%22&ws.op=getPPAByName-application,json,3e7f57526cbe06a4718fa35e0f08433f'
```

philippe44, 2023-10-25:
> Hmm, couldn't reproduce. Command line: `PREFER_DOCKER=no ./compile.sh build BOARD=uefi-x86 BRANCH=edge BUILD_DESKTOP=yes DESKTOP_APPGROUPS_SELECTED=browsers DESKTOP_ENVIRONMENT=kde-plasma [...]
Maybe because my build was made under armbian/bookworm/current/uefi_x86?

amazingfate, 2023-10-26:
I added `chroot_sdcard rm -rf /home/*` after `do_with_retries 3 chroot_sdcard add-apt-repository ppa:liujianfeng1994/qcom-mainline --yes --no-update` then the issue is gone.

amazingfate, 2023-10-26:
@philippe44 You can mount the image you build to see what kind of file is in the home directory. In my situation I have launpad cache used for adding ppa repo.
I think we can add clean up step to clean home dir after every chroot operation @rpardini .

philippe44, 2023-10-26:
> @philippe44 You can mount the image you build to see what kind of file is in the home directory. In my situation I have launpad cache used for adding ppa repo. I think we can add clean up step to clean home dir after every chroot operation @rpardini .
>
I already did. Assuming that "user" is the user that launched "compile.sh", I have `/home/<user>` owned by root. This was why I was getting insane because when I was using that image, I was creating the same "user" in first run (it's a recurring name I'm using) and then when KDE was launched, it was failing. As soon as I was using any other name, it was working and it took me a while to make the connection 😄.

amazingfate, 2023-10-26:
> > @philippe44 You can mount the image you build to see what kind of file is in the home directory. In my situation I have launpad cache used for adding ppa repo. I think we can add clean up step to clean home dir after every chroot operation @rpardini .
> 
> I already did. Assuming that "user" is the user that launched "compile.sh", I have `/home/<user>` owned by root. This was why I was getting insane because when I was booting that image, I was creating the same "user" at first run (it's a recurring name I'm using) and then when KDE was launched, it was failing. As soon as I was using any other name, it was working and it took me a while to make the connection 😄.

So what files are in your `/home/<user>` dir?

philippe44, 2023-10-26:
oh, sorry. There is /home/<user>/.cache/appstream. No file, just that tree

igorpecovnik, 2023-10-26:
Basically we need one clean command before closing the image and before calling user customization script?

Just to make sure nothing is there.

rpardini, 2023-10-26:
Hmm. Seems like in Amazingfate's case, it is the `add-apt-repository` that is causing the "leak". I'll try to confirm this.
Unsure about philippe44's case: what is "appstream"? I couldn't reproduce this at all.
Either way, what puzzles me is 

1) that the `rsync` that transfers from SDCARD to MOUNT is already (for many years) trying to ignore `/home` and somehow failing? Or maybe it's a ppa being added after MOUNT?

2) What is leaking that is making `add-apt-repository` do that? We probably should sanitize the environment, so `add-apt-repository` (and others / appstream) simply _can't_ get confused.

rpardini, 2023-11-06:
> 1. that the `rsync` that transfers from SDCARD to MOUNT is already (for many years) trying to ignore `/home` and somehow failing?

I was mistaken; only the `rootfs` tar-ing had the `/home` exclusion. rootfs-to-image's `rsync` didn't.

closed 2023-11-06 by igorpecovnik

rpardini, 2023-11-15:
Ooops, fix one, break another: https://github.com/armbian/build/issues/5918

PR #5886 (2023-11-06): rootfs-to-image: create_image_from_sdcard_rootfs(): ignore `/home` during rsync; fixes #5864
#### rootfs-to-image: create_image_from_sdcard_rootfs(): ignore `/home` during rsync; fixes #5864

- rootfs-to-image: create_image_from_sdcard_rootfs(): ignore `/home` during rsync; fixes #5864
commit 46728e94 (2023-11-06): rootfs-to-image: create_image_from_sdcard_rootfs(): ignore `/home` during rsync; fixes #5864