armbian/build #5870: Unable to build Armbian for Orange Pi 5

gjsmo, 2023-10-28:
### What happened?

During the build process, Armbian tries to clone the repository [morrownr/8821cu-20210118/](https://github.com/morrownr/8821cu-20210118/). This does not appear to exist anymore, causing the build to fail. It also seems to prevent the `compile.sh` from properly creating a logfile, or at least giving me the path to and/or uploading it with the `SHARE_LOG` option.

### How to reproduce?

`./compile.sh docker BOARD=orangepi5 BRANCH=legacy RELEASE=bookworm KERNEL_CONFIGURE=no BUILD_DESKTOP=no BUILD_MINIMAL=no INSTALL_HEADERS=yes EXTERNAL=yes EXTERNAL_NEW=compile PROGRESS_DISPLAY=dialog BOOTSIZE=256 SKIP_BOOTSPLASH=yes CONSOLE_AUTOLOGIN=no BUILD_ONLY=default SHARE_LOG=yes`

I have tried other combinations, including zero options whatsoever, with no effect.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-10-28:
You're using the wrong branch. Try `main`. This was solved months ago.

closed 2023-10-28 by rpardini

gjsmo, 2023-10-28:
@rpardini Ok I'll admit I'm actually baffled here how I got onto `master` instead of `main`. My bash history shows that I intentionally checked out `master`: `git clone --depth=1 --branch=master https://github.com/armbian/build`, and I'm sure I didn't type that because I wouldn't have bothered with the long form of the branch option. Is it possible there are some docs which have not been properly updated?