armbian/build #5918: /home directory unexpectedly cleared by build script

theCalcaholic, 2023-11-15:
### What happened?

Hi, I'm the maintainer of [NextcloudPi](https://github.com/nextcloud/nextcloudpi) and use armbian to generate various system images for SBCs. Recently, the build script was changed to clear the home directory before finalizing the image, which breaks my build process, as I rely on a preconfigured home directory for the `www-data` user.

In particular, commit 46728e9 introduces the change which breaks my builds.

If I understand the situation correctly and am not missing anything, I think, this should not be the default behavior or at least there should be a configuration option to turn this off. I'm looking forward to hear your opinions or solutions, if I overlooked any :)

### How to reproduce?

Here's my build script: https://github.com/nextcloud/nextcloudpi/blob/master/build/build-SD-armbian.sh
And here's the customize-image.sh: https://github.com/nextcloud/nextcloudpi/blob/master/build/armbian/armbian.sh

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-11-15:
Hey @theCalcaholic -- did you see https://github.com/armbian/build/issues/5864 ? That's the reason https://github.com/armbian/build/pull/5886 was done, to ignore and any `/home` data that "leaked" during the build process. Unfortunately it seems you're _intentionally_ leaking into `/home`.

I fully understand your position though and I'm sorry it broke your use case. It might very well break similar others, although pre-deploying stuff to `/home` does feel a bit strange. Feel free to send a revert (or similar) PR and tag the ppl involved in the ones above so there's some discussion.

viraniac, 2023-11-21:
@theCalcaholic you can now set INCLUDE_HOME_DIR=yes, to keep contents of /home/ directory intact inside the image. I hope this will help to get your usecase working again. Let us know if this solves the issue.

theCalcaholic, 2023-11-21:
@rpardini @viraniac Thanks so much! I was planning on doing a writeup on possible solutions or even provide a PR to make it configurable, but you were faster than I :)

This flag should solve my issue.

closed 2023-11-21 by theCalcaholic