armbian/build #5948: rock64 compile error

liwei19920307, 2023-11-21:
### What happened?

Unable to compile due to some patches

### How to reproduce?

./compile.sh BOARD=rock64 BRANCH=current BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=jammy

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/zucomozapo.rust

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Tonymac32, 2023-11-21:
@SteeManMI if a patch included in the build system is failing (it is) and breaking the compilation of a family of boards then I would consider this appropriate.

igorpecovnik, 2023-11-21:
Yeah, we have to sort this out.

closed 2023-11-21 by igorpecovnik

rpardini, 2023-11-21:
[CI knew way before any of us](https://github.com/armbian/os/actions/runs/6931128755/job/18857348102) -- where just need to make CI let us know somehow. Maybe some GHA-to-Issues or GHA-to-Discord thing?

igorpecovnik, 2023-11-21:
I am getting emails on failure but ... there is simply too much noise and even I noticed this before issue was opened, I couldn't find time it out faster as I did. I posted to Discord in case someone else would have time. Not sure if automation can help here since our capacity & organisation is limited.

liwei19920307, 2023-11-22:
This problem was fixed promptly. It bothered me for several days. Thank you for your efficient work.