armbian/build #5954: armbian latest offical for radxa zero， the /boot/dtb/amlogic/overlay/meson-g12a-radxa-zero-spi-spidev.dtbo can't work at all !!!

Logan147, 2023-11-22:
### What happened?

appear /dev/spi* device

### How to reproduce?

just download the latest armbian for radxa zero:https://www.armbian.com/radxa-zero/
iso version ： Armbian Jammy | Minimal CLI
and then copy this https://github.com/armbian/build/blob/23b94a05aa52f9ada4ec817df1d6f1652f23c7df/patch/kernel/archive/meson64-6.6/overlay/meson-g12a-radxa-zero-spi-spidev.dts
to your /boot/dtb/amlogic/overlay/ directory and use dtc to compile the dts to dtbo 
command :
dtc -I dts -O dtb -o meson-g12a-radxa-zero-spi-spidev.dtbo meson-g12a-radxa-zero-spi-spidev.dts

next step:
use vim to add these lines to /boot/armbianEnv.txt and reboot ：
overlays=g12a-radxa-zero-spi-spidev
param_spidev_spi_bus=1
param_spidev_max_freq=10000000

then you will found  ls /dev/spi* output is nothing !!!!!!!
fuc...k
[image]

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Logan147, 2023-11-22:
If anyone can help I would be grateful

viraniac, 2023-11-22:
@Logan147 could you please share the serial console logs and logs from armbianmonitor -m?

Without that info, Anyone that does not possess that device will not be able to debug this and help you with the same. Also I would suggest you move this discussion to forums as this might not qualify as an issue in the build framework

igorpecovnik, 2023-11-22:
> If anyone can help I would be grateful

Try Armbian community forums, contact Radxa technical support. This is build framework tracker. We have nothing to do with this.

closed 2023-11-22 by igorpecovnik