armbian/build #5962: Armbian boot partition unreadable on Windows machines due to weird partition type

meganukebmp, 2023-11-25:
### What happened?

Armbian build sets the boot partition as type 0xEA. This partition type is not standard in any specific distribution or operating system. As such Windows is unable to discover and mount the partition which makes it impossible to mount and modify the network config or other first boot configurations on Windows machines. After changing the partition type to either 0x0B or 0x0C, for DOS FAT32, the partition mounts correctly and is editable fine.

I haven't observed any side effects of doing so, u-boot seems to correctly discover and use these partitions regardless.

I tried tracking down this change and why it was set to 0xEA but found no definitive answer, it might very well be intentional and in that case, I'd like to know as to why.

### How to reproduce?

N/A

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

meganukebmp, 2023-11-25:
Relevant function: https://github.com/armbian/build/blob/main/lib/functions/image/partitioning.sh#L188

meganukebmp, 2023-11-25:
Some relevant links to this discussion would be:

https://en.wikipedia.org/wiki/Partition_type - which lists the MBR partition types that are IBM PC compatible (any relevant OS will likely support identifying ones in this list)

https://uapi-group.org/specifications/specs/boot_loader_specification/ - The Linux Userspace API docs (from freedesktop and systemd devs) Suggest the use of 0xEA, however I am yet to actually see this utilized in any distribution by default.

https://github.com/pbatard/rufus/issues/609 - Popular image flashing tool Rufus, uses 0xEA as extra metadata partition for itself. The developers claim they've used it for longer than the UAPI spec. Users might be using this tool to image their SDcards which might lead to potential conflicts.

SteeManMI, 2024-07-08:
The aml-s9xx-box builds have a fat boot partition (so users can edit the extlinux.conf file for their specific type of TV Box).  I've see reports of issues in the forums (since the move to armbian-next where I think this change was introduced) that users can no longer see the fat partition on Windows machines (often TV Box users don't have other linux machines to use to pop their sd card into to edit).
This is something I've wanted to followup on, but haven't had the time.

SteeManMI, 2024-07-08:
I don't know.  For my use case, as long as the SD card boot partition is readable by modern Windows OSs and MacOS that is my particular issue.  I can't speak for the original author of this particular issue as to what his use case was.

SteeManMI, 2024-07-08:
@ColorfulRhino   I think you are really discussing a separate issue (MBR vs GPT).  This issue could probably be solved by changing the partition type from 0xEA to either 0x0B or 0x0C.  The change to 0xEA occurred in the move to the new framework.  I don't know if @rpardini had a particular reason for the change or not.

rpardini, 2024-07-08:
On Amlogic, we can't have GPT _and_ u-boot on the same media, as the load offsets conflict.
It is possible to use for example u-boot in SPI and a GPT SD, but not _on the same media_

rpardini, 2024-07-08:
> Relevant function: [`main`/lib/functions/image/partitioning.sh#L188](https://github.com/armbian/build/blob/main/lib/functions/image/partitioning.sh?rgh-link-date=2023-11-25T01%3A03%3A10Z#L188)

Nah. That is for _legacy BIOS_ -- a x86 thing controlled by `UEFISIZE > 0 and BIOSSIZE > 0`

rpardini, 2024-07-08:
> Armbian build sets the boot partition as type 0xEA.

That's done here: https://github.com/armbian/build/blob/main/lib/functions/image/partitioning.sh#L200-L210

I can't remember writing this code. `git blame` says it's mine. intriguing

rpardini, 2024-07-08:
I _didn't_ write this code, but we've split it around during armbian-next. This used to be on the old `debootstrap.sh`, heh.

See https://github.com/armbian/build/pull/4189 -- thus @hzyitc

rpardini, 2024-07-08:
> The change to 0xEA occurred in the move to the new framework. I don't know if @rpardini had a particular reason for the change or not.

See above. I'm fine with changing it. I personally don't use /boot partition as much as possible, as their presence is generally indicative of legacy bootloaders.

rpardini, 2024-07-08:
> But from this, I assume MBR is needed for Amlogic? What a shame :( Whyyyy Amlogic? I am disappointed yet again by vendors doing weird stuff.

This has a long history. Even MBR was a challenge. https://github.com/u-boot/u-boot/blob/master/doc/board/amlogic/boot-flow.rst & https://github.com/LibreELEC/amlogic-boot-fip/pull/8

I've no idea if Amlogic changed this for their new (S4/T7+) SoCs though.

rpardini, 2024-07-08:
> Which is why I was thinking, if there is no real reson to support all this legacy stuff (MBR as well as legacy boot), we could simplify the code massively :) But I have no idea if this stuff is maybe needed for something? This is why I was asking instead of just rushing to remove more stuff again xD

That is all absolutely needed, either for x86 legacy BIOS (part of `uefi-x86`) or for non-GPT-capable boards.
Here we only need to fix the type as @meganukebmp indicated.

hzyitc, 2024-07-09:
This is why: https://github.com/util-linux/util-linux/commit/d0c430068206e1215222792e3aa10689f8c632a6

BTW, IIRC, Windows doesn't mount boot partition by default. As you can see that the EFI boot partition doesn't be shown in explorer. It maybe a safety feature.

rpardini, 2024-07-09:
> How does `uefi-x86` use legacy BIOS though? isn't legacy BIOS the opposite of UEFI?

It's... mashed together. Some machines, even though they're in theory UEFI, use a CSM -- the CSM will end up booting a legacy BIOS thing. Other machines (eg virtual machines), without OVMF, will also default to BIOS booting.

SteeManMI, 2024-07-09:
> [](https://uapi-group.org/specifications/specs/boot_loader_specification/)
> 
> > * On disks with an MBR partition table:
> >   
> >   * The boot partition — a partition with the type ID of 0xEA — shall be used as the single location for boot loader menu entries.

Getting back to the original issue that was logged.  The above behavior (which isn't really intended for classic u-boot senarios from my understanding) breaks existing functionality.  It is desired and has long been the case that you can take an SD card with an Armbian image (for those builds that still use a FAT partition for /boot) and pop it into a windows machine to tweek things (armbianEnv being one and extlinux.conf being another).  This change breaks that long standing capability.  So until we move to modern booting systems, we really need to go back to the /boot partition being 0x02 for MBR cases when the board configures the boot partition as a fat file system type.  (I'm not sure what should be done for GPT cases as my boards are MBR).

rpardini, 2024-07-12:
We just need to change the (MBR-only) type back and be done with it for a while.

Further changes should be carefully considered -- we've already lost capabilities that were result of hard labor and were removed in the interest of "keeping it simple". It's a thin line, but booting on exotic hardware is one of Armbian's strengths -- let's not throw that away.

rpardini, 2026-08-15:
Well, that's a different argument ("... is arbitrary" vs "..is actively breaking functionality with another OS").

You could bring it up with UAPI group, I've no idea on how though.

meganukebmp, 2026-08-15:
> Well, that's a different argument ("... is arbitrary" vs "..is actively breaking functionality with another OS").
> 
> You could bring it up with UAPI group, I've no idea on how though.

Fair enough. I'd like for this issue to remain open while I discuss this with them. Perhaps this would be considered a special case and they wouldn't budge on it. In which case it's up to Armbian maintainers to decide whether or not to go against the spec on this matter.