armbian/build #5973: Automatically generated kernel-patch fails to apply

rossbcan, 2023-11-26:
### What happened?

Expected automatically generated patch to apply properly

### How to reproduce?

/compile.sh BOARD=khadas-vim2 BRANCH=edge kernel-patch

edited "cache/sources/linux-kernel-worktree/6.6__meson64__arm64/arch/arm64/boot/dts/amlogic/Makefile"

resulting in this patch:
` arch/arm64/boot/dts/amlogic/Makefile | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/arch/arm64/boot/dts/amlogic/Makefile b/arch/arm64/boot/dts/amlogic/Makefile
index 3d0099dc2..4a163dfce 100644
--- a/arch/arm64/boot/dts/amlogic/Makefile
+++ b/arch/arm64/boot/dts/amlogic/Makefile
@@ -66,10 +66,12 @@ dtb-$(CONFIG_ARCH_MESON) += meson-gxm-nexbox-a1.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-q200.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-q201.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-rbox-pro.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-s912-libretech-pc.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-t95z-plus.dtb
+dtb-$(CONFIG_ARCH_MESON) += meson-gxm-tx9-pro.dtb
+dtb-$(CONFIG_ARCH_MESON) += meson-gxm-tx9-pro-100M.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-vega-s96.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-gxm-wetek-core2.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-s4-s805x2-aq222.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-a95xf3-air.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-a95xf3-air-gbit.dtb
@@ -82,6 +84,6 @@ dtb-$(CONFIG_ARCH_MESON) += meson-sm1-odroid-c4.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-odroid-hc4.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-sei610.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-x96-air.dtb
 dtb-$(CONFIG_ARCH_MESON) += meson-sm1-x96-air-gbit.dtb
 
-subdir-y       := $(dts-dirs) overlay
\ No newline at end of file
+subdir-y       := $(dts-dirs) overlay
-- 
Created with Armbian build tools https://github.com/armbian/build
`

moved patch to "userpatches/kernel/meson64-edge/kernel-meson64-edge.patch"

compiled:
"./compile.sh build BOARD=khadas-vim2 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=no \
                KERNEL_CONFIGURE=yes RELEASE=jammy KERNEL_GIT=shallow"

which failed with:
[image]

and the file "/tmp/tmp26wqmycp" does not exist.

It is a sample change, just added two lines:
+dtb-$(CONFIG_ARCH_MESON) += meson-gxm-tx9-pro.dtb
+dtb-$(CONFIG_ARCH_MESON) += meson-gxm-tx9-pro-100M.dtb

What's going on, how to fix?

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/osoxebalac

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2023-11-26:
I will say this is not a problem of build framework per se, and can easily be solved by the user on their side. When the user created the patch, all other patches were already applied. So the created patch was supposed to be the last patch in the list. But its clear from the logs, that the patch is now being applied at the 52nd position instead of the last position. Just adding a `zzzz-` as a prefix to patch filename would have resolved the issue.

rossbcan, 2023-11-26:
@viraniac 

disagree. Adding zzz- as patch name prefix has same problem. New log: https://paste.next.armbian.com/oyufowemih

The "problem" is that patches: Armbian Bare DT auto-patch, Armbian Bare OVERLAY auto-patch and Armbian DT Makefile auto-patch appear to be applied BEFORE the kernel-patch (auto create patch) process and AFTER the created / user patch is applied, causing patch failure.

This is because I am making changes to files affected by the above auto-patch's 

I am unsure how to resolve this without code consistency between when patch created and applied.

viraniac, 2023-11-26:
@rossbcan Sorry, my bad. I take care of sunxi and meson-s4t7 family and that feature is not used there. Hence completely forgot about the same. Reopened the issue. Maybe someone else can take a look

rossbcan, 2023-11-26:
Possible solution hint...

In file "lib/tools/common/dt_makefile_patcher.py", make auto-patch run dependent on "kernel-patch" NOT defined in compile.sh parameters.

may be "unintended" consequences. Need a guru...

rpardini, 2023-11-26:
The purpose of the auto patcher is exactly to avoid needing to patch Makefiles, which are frequent cause of conflicts across version bumps.
Just add the DTs you want to the `dt` dir in the patch folder, everything else will be done automatically. See the others already in there: https://github.com/armbian/build/tree/main/patch/kernel/archive/meson64-6.6:
- 000.patching_config.yaml configures the auto patcher
- `dt` folder, you can just drop your .dts here and they'll be added to makefile automatically

rpardini, 2023-11-26:
Oh, and you can have your bare DTs also in `userpatches` if you prefer, I have them here https://github.com/rpardini/armbian-build/tree/extensions/userpatches/kernel/meson64-edge/dt (thus in `userpatches/kernel/meson64-edge/dt`)

closed 2023-11-26 by rossbcan