armbian/build #6020: chroot: failed to run command ‘/bin/bash’: No such file or directory...

zerog2k, 2023-12-07:
### What happened?

I am encountering this `chroot: failed to run command ‘/bin/bash’: No such file or directory... `
error when trying to build.

FYI, I am building on ubuntu-18.04 LTS, however, I don't believe this is the problem, as the issue is occurring inside the docker build container, which should not care about the host os.

### How to reproduce?

`./compile.sh build BOARD=onecloud BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=yes RELEASE=jammy EXTRAWIFI=no`

I put a long sleep before `lib/functions/rootfs/distro-specific.sh:193`, then did a docker exec to get a shell into the container and poke around.

I was able to reproduce this
```
root@7727cfee645e:/armbian# chroot /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277 /bin/bash
chroot: failed to run command '/bin/bash': No such file or directory
```

the file is indeed there:
```
root@7727cfee645e:/armbian# ls -l /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/bin/bash
-rwxr-xr-x 1 root root 813652 Jan  6  2022 /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/bin/bash
root@7727cfee645e:/armbian# file /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/bin/bash
/armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/bin/bash: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=fdaf25ce6a523e143292cc57223b5fc1141f7e95, for GNU/Linux 3.2.0, stripped
```

but chroot's output is misleading - it's actually having some issue executing it, perhaps due to something w.r.t. qemu-user-static, which was installed:

```
root@7727cfee645e:/armbian# dpkg -l |grep qemu
ii  qemu-user-static                      1:6.2+dfsg-2ubuntu6.15                  amd64        QEMU user mode emulation binaries (static version)
ii  qemu-utils                            1:6.2+dfsg-2ubuntu6.15                  amd64        QEMU utilities
root@7727cfee645e:/armbian# qemu-arm-static
qemu: no user program specified
```
but something is not right with it in the build container.

By simply uninstalling and reinstalling qemu-user-static, i.e. `apt remove qemu-user-static && apt install qemu-user-static`, (the exact same pkg version is reinstalled) this does something to the build container which makes the chroot work.

```
root@7727cfee645e:/armbian# chroot /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/ /bin/sh
# quit
/bin/sh: 1: quit: not found
# exit
root@7727cfee645e:/armbian# chroot /armbian/.tmp/rootfs-f10848de-ba80-4825-a9e3-3c68b2f2a277/ /bin/bash
root@7727cfee645e:/# exit
exit
root@7727cfee645e:/armbian# dpkg -l | grep qemu-user-static
ii  qemu-user-static                      1:6.2+dfsg-2ubuntu6.15                  amd64        QEMU user mode emulation binaries (static version)
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log URL

https://gist.github.com/zerog2k/e5987095fd0a67d648d402f3f7e7bbf3#file-gistfile1-txt-L11480-L11494

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

zerog2k, 2023-12-07:
I want to dig into the Dockerfile and container build process a bit more, but this appears to be a bit more dynamically built.

```
CONTAINER ID                                                       IMAGE                                                                     COMMAND                                                                                                                                                                                                                                                                                        CREATED          STATUS                PORTS                                                                                                                                                                     NAMES
7727cfee645eab8fdcbe0dc4eadf37d29ea30bf55da3b7f16784b5816f60b3c9   armbian.local.only/armbian-build:initial                                  "/bin/bash /armbian/compile.sh PREFER_DOCKER=no BUILD_MINIMAL=no BUILD_DESKTOP=no ARMBIAN_BUILD_UUID=f10848de-ba80-4825-a9e3-3c68b2f2a277 ARMBIAN_RELAUNCHED=yes BRANCH=edge EXTRAWIFI=no BOARD=onecloud SKIP_LOG_ARCHIVE=yes SET_OWNER_TO_UID=1000 RELEASE=jammy KERNEL_CONFIGURE=no build"   25 minutes ago   Up 25 minutes                                                                                                                                                                                   sweet_aryabhata
```

viraniac, 2023-12-08:
Try installing binfmt-support on your host OS (ubuntu-18.04). That will add a service which will update binary format information. I think the issue is caused as nothing is running `/usr/sbin/update-binfmts --enable` on your system and hence even though docker container has binfmt-support and qemu-user-static installed, its not able to run the binary. When you reinstalled qemu-user-static, the post install procedure of the deb file would have called the same, making it look like that the issue is resolved. If my understanding is correct, the issue might re-appear if you restart your system

zerog2k, 2023-12-08:
thanks @viraniac but looking at my host os I see this:
```
jens@mars:~$ dpkg -i | grep binfmts
dpkg: error: requested operation requires superuser privilege
jens@mars:~$ dpkg -l | grep binfmt
ii  binfmt-support                         2.1.8-2                                              amd64        Support for extra binary formats
ii  qemu-user-binfmt                       1:2.11+dfsg-1ubuntu7.42+esm1                         amd64        QEMU user mode binfmt registration for qemu-user
jens@mars:~$ /usr/sbin/update-binfmts --display qemu-arm
qemu-arm (enabled):
     package = qemu-user-binfmt
        type = magic
      offset = 0
       magic = \x7f\x45\x4c\x46\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00
        mask = \xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff
 interpreter = /usr/bin/qemu-arm
    detector = 
```

I'll try to remove the armbian build container and reproduce the issue...

zerog2k, 2023-12-09:
So I had to do a different arm64 build, and I run into the same thing:
 https://paste.next.armbian.com/ebediyagof

If i run `compile.sh docker-shell`, then I can see with `/usr/sbin/update-binfmts --display` that all of them are disabled. Back in the host os, `/usr/sbin/update-binfmts --display` shows all are enabled.

so after a bit of reading on qemu & binfmt, i found this _extremely_ helpful article:
https://medium.com/@artur.klauser/building-multi-architecture-docker-images-with-buildx-27d80f7e2408

it has two scripts which were helpful, as ubuntu 18.04 seems to suffer from the 
`ERROR: QEMU not registered in binfmt_misc with fix-binary (F) flag.` issue

```
jens@mars:~$ bin/check-qemu-binfmt.sh 
OK: docker 20.10.21 supports buildx experimental feature.
OK: kernel 4.15.0-219-generic has binfmt_misc fix-binary (F) support.
OK: /proc/sys/fs/binfmt_misc is mounted
OK: update-binfmts 2.1.8 has fix-binary (F) support.
ERROR: QEMU not registered in binfmt_misc with fix-binary (F) flag.

jens@mars:~$ sudo bin/reregister-qemu-binfmt.sh 
Reregistering arch aarch64
Reregistering arch alpha
Reregistering arch arm
Reregistering arch armeb
Reregistering arch cris
Reregistering arch m68k
Reregistering arch microblaze
Reregistering arch mips
Reregistering arch mips64
Reregistering arch mips64el
Reregistering arch mipsel
Reregistering arch ppc
Reregistering arch ppc64
Reregistering arch ppc64abi32
Reregistering arch ppc64le
head: cannot open '/var/lib/binfmts/qemu-riscv64' for reading: No such file or directory

jens@mars:~$ bin/check-qemu-binfmt.sh 
OK: docker 20.10.21 supports buildx experimental feature.
OK: kernel 4.15.0-219-generic has binfmt_misc fix-binary (F) support.
OK: /proc/sys/fs/binfmt_misc is mounted
OK: update-binfmts 2.1.8 has fix-binary (F) support.
OK: QEMU registered in binfmt_misc with flags OCF (F is required).
Host looks good for docker buildx multi-architecture support.
```
armbian compile _seems_ to be running now, although I'm getting alot of `qemu: Unsupported syscall: 293` types of errors.

will do some more testing

zerog2k, 2023-12-09:
ubuntu-18.04 is EOL now anyhow, but unfortunately my ancient build box still has to run this (with 4.x kernel) due to ancient nvidia card and driver support which doesnt work with newer kernels/xorg in ubuntu-20+ :( 
so I'm going to chalk this up to, an ubuntu-18 issue which, is a wontfix

closed 2023-12-09 by zerog2k

zerog2k, 2023-12-12:
just dropping a note of closure here, just in case anyone else is trying to figure this out

looks like I _also_ needed to uncomment the 
```
DOCKER_FLAGS+=(--privileged)
``` 
in userpatches/config-docker.conf, and this took care of the qemu unsupported syscalls, and everythign is building fine

SeeleVolleri, 2024-05-12:
> just dropping a note of closure here, just in case anyone else is trying to figure this out
> 
> looks like I _also_ needed to uncomment the
> 
> ```
> DOCKER_FLAGS+=(--privileged)
> ```
> 
> in userpatches/config-docker.conf, and this took care of the qemu unsupported syscalls, and everythign is building fine

Hi! I have the same problem, where can I find the `config-docker.conf`? I searched the repo, but only seems exist in 2021 but not mainline...