armbian/build #6029: khadas-vim2 (s912) wont boot after current kernel updated to 6.1.66, hangs at mounting /boot

rossbcan, 2023-12-10:
### What happened?

Successful boot, like previous 6.1.{64,65} kernels

last thing I see on console (during first and subsequent boots) is:
"systemd-fsck@dev-disk-by\x2duuid-aA58\x2dD037.service (UUID is correct)
sometimes this is followed by "mounting /boot"
and perpetual hang

NOTE: I modified the khadas-vim2 config for FAT bootfs and size.

None of my build settinges or config changed between functioning with previous and latest kernels.

### How to reproduce?

./compile.sh build BOARD=khadas-vim2 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no RELEASE=jammy KERNEL_GIT=shallow

Build log: https://paste.next.armbian.com/ibulawabon

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/ibulawabon

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2023-12-10:
From the logs, it looks like you have made a lot of changes. Without looking at them, its hard to tell what could have caused the issue. I would suggest you to add verbosity=7 and extraargs=debug to your boot/armbianEnv.txt file to get more detailed output of where it hangs. 

PS: I don't have the board so I can't do any testing on my end

igorpecovnik, 2023-12-10:
Thankfully we have automated test system which proves there is nothing wrong at our side:
https://github.com/armbian/os?tab=readme-ov-file#khadas-vim2

igorpecovnik, 2023-12-10:
If this is Khadas hardware and you have been changing to FAT boot and change other things, verify if boot scripts are compatible with the change. They should be, but who knows. Are you sure you are using boot loader from SD card and not the one for internal media? But problems could also be hidden deep in the boot loader code (especially if this is some random act) which automatically means several hours of debugging. That is something you can't even ask for. From Armbian perspective, building and running of this hardware works perfectly. IMO best way is to stick to defaults. Closing.

closed 2023-12-10 by igorpecovnik

rossbcan, 2023-12-10:
after verbosity=7 and extraargs=debug to boot/armbianEnv.txt ...

[ TIME ] Timed out waiting for device /dev/disk/by-uuid/1A58-D037
[DEPEND] Dependency failed for /boot
[DEPEND] Dependency failed for Local File Systems
[DEPEND] Dependency failed for File System Check on /dev/disk/by-uuid/1A58-D037

boot (with verb, debug changes) proceeded to emergency mode, "journalctl -xb" hangs, ^C, back to shell prompt

fdisk -l /dev/sda - partitions, UUIDS OK, rootfs mounted
"mount /dev/sda1 /boot" hangs

dmesg contains: FAT-fs (sda1) IO charset iso-8859-1 not found
Fails too early in boot process for syslog existence.

added: iso_8859-1 to /etc/modules, rebooted, failed same symptoms
oddity (after fresh reboot to emergency mode):
"mount /dev/sda1 /boot" hangs (nothing in dmesg)
"mount -a" mounts /dev/sda1 to /boot properly

I am confused why iso_8859-1.ko is not loaded on demand and must be specified in /etc/modules

I am considering (kconfig) making iso-8859-1 builtin, to eliminate parsing /etc/modules boot timing issues. Good idea? It may solve issue, but, I believe this should work with standard build.

As to my customizations, Unrelated to this issue: patches to add openvfd and a few dtb's for the hardware I am using (Tanix TX9 Pro, TX92, both S912). Unrelated, working on Rockchip H96 Max V56 (working) and V58 (not started yet, just received hardware)  

Any other debug suggestions?
Thanks; Bill

igorpecovnik, 2023-12-10:
> Any other debug suggestions?

https://forum.armbian.com/ not here. This is build framework ticketing system and this issue is unrelated. Please understand ans respect our attention.