armbian/build #6036: ERROR in function prepare_host

abirabedinkhan, 2023-12-12:
### What happened?

```
[ o.k. ] Preparing [ host ]  
[ o.k. ] Build host OS release [ jammy 
[ error ] ERROR in function prepare_host [ main.sh:589 -> main.sh:434 -> general.sh:1408 -> general.sh:0
[ error ] It seems you ignore documentation and run an unsupported build system: jammy ming                                                          
[ o.k. ] Process terminated
```

### How to reproduce?

I ran 
```
./compile.sh EXPERT="yes"
```
in WSL2

I was using this branch
```
https://github.com/awawa-dev/build/tree/v2022.03.08
```

### Branch

other

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2023-12-12:
These is an old fork from when ubuntu jammy is not release.
https://github.com/awawa-dev/build/blob/v2022.03.08/lib/general.sh#L1398

abirabedinkhan, 2023-12-12:
then which version of Ubuntu should I use for the compilation

abirabedinkhan, 2023-12-12:
how can I downgrade my WSL to 21.04

igorpecovnik, 2023-12-12:
Check this if it helps
https://github.com/armbian/build/commit/204d2e79b9a337722af292a06a309c49a51bee6e

abirabedinkhan, 2023-12-12:
UPDATE:
using this guide: https://www.golinuxcloud.com/downgrade-ubuntu/
this source list: https://gist.github.com/airglow923/71aa28e4def1120354c5a9f2295de532#file-sources-list-hirsute
```
sudo apt-get update --allow-insecure-repositories
```
trying to downgrade.

igorpecovnik, 2023-12-12:
Before you downgrade, try to install our kernel - is in beta.armbian.com repository.

abirabedinkhan, 2023-12-12:
Does it support `Tanix TX6 Allwinner H6`?

igorpecovnik, 2023-12-12:
> Does it support Tanix TX6 Allwinner H6?

Who? Kernel for WSL2 has nothing to do with what you will build. It might fix problems you have running build framework.

abirabedinkhan, 2023-12-12:
ok let me try

abirabedinkhan, 2023-12-15:
After switching OS it shows this ->
[image]

igorpecovnik, 2023-12-15:
> it shows this

This tells nothing. You need to provide logs 

`./compile.sh SHARE_LOG=yes`

and keep in mind that support is limited to build framework functioning. If there is a problem in source code, you need to resolve it on your own.

viraniac, 2023-12-15:
You asked for Tanix TX6. I wonder why you are not building from our main branch using BOARD=tanix-tx6 and instead are relying on a fork?

igorpecovnik, 2023-12-15:
I didn't even notice it was a fork. Then we can only close this down.

closed 2023-12-15 by igorpecovnik

abirabedinkhan, 2023-12-15:
> You asked for Tanix TX6. I wonder why you are not building from our main branch using BOARD=tanix-tx6 and instead are relying on a fork?

Cuz in the community edition wifi and ethernet don't work. But from the release of the fork, works perfectly fine. But I need a different version of the OS. I want to compile focal.

igorpecovnik, 2023-12-15:
I am sorry, please respect conditions under which we are willing to sponsor your hobby / business needs.