armbian/build #6054: Add ASUA Tinker Edge R board.

ARC-MX, 2023-12-16:
### Which feature would you like to have?

ASUS Tinker Edge R is an Single Board Computer (SBC) specially designed for AI applications.  But the official firmware version is too old, request to add a new board. ASUS official u-boot source code and kernel source code are
[1](https://github.com/TinkerEdgeR/debian-u-boot)
[2](https://github.com/TinkerEdgeR/debian-kernel),
which have a complete device tree description file and board configuration file. However, my ability is limited and I cannot complete the adaptation, so I request official support

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2023-12-16:
WIthout funding or someone that is willing to do it for free (we won't) it won't happen.

Tonymac32, 2023-12-16:
Besides the unfortunate situation with Asus and their community support problems, there are no plans for me to support a vendor kernel for the RK3399.  The edge R doesn't have a mainline device tree because of some component sourcing decisions by Asus, the voltage regulators have no driver outside the Asus tree.  

Let me put it this way:  I have had this board for a couple years and I have not felt like dealing with the effort because Asus isn't serious about hobbyists or small users and won't invest in the mainline support.

viraniac, 2024-01-03:
As stated above, we don't have this planned to be supported. But if a PR is raised for the same, we will add it to build framework

closed 2024-01-03 by viraniac