armbian/build #6094: Linux headers package is missing after compilation - kernel branch 4.19

xnosek00, 2023-12-26:
### What happened?

Hello,

I compiled the kernel for the **bananapir2** device on **Debian Bullseye**, which uses the kernel from the **4.19 branch**.

Unfortunately, after compilation, the **linux headers package is not created**. Just a year ago, the compilation was fine. At that time, I compiled the 4.19.265 kernel in the Armbian branch 22.11.

I want to ask whether this is a minor issue that is easily fixable, or if it's a major problem that makes implementing the linux-headers package impossible.

Unfortunately, I don't have enough kernel expertise to assess this or to build the package myself.

The kernel compilation went fine; the issue I see now is only with the missing linux-headers package.

I tried compile:

`git clone --depth=1 --branch=main https://github.com/armbian/build  `

then I tried this branch: 

`git clone --depth=1 --branch=v23.11 https://github.com/armbian/build`

And I have the same result:

```
user@armbian:~/build/output/debs$ ls -l
total 137632
-rw-rw-r-- 1 user root  1566912 Dec 25 22:00 armbian-bsp-cli-bananapir2-legacy_24.2.0-trunk_armhf__1-PCf77f-V43d0-H01ba-Be8be-R6632.deb
-rw-rw-r-- 1 user root    20672 Dec 25 22:00 armbian-bsp-cli-bananapir2_24.2.0-trunk_armhf__1-PCf77f-V43d0-H01ba-Be8be-R6632.deb
-rw-rw-r-- 1 user root   135092 Dec 25 22:00 armbian-config_24.2.0-trunk_all__1-SA6761-B0293-R448a.deb
-rw-rw-r-- 1 user root 74882604 Dec 25 22:00 armbian-firmware_24.2.0-trunk_all__1-SAe36a-B90f5-R448a.deb
-rw-rw-r-- 1 user root 11815352 Dec 25 22:00 armbian-zsh_24.2.0-trunk_all__1-SAbfee-Baafd-R448a.deb
drwxrwxr-x 2 user root     4096 Dec 25 22:00 bullseye
-rw-rw-r-- 1 user root   174272 Dec 25 21:59 linux-dtb-legacy-mt7623_24.2.0-trunk_armhf__4.19.303-S74ad-D0000-Pde20-C999999Hfe66-HK01ba-V014b-B1743-R448a.deb
-rw-rw-r-- 1 user root 52009152 Dec 25 21:59 linux-image-legacy-mt7623_24.2.0-trunk_armhf__4.19.303-S74ad-D0000-Pde20-C999999Hfe66-HK01ba-V014b-B1743-R448a.deb
-rw-rw-r-- 1 user root   308616 Dec 25 21:38 linux-u-boot-bananapir2-legacy_24.2.0-trunk_armhf__2023.04-Sfd4e-Pf377-H1a12-Va21c-B1179-R448a.deb
```

I tried to compile another device - like cubieboard2. There is another kernel branch and the linux headers are included after compilation. So I think, it's only problem for bananapir2 or kernel branch 4.19.

### How to reproduce?

./compile

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xnosek00, 2023-12-26:
I edited step - how to reproduce the problem. This could be more helping:

`./compile.sh build BOARD=bananapir2 BRANCH=legacy BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=bullseye`

rpardini, 2023-12-26:
We've no support for producing _working_ kernel-headers when cross-compiling 4.x kernels.
You're welcome to dig in and try, but I did years ago and gave up; it is doable, but the sheer amount of hacks required is simply not worth it.

xnosek00, 2023-12-26:
Thank you @rpardini for you answer and explanation. I was afraid that it would be a major problem.

I know that the Armbian branch 21.11 worked for me with the 4.19.265 kernel. That's probably the kernel version where you put in the effort to fine-tune the compilation of linux headers. Is there a way I can compile that specific 4.19.265 kernel version?

I looked in the compilation parameters, but I couldn't find any parameter for selecting the kernel version. So, I just want to ask if it's possible to choose a specific kernel version during compilation. It could be solution for now.

rpardini, 2023-12-26:
Depends on what you call "worked". It might have given you a headers package, but could it really be used to build stuff?
If yes, you can go archeology the old (not `main`, can't remember what is is now) branch and try to reproduce. I _absolutely_ can't help you there though.

EvilOlaf, 2023-12-26:
>  (not main, can't remember what is is now) 

`master` :-)

xnosek00, 2023-12-27:
thank you for you answer, I'm closing this thread, because it is not a bug. The result is: obsolete kernel and device, support is not possible.

closed 2023-12-27 by xnosek00

xnosek00, 2023-12-31:
Finally, I immersed myself for several hours into the Armbian script and found a solution for one variant that I asked about in this thread. It's not an ideal solution, but it can be useful in some situations and might save someone a lot of time. Therefore, I will describe it.

# How to compile a specific kernel version in Armbian build

To obtain Linux headers for the kernel, I needed an older branch of Armbian and a specific kernel version. Armbian scripts always download the last stable kernel.

Downloading an older branch of Armbian is simple and might look something like this:

`git clone --depth=1 --branch=v22.11 https://github.com/armbian/build`

The downloaded scripts then need to be slightly edited. The configuration scripts are in the lib folder. I searched through all the scripts and found the function fetch_from_repo(). This ensures the downloading of source codes from Github. However, this function can be used to download data other than the Linux kernel. Editing this function is therefore not a solution, but a clue.

The third parameter of the function was a clue, which is stored in the variable $ref. Upon closer examination, I found out that if the parameter ref is passed with the variable **branch:linux-4.19.y**, the most current stable version of the kernel is downloaded. If the variable is passed with the value **tag:v4.19.292**, a specific version of the kernel is downloaded.

In the next step, I read in the file **build/config/boards/bananapir2.csc** that I have **BOARDFAMILY='mt7623'**.

All these clues led to this modification:

I edited the file **config/sources/families/mt7623.conf**. At the beginning of the file, I commented out the original KERNELBRANCH and set it to the kernel version I needed. Here is the actual code part:

```
case $BRANCH in

        legacy)

                #KERNELBRANCH='branch:linux-4.19.y'
                KERNELBRANCH='tag:v4.19.262'

                ;;

esac
```

Hopefully, this guide will help someone. There is no need to do anything. I was just documenting the instructions for one of the solutions to the problem I was asking about.

xnosek00, 2024-01-01:
Further investigation allowed me to compile the current stable kernel (branch 4.19.) -> 4.19.303 including linux-headers.

I used the armbian branch v23.02:

`git clone --depth=1 --branch=v23.02 https://github.com/armbian/build`

I had to modify the file **config/sources/families/mt7623.conf** because there is a kernel from a different branch specified. I think there's an error because two variables are set one after the other, so one overrides the other.

So, I commented out the line

`#KERNELBRANCH='commit:53b696f0584acce2e90db69272a2a11aab138370"`

Furthermore, in the file **lib/functions/compilation/patch/drivers_network.sh**, I also commented out the line in the function **patch_drivers_network()**:

`#driver_rtl8811CU_rtl8821C`

because the repository with the drivers no longer exists, and during the kernel compilation in menuconfig, I excluded some RTL wifi drivers and the kernel compiled successfully including the kernel-headers package.

The compilation and package creation went smoothly, so hopefully, it will work for me. Maybe it will help someone else.