armbian/build #6110: Add support for the Orange pi 2w 4gb models

renewedbullet, 2023-12-30:
### Which feature would you like to have?

Addition to the build config would be awesome!

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

viraniac, 2023-12-30:
@renewedbullet Feel free to do that yourself and raise a PR. If not, then cover the expenses for the same. We can't take demands when total donation from all the users only amounts to 60 USD per month

closed 2023-12-30 by viraniac

renewedbullet, 2023-12-30:
> @renewedbullet Feel free to do that yourself and raise a PR. If not, then cover the expenses for the same. We can't take demands when total donation from all the users only amounts to 60 USD per month

any guides how to do that?

EvilOlaf, 2023-12-30:
https://docs.armbian.com/Developer-Guide_Adding-Board-Family/