armbian/build #6132: Error Partition(s) 1, 2, 3 (...) 255, 256 on /dev/loop0 have been written, but we have been unable to inform the kernel...

Ekmansoft, 2024-01-06:
### What happened?

I'm trying to build a complete Armbian image for the Rock 5A board, but when the image is being written, it fails with this error message 

 [💲|🔨]   The partition table has been altered.
[💲|🔨]   Syncing disks.
[💲|🌿] Allocated loop device [ LOOP=/dev/loop0 ]
[💲|🔨]   Error: Partition(s) 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256 on /dev/loop0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
[💲|💥] Error 1 occurred in main shell [ at /home/kenneth/armbian/build/lib/functions/logging/runners.sh:211    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        run_host_command_logged() --> lib/functions/logging/runners.sh:193
             prepare_partitions() --> lib/functions/image/partitioning.sh:229
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:86
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💲|💥] Cleaning up [ please wait for cleanups to finish ]
[💲|🌿] Unmounting recursively [ SDCARD - be patient ]

I realize that this might well be classified as more of a behavior in some other part of Linux than a bug in the Armbian build framework, but since it can often occur when building Armbian, we should either fix the it or at the very least link to a page that describes how to solve it. 

### How to reproduce?

./compile.sh build BOARD=rock-5b BRANCH=edge BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='3dsupport browsers' DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=no RELEASE=bookworm

#uname -a 
Linux Maximus 5.15.133.1-microsoft-standard-WSL2 #1 SMP Thu Oct 5 21:02:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

I'm currently building from Ubuntu run under Windows Subsystem for Linux, but the problem has also been seen in native Ubuntu.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log URL

 https://paste.armbian.com/rabololire

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-01-06:
We are aware there might be bugs introduced in WSL2 kernels, which are currently out of our control. Those hints / files could help you resolve this, but its untested / WIP / not supported yet:
https://fi.mirror.armbian.de/beta/pool/main/l/linux-6.1.71/linux-image-current-wsl2-x86_24.2.0-trunk.314_amd64__6.1.71-S38fb-D0000-Pa1f9-Cdc84Hfe66-HK01ba-Vf5f1-B6a41-R448a.deb
https://github.com/armbian/build/commit/204d2e79b9a337722af292a06a309c49a51bee6e

igorpecovnik, 2024-01-06:
Loop device management is indeed buggy, but retry helps. We have no solution for that part yet.

viraniac, 2024-01-24:
This is a kernel bug not in Armbian, but from wsl2 ubuntu. You can try installing our kernel and it might fix the issue.

closed 2024-01-24 by viraniac

NicolasDorier, 2024-01-30:
> We are aware there might be bugs introduced in WSL2 kernels, which are currently out of our control. Those hints / files could help you resolve this, but its untested / WIP / not supported yet: https://fi.mirror.armbian.de/beta/pool/main/l/linux-6.1.71/linux-image-current-wsl2-x86_24.2.0-trunk.314_amd64__6.1.71-S38fb-D0000-Pa1f9-Cdc84Hfe66-HK01ba-Vf5f1-B6a41-R448a.deb [204d2e7](https://github.com/armbian/build/commit/204d2e79b9a337722af292a06a309c49a51bee6e)

Tried [v23.11.2](https://github.com/armbian/build/releases/tag/v23.11.2), encountered the error as well.
First time the build worked, and then always this error. Restarting wsl probably fix it.

adeepn, 2024-01-30:
ubuntu with kernel 5.15.x in same trouble

Tonymac32, 2024-01-31:
Any/all 5.15 wsl will fail.  6.1 should work though, can you give us a uname -a on that Armbian image?  WSL uses a kernel stored elsewhere in the filesystem and might be ignoring the Armbian one

NicolasDorier, 2024-01-31:
Restarting with `wsl --shutdown` didn't fix the issue.

I have no clue why it worked once, but since then no luck.

NicolasDorier, 2024-01-31:
`Linux hack0 5.15.146.1-microsoft-standard-WSL2+ #1 SMP Tue Jan 30 19:09:49 JST 2024 x86_64 x86_64 x86_64 GNU/Linux`

NicolasDorier, 2024-01-31:
I changed the WSL kernel, no more issue!

`Linux hack0 6.1.21.2-microsoft-standard-WSL2+ #1 SMP Wed Jan 31 16:12:58 JST 2024 x86_64 x86_64 x86_64 GNU/Linux`

hmdevelopers, 2024-02-12:
@adeepn I had the same issue on Ubuntu 22.04.03 LTS w/kernel 5.15.0-94. I fixed by installing kernel 6.7.3 using the "mainline" CLI tool (https://github.com/bkw777/mainline).

adeepn, 2024-02-12:
> @adeepn I had the same issue on Ubuntu 22.04.03 LTS w/kernel 5.15.0-94. I fixed by installing kernel 6.7.3 using the "mainline" CLI tool (https://github.com/bkw777/mainline).

i had this error inside build-vms with current ubuntu/jammy. seems kernel/partprobe bug

ig3, 2024-02-20:
I had this issue with armbian / build commits b600ead20d722434c062c3303316195413de70c0 and 2afd8fe9bb8aca33a2a3913e931fc43c43c8e309. Multiple executions of ./compile.sh ended with this error.

After installing armbian / build commit b600ead20d722434c062c3303316195413de70c0 late January, I ran ./compile.sh multiple times without the error. When I tried building today, I got the error with both my original install of armbian / build (unchanged since Jan) and a fresh install today: 100% failure rate.

I did have one other error today, but I can't imagine how it could be relevant. Downloading base-files was failing due to a 10 second time limit on the curl command. I have a rather slow broadband connection and downloading the page in my browser was taking more than 9 seconds without cache. I increased the time limit to 30 seconds, after which no further problem. But this caused early failure of the script. I don't think it was getting anywhere near manipulating the loopback device. However, since solving this problem, I have had the problem with /dev/loop0 partitions every attempt with both installations of armbian / build.

On investigation I found in lib/functions/image/partitioning.sh line 393:

`        run_host_command_logged losetup "${LOOP}" "${SDCARD}".raw # @TODO: had a '-P- here, what was it?`

I restored the -P, making it:

`        run_host_command_logged losetup -P "${LOOP}" "${SDCARD}".raw # @TODO: had a '-P- here, what was it?`

After which ./compile.sh ran to successful completion. Only two trials thus far, but after a day of every attempt failing, that's looking like a solution to me.

The -P option is described as: Force  the kernel to scan the partition table on a newly created loop device.

I am running armbian / build on:

```
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

The root cause of the problem may be in the ubuntu kernel but the -P option to losetup appears to at least be a workaround for me.

rpardini, 2024-03-02:
@ig3 can you produce a log of this working with the -P while running Microsoft's default WSL2 kernel?

ig3, 2024-03-02:
@rpardini I can provide logs from successful builds but I am building on Ubuntu Jammy running on a QEMU emulation of x86_64, running on Debian Bullseye, until I upgraded to Bookworm this weekend, so I don't know how running Microsoft's default WSL2 kernel is relevant, unless you are asking if I can build Armbian with Microsoft's kernel instead of the default. I'm willing to have a go doing that if it would help, but I have no idea how, so would need guidance. I am very new to the Armbian build system.

tntturtl, 2024-05-09:
I can confirm that @ig3 workaround was successful in my case. 
Logs 
[https://paste.next.armbian.com/raw/juzefijefe](url)