armbian/build #6136: failed to compile late in process "error in function create_rootfs_cache"

KurdtNervanna, 2024-01-07:
### What happened?

<pre>[<font color="#26A269"> o.k. </font>] Checking MD5 sum of installed packages [<font color="#A2734C"> debsums </font>]
debsums: changed file /usr/share/doc/base-files/changelog.gz (from base-files package)
debsums: changed file /etc/os-release (from base-files package)
debsums: changed file /etc/issue.net (from base-files package)
debsums: changed file /etc/issue (from base-files package)
[<font color="#C01C28"> error </font>] ERROR in function create_rootfs_cache [<font color="#A2734C"> functions/cli/cli-entrypoint.sh:109 -&gt; functions/main/build-tasks.sh:302 -&gt; functions/main/build-tasks.sh:225 -&gt; functions/main/rootfs-image.sh:66 -&gt; functions/rootfs/create-cache.sh:229 -&gt; functions/logging/traps.sh:0 </font>]
[<font color="#C01C28"> error </font>] MD5 sums check of installed packages failed 
[<font color="#26A269"> o.k. </font>] Process terminated 
[<font color="#C01C28"> error </font>] unmount_on_exit() called! [<font color="#A2734C"> functions/cli/cli-entrypoint.sh:109 -&gt; functions/main/build-tasks.sh:302 -&gt; functions/main/build-tasks.sh:225 -&gt; functions/main/rootfs-image.sh:66 -&gt; functions/rootfs/create-cache.sh:229 -&gt; functions/logging/traps.sh:1 -&gt; functions/main/rootfs-image.sh:0 </font>]
[<font color="#26A269"> o.k. </font>] Unmounting [<font color="#A2734C"> /home/nucci/retrostone2_image/.tmp/rootfs-3aa36241-245c-41da-a0c4-1d4564c6c663/ </font>]
[<font color="#C01C28"> error </font>] ERROR in function unmount_on_exit [<font color="#A2734C"> functions/cli/cli-entrypoint.sh:109 -&gt; functions/main/build-tasks.sh:302 -&gt; functions/main/build-tasks.sh:225 -&gt; functions/main/rootfs-image.sh:66 -&gt; functions/rootfs/create-cache.sh:229 -&gt; functions/logging/traps.sh:1 -&gt; functions/main/rootfs-image.sh:27 -&gt; functions/logging/traps.sh:0 </font>]
[<font color="#C01C28"> error </font>] debootstrap-ng was interrupted 
[<font color="#26A269"> o.k. </font>] Process terminated 
</pre>

### How to reproduce?

I used the retrostone compile file from here: [https://github.com/maximilianvoss/retrostone2_image/blob/retrostone2_uboot/RETROSTONE2.md](url)

I do not know how to fix, please advise.

### Branch

other

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-01-07:
We cannot provide support for 3rd party forks.

closed 2024-01-07 by EvilOlaf

KurdtNervanna, 2024-01-07:
ok, thank you anyway, sorry for the bother.