armbian/build #6144: cannot compile uboot without existing cache anymore

prahal, 2024-01-09:
### What happened?

if BOOTBRANCH is not set in board config (most boards), u-boot compilation fails when trying to delete the parent directory that is a mount point (thus errors out with "device busy").
```
[🐳|🌱] Getting sources from Git [ u-boot v2022.07 ]                                                                                                                                                               
[🐳|🚸] Removing previously half-checked-out tree [ /armbian/cache/sources/u-boot-worktree// ]                                                                                                                     
rm: cannot remove '/armbian/cache/sources/u-boot-worktree/': Device or resource busy  
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/general/git.sh:126               
```

This is because git_work_dir is set to `GIT_FIXED_WORKDIR` just above in lib/functions/general/git.sh
```
        # if GIT_FIXED_WORKDIR has something, ignore above logic and use that directly.                                                                                                                            
        if [[ "${GIT_FIXED_WORKDIR}" != "" ]]; then                                                                                                                                                                
                display_alert "GIT_FIXED_WORKDIR is set to" "${GIT_FIXED_WORKDIR}" "git"                                                                                                                           
                git_work_dir="${SRC}/cache/sources/${GIT_FIXED_WORKDIR}"                                                                                                                                           
        fi
```
GIT_FIXED_WORKDIR is set to `BOOTSOURCEDIR` in lib/functions/compilation/uboot-git.sh uboot_prepare_git
```
GIT_FIXED_WORKDIR="${BOOTSOURCEDIR}" \                                                                                                                                                         
                         GIT_BARE_REPO_FOR_WORKTREE="${uboot_git_bare_tree}" \                                                                                                                                  
                         GIT_BARE_REPO_INITIAL_BRANCH="master" \                                                                                                                                                
                         GIT_SKIP_SUBMODULES="${UBOOT_GIT_SKIP_SUBMODULES}" \                                                                                                                                   
                         fetch_from_repo "$BOOTSOURCE" "$BOOTDIR" "$BOOTBRANCH" "yes" # fetch_from_repo <url> <dir> <ref> <subdir_flag>        
```
and `BOOTSOURCEDIR` is set in lib/functions/main/config-prepare.sh config_post_main:
`BOOTSOURCEDIR="u-boot-worktree/${BOOTDIR}/$(branch2dir "${BOOTBRANCH}")"`

but adding logs above BOOTSOURCEDIR assignment in lib/functions/main/config-prepare.sh config_post_main I see that at this point BOOTDIR and BOOTBRANCH are empty. These variables are set later in the preparation process!
Thus Armbian build framework tries to delete the parent directory (the mount point) instead of the u-boot git working directory.
Sadly I do not know how to fix this issue.

Note probably another bug: if I set BOOTBRANCH in the config/board file this workaround the above issue. But if I set BOOTBRANCH_BOARD as most board configs do then I also have the above bug. (this second bug is I believe BOOTBRANCH defaulting to BOOTBRANCH_BOARD should be done before any use of BOOTBRANCH by the build framework).

In the log pastebin, the config_post_main BOOTBRANCH, BOOTDIR and BOOTSOURCEDIR are the logs I added in lib/functions/main/config-prepare.sh config_post_main (their values are empty) :
```
--> (0) INFO: config_post_main BOOTBRANCH
--> (0) INFO: config_post_main BOOTDIR
--> (0) INFO: config_post_main BOOTSOURCEDIR [ u-boot-worktree// ]
--> (0) EXTENSIONS: Extension Method 'late_family_config' being called from [ cli/entrypoint.sh:176 -> cli/utils-cli.sh:136 -> cli/cli-artifact.sh:36 -> artifacts/artifacts-obtain.sh:34 -> artifacts/artifact-uboot.sh:194 -> main/config-prepare.sh:79 -> logging/section-logging.sh:111 -> logging/section-logging.sh:81 -> main/config-prepare.sh:233 ]
--> (0) EXTENSIONS: Extension Method being called: late_family_config [ hook: late_family_config ]
--> (0) EXTENSIONSTRACE: Extension-managed hook starting 'late_family_config': will run 2 functions [ common_defaults_for_mainline_kernel common_defaults_for_mainline_uboot ]
--> (0) EXTENSIONSTRACE: Extension Method late_family_config [ 1/2 (ext:built-in) common_defaults_for_mainline_kernel ]
--> (0) EXTENSIONSTRACE: Extension-managed hook starting 1/2 [ late_family_config__common_defaults_for_mainline_kernel ]
--> (0) DEBUG: KERNEL_MAJOR_MINOR ('6.6') is set and KERNELBRANCH is unset [ hopefully mainline ]
--> (0) DEBUG: KERNELSOURCE is unset [ common_defaults_for_mainline ]
--> (0) DEBUG: KERNELBRANCH is unset [ common_defaults_for_mainline ]
--> (0) DEBUG: LINUXFAMILY is set by board/family/hooks to 'rockchip64' [ common_defaults_for_mainline ]
(...)
--> (0) DEBUG: BOOTDIR is unset [ common_defaults_for_mainline ]
--> (0) DEBUG: BOOTSOURCE is unset [ common_defaults_for_mainline ]
--> (0) DEBUG: BOOTBRANCH is unset; BOOTBRANCH_BOARD= or tag:v2022.07 [ common_defaults_for_mainline ]
--> (0) EXTENSIONSTRACE: Extension-managed hook finished 2/2 [ late_family_config__common_defaults_for_mainline_uboot ]
--> (0) EXTENSIONSTRACE: Extension-managed hook ending 'late_family_config': completed [ late_family_config ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ KERNELSOURCE='https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ KERNELBRANCH='branch:linux-6.6.y' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ KERNELPATCHDIR='archive/rockchip64-6.6' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ KERNEL_PATCH_ARCHIVE_BASE='rockchip64' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ BOOTSOURCE='https://github.com/u-boot/u-boot' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ BOOTBRANCH='tag:v2022.07' ]
--> (0) CHANGE-TRACKING: change-tracking: after late_family_config hooks [ BOOTDIR='u-boot' ]
```

### How to reproduce?

`./compile.sh uboot BOARD=helios64 BRANCH=edge RELEASE=bookworm DEBUG="yes" ARTIFACT_IGNORE_CACHE="yes"`
or
`./compile.sh uboot BOARD=rockpro64 BRANCH=edge RELEASE=bookworm DEBUG="yes" ARTIFACT_IGNORE_CACHE="yes"`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/ogifamunak

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-01-13:
Happened in CI https://paste.armbian.com/yikiwipiku

rpardini, 2024-01-13:
Thanks @prahal for identifying this, I introduced this bug and will be sending a fix soon

PR #6157 (2024-01-13): config: calculate BOOTSOURCEDIR & friends only _after_ `late_family_config` hook; fixes #6144
#### config: calculate BOOTSOURCEDIR & friends only _after_ `late_family_config` hook; fixes #6144

- config: calculate BOOTSOURCEDIR & friends only _after_ `late_family_config` hook; fixes #6144
  - snafu on my part, sorry
  - also: add more change-tracking variables; add more calls to change tracking in some key places, for future clarity
commit 04bde7ac (2024-01-13): config: calculate BOOTSOURCEDIR & friends only _after_ `late_family_config` hook; fixes #6144

- snafu on my part, sorry
- also: add more change-tracking variables; add more calls to change tracking in some key places, for future clarity
closed 2024-01-13 by igorpecovnik