armbian/build #6179: Unknown git source

tburschka, 2024-01-19:
### What happened?

try to build (on my nanopc-t4: Armbian 23.8.1 Bookworm with Linux 6.1.63-current-media) for nanopc-t4 with custom kernel source, getting this error

```
[🌱] Producing new & caching [ GIT_INFO_KERNEL ]
[🌱] Fetching SHA1 of 'branch' 'refs/heads/master' [ https://git.linuxtv.org/media_tree.git ]
[🌱] SHA1 of branch refs/heads/master [ '60a031b64984ad4a219a13b0fe912746b586bb9b' ]
[💥] error! [ Unknown git source 'https://git.linuxtv.org/media_tree.git'  ]
[💥] Exiting with error 43 [ at /home/user/build/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
   obtain_makefile_body_from_git() --> lib/functions/general/git-ref2info.sh:127
       memoized_git_ref_to_info() --> lib/functions/general/git-ref2info.sh:188
                   run_memoized() --> lib/functions/general/memoize-cached.sh:69
   obtain_kernel_git_info_and_makefile() --> lib/functions/artifacts/artifact-kernel.sh:218
   artifact_kernel_prepare_version() --> lib/functions/artifacts/artifact-kernel.sh:75
       artifact_prepare_version() --> lib/functions/artifacts/artifacts-obtain.sh:34
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
       obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:88
       build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:390
    main_default_build_packages() --> lib/functions/main/build-packages.sh:108
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:31
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
```

### How to reproduce?

./compile.sh build BOARD=nanopct4 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=yes CLEAN_LEVEL=make,debs,oldcache DEST_LANG=de_DE.UTF-8 EXTERNAL_NEW=prebuilt KERNELBRANCH=branch:master KERNEL_CONFIGURE=yes KERNELSOURCE=https://git.linuxtv.org/media_tree.git RELEASE=bookworm

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tburschka, 2024-01-19:
seems it was fixed but not work anymore: https://github.com/armbian/build/issues/5381

igorpecovnik, 2024-01-19:
It looks like syntax / git url error. Try this for example:

./compile.sh kernel BOARD=uefi-arm64 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes CLEAN_LEVEL=make,debs,oldcache DEST_LANG=de_DE.UTF-8 EXTERNAL_NEW=prebuilt KERNELBRANCH=branch:rk3588-v6.7 KERNELSOURCE=https://gitlab.collabora.com/hardware-enablement/rockchip-3588/linux RELEASE=bookworm EXTRAWIFI=no

tburschka, 2024-01-20:
@igorpecovnik this is compiling, but this kernel source is not working for my nanopct4/rk3399, isn't it? I try do set BOARD correctly but patching failed:

```
[🔨]   ╭─────────────────────┬──────────────────────────────────┬───────────────────────────────────╮
[🔨]   │ Patch               │ Patching output                  │ Rejects                           │
[🔨]   ├─────────────────────┼──────────────────────────────────┼───────────────────────────────────┤
[🔨]   │ rk356x-vop2-support │ patching file                    │ ---                               │
[🔨]   │                     │ "drivers/gpu/drm/rockchip/dw_hdm │ drivers/gpu/drm/rockchip/dw_hdmi… │
[🔨]   │                     │ i-rockchip.c"                    │ +++                               │
[🔨]   │                     │ Hunk #1 succeeded at 228 (offset │ drivers/gpu/drm/rockchip/dw_hdmi… │
[🔨]   │                     │ 136 lines).                      │ @@ -192,6 +180,7 @@ static const  │
[🔨]   │                     │ Hunk #2 succeeded at 317 (offset │ struct dw_hdmi_phy_config         │
[🔨]   │                     │ 136 lines).                      │ rockchip_phy_config[] = {         │
[🔨]   │                     │ Hunk #3 FAILED at 192.           │     { 74250000,  0x8009, 0x0004,  │
[🔨]   │                     │ Hunk #4 succeeded at 1283        │ 0x0272},                          │
[🔨]   │                     │ (offset 1038 lines).             │     { 148500000, 0x802b, 0x0004,  │
[🔨]   │                     │ 1 out of 4 hunks FAILED --       │ 0x028d},                          │
[🔨]   │                     │ saving rejects to file           │     { 297000000, 0x8039, 0x0005,  │
[🔨]   │                     │ "/tmp/tmpjsak_7nj"               │ 0x028d},                          │
[🔨]   │                     │                                  │ +   { 594000000, 0x8039, 0x0000,  │
[🔨]   │                     │                                  │ 0x019d},                          │
[🔨]   │                     │                                  │     { ~0UL,      0x0000, 0x0000,  │
[🔨]   │                     │                                  │ 0x0000}                           │
[🔨]   │                     │                                  │  };                               │
[🔨]   │                     │                                  │                                   │
[🔨]   │                     │                                  │                                   │
[🔨]   ╰─────────────────────┴──────────────────────────────────┴───────────────────────────────────╯
```

I try to create a build that enables me to use the camera modules in combination with crownest.
I try to use this documentation and custom patches as source: https://github.com/initBasti/NanoPC-T4_armbian_configuration

igorpecovnik, 2024-01-20:
> but this kernel source is not working for my nanopct4/rk3399, isn't it?

It looks that way. Armbian build framework is a tool and we support its functioning. I proved you its nothing wrong on our side. At least not in this context. 

> I try to create a build that enables me to use the camera modules in combination with crownest.

If you don't know how to proceed from here, you need to invest some time into learning. Perhaps try to contact that person, make some bond with Armbian community and try together bringing the code into main code. We have no resources to get involved into hw / 3rd party problems or educating people how to use framework. Hints, responds as such, only. You will need to invest some time and get familiar with, start with some simple problem, learn how to interpret logs, ... Nothing here is Armbian specific - doing by hand would be similar problematic. Its general research and development, thus user's problem. I am truly sorry.