armbian/build #6186: Rock Pi 4C+ - builds were not booting recently - fix needed to armbianEnv.txt

vijaygill, 2024-01-21:
### What happened?

After buring the image, the SBC would just show the uboot logo and do nothing.

So I mounted the SD card under linux, edited armbianEnv.txt by hand to update the file name in the entry for "fdtfile". Then after putting the SD card back in the SBC, it booted just fine.

Then I decided to make changes to code in build repo itself to see if a new image could be built with that value. And that worked too!

Here is the output of  git diff 

```
diff --git a/config/boards/rockpi-4cplus.csc b/config/boards/rockpi-4cplus.csc
index 39e7155bd..fe70380c5 100644
--- a/config/boards/rockpi-4cplus.csc
+++ b/config/boards/rockpi-4cplus.csc
@@ -6,6 +6,6 @@ BOOTCONFIG="rock-pi-4c-rk3399_defconfig"
 KERNEL_TARGET="current,edge"
 FULL_DESKTOP="yes"
 BOOT_LOGO="desktop"
-BOOT_FDT_FILE="rockchip/rk3399-rock-pi-4c-plus.dtb"
+BOOT_FDT_FILE="rockchip/rk3399-rock-4c-plus.dtb"
 BOOT_SCENARIO="tpl-spl-blob"
 BOOT_SUPPORT_SPI=yes
```

### How to reproduce?

Just build the image by following the standard procedure. I just build a standard default CLI image with edge kernel.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

vijaygill, 2024-01-22:
@igorpecovnik - is it ok for me to raise a PR for this? Or is it something the developers would like to do themselves, because they have wider view whereas I was just concentrating on getting my 4C+ booting up.

viraniac, 2024-01-22:
> @igorpecovnik - is it ok for me to raise a PR for this? Or is it something the developers would like to do themselves, because they have wider view whereas I was just concentrating on getting my 4C+ booting up.

go ahead and raise a PR.

PR #6189 (2024-01-22): Fix #6186 - update armbianEnv.txt for Rock Pi 4C+.
# Description

This fixes #6186 

Jira reference number [AR-2031]

# How Has This Been Tested?

I built images and checked the contents of armbianEnv.txt in every image.

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] Any dependent changes have been merged and published in downstream modu [... cut]
commit 748fc87c (2024-01-22): Fix #6186 - update armbianEnv.txt for Rock Pi 4C+.
closed 2024-01-22 by igorpecovnik