armbian/build #6197: Cloud-Init Supprot and Armbian First

rvalle, 2024-01-24:
### Which feature would you like to have?

I am in the process of adding cloud-init support to Armbian. 

Cloud Init is industry standard for customizing virtual targets, but with the NoCloud source it is also possible to use it to customize physical devices. Ubuntu's official Raspberry Pi distributions use it for some time.

I believe the best way to adopt it for Armbian would be to  move Armbian_First_Run scripts to an optional extension and then create an extension that provides Cloud Init. It does not make sense to have 2 different, and possibly competing, initialization systems on the same target.

Cloud Init requires the existence of a FAT partition, which is identified by label. PIs use the boot partition, but it is also posisible to use a partition for the purpose, will be identified by label.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2024-01-24:
@rpardini already did something on the topic, but I haven't tried it so I don't know how (well) it function. Check if you can use that as a base: https://github.com/rpardini/armbian-build/tree/extensions/userpatches/extensions

rvalle, 2024-01-24:
thanks @igorpecovnik. I will see what I can pick from the branch.

I actually have it working, but the intgration in armbian is not clean... and there is also the question of armbian first run.

I recommend following the implementation of Ubuntu itself for the RPI.

closed 2024-02-26 by igorpecovnik