armbian/build #6200: OdroidM1 no Console or SSH configured

rvalle, 2024-01-25:
### What happened?

It is not possible to access OdroidM1 after installing armbian, there is no serial console or SSH installed. The target can be pinged on Ethernet. 

### How to reproduce?

Install the image from Armbian Downloads:  Armbian_23.11.2_Odroidm1_jammy_current_6.1.68.img

Or build your own with:

```
./compile.sh \
  BOARD=odroidm1 \
  RELEASE=jammy \
  BRANCH=current \
  KERNEL_CONFIGURE=no \
  KERNEL_GIT=shallow \
  BUILD_MINIMAL=yes \
  BUILD_DESKTOP=no \
  CONSOLE_AUTOLOGIN=no

```

The target will boot, and welcome:
```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 4.19.206 (odroid@943431a2eaa2) (gcc version 6.3.1 20170109 (Linaro GCC 6.3-2017.02), GNU ld (Linaro_Binutils-2017.022
[    0.000000] Machine model: Hardkernel ODROID-M1
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] bootconsole [uart8250] enabled
```

However no terminal comes up on the Serial interface, the target will get IP on ethernet and responds to Ping, but no SSH is present there.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viraniac, 2024-01-25:
> Armbian_23.11.2_Odroidm1_jammy_current_6.1.68.img

> [    0.000000] Linux version 4.19.206 (odroid@943431a2eaa2) (gcc version 6.3.1 20170109 (Linaro GCC 6.3-2017.02), GNU ld (Linaro_Binutils-2017.022

That doesn't look right. Are you sure you have flashed it correctly? If you used Armbian_23.11.2_Odroidm1_jammy_current_6.1.68.img image, kernel version should have been 6.1.68 and not 4.19.206

igorpecovnik, 2024-01-25:
> Linux version 4.19.206 (odroid@943431a2eaa2)

This is for sure not Armbian. Please keep this issue tracker free from hardware related issues. 

https://www.armbian.com/odroid-m1/ contains info how to start with this board.

closed 2024-01-25 by igorpecovnik

rvalle, 2024-01-25:
@igorpecovnik @viraniac 

Yes! you are absolutely right. I looks like I have been booting a system installed in the SPI memory instead of SD/eMMC.

My builds were working, all along.

Thanks for the heads up.