armbian/build #6210: Error compiling

Gjuju, 2024-01-27:
### What happened?

When building an bullseye standard console image for Orange pi 3 LTS i get this error :
i'm on Ububntu 22.04 LTS

`[🔨]   Usage:   
[🔨]     /usr/bin/python3 -m pip install [options] <requirement specifier> [package-index-options] ...
[🔨]     /usr/bin/python3 -m pip install [options] -r <requirements file> [package-index-options] ...
[🔨]     /usr/bin/python3 -m pip install [options] [-e] <vcs project url> ...
[🔨]     /usr/bin/python3 -m pip install [options] [-e] <local project path> ...
[🔨]     /usr/bin/python3 -m pip install [options] <archive url/path> ...
[🔨]   
[🔨]   no such option: --break-system-packages
[💥] Error 2 occurred in main shell [ at /home/jubuntu/build/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        run_host_command_logged() --> lib/functions/logging/runners.sh:193
         prepare_python_and_pip() --> lib/functions/general/python-tools.sh:131
   aggregate_all_packages_python() --> lib/functions/configuration/aggregation.sh:47
             aggregate_packages() --> lib/functions/configuration/aggregation.sh:35
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
   aggregate_packages_in_logging_section() --> lib/functions/configuration/aggregation.sh:23
       main_default_start_build() --> lib/functions/main/start-end.sh:25
          do_with_default_build() --> lib/functions/main/default-build.sh:41
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
`

Can anyone point me in any direction to solve the problem ?
Best regards

### How to reproduce?

Command you gave to the framework!

./compile.sh build BOARD=orangepi3-lts BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no EXPERT=yes KERNEL_CONFIGURE=no RELEASE=bullseye

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log URL

https://paste.armbian.com/wijerunasa

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-01-27:
Retry with clean build environment. It seems like Python has been altered.
Also installing docker (and making it available for the user account you are using to build) should circumvent this.

SteeManMI, 2024-01-27:
It does seem like something is non-standard in your python environment.

The code is adding --break-system-packages for a specific version level of pip3:
from lib/functions/general/python-tools.sh:
	declare -a pip3_extra_args=("--no-warn-script-location" "--user")
	# if pip 23+, add "--break-system-packages" to pip3 invocations.
	# See See PEP 668 -- System-wide package management with pip
	# but the fact is that we're _not_ managing system-wide, instead --user
	if linux-version compare "${pip3_version_number}" ge "23.0"; then
		pip3_extra_args+=("--break-system-packages")
	fi
	if linux-version compare "${pip3_version_number}" ge "22.1"; then
		pip3_extra_args+=("--root-user-action=ignore")
	fi

According to https://pip.pypa.io/en/stable/cli/pip_install documentation the version of pip you are running 23.3.2 should support --break-system-packages 

Also on my 22.04 jammy environment I have pip3 version 22.0.2 installed with python3 version 3.10.12.  Your environment seems to have a pip3 of 23.3.2 on 3.10.12.

Gjuju, 2024-01-30:
Thanks a lot, you were right.

It works perfectly on another system with stock py3.

Best regards.

viraniac, 2024-01-30:
ok, as you are able to build. Closing the issue.

closed 2024-01-30 by viraniac