armbian/build #6280: Build broken with CRYPTROOT_ENABLE=yes

doodlebro, 2024-02-16:
### What happened?

Building with CRYPTROOT_ENABLE=yes leads to an error with grub configuration. Built on docker with a Mac.

grub-install: error: attempt to install to encrypted disk without cryptodisk enabled. Set `GRUB_ENABLE_CRYPTODISK=y' in file `/etc/default/grub'.

### How to reproduce?

./compile.sh build SHARE_LOG=yes BOARD=uefi-x86 BOOTSIZE=1024 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE=PASSWORD CRYPTROOT_SSH_UNLOCK_PORT=22 INSTALL_HEADERS=yes KERNEL_CONFIGURE=prebuilt KERNEL_GIT=shallow RELEASE=bookworm ROOTFS_TYPE=btrfs BTRFS_COMPRESSION=zstd:5 COMPRESS_OUTPUTIMAGE=gz PROGRESS_DISPLAY=none;

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/fejaheneno

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-02-17:
It passes compilation when added:
```
--- a/extensions/grub.sh
+++ b/extensions/grub.sh
@@ -276,6 +276,7 @@ configure_grub() {
                GRUB_DISABLE_OS_PROBER=false                             # Have to be explicit about enabling os-prober
                GRUB_FONT="/usr/share/grub/unicode.pf2"                  # Be explicit about the font to use so Ubuntu does not freak out and mess gfxterm
                GRUB_GFXPAYLOAD=keep
+                GRUB_ENABLE_CRYPTODISK=y
```

... but resulting image does not work properly. Grub prompts for password, it prompts for disk-unlock but then it doesn't mount /root ... so there is some other problem too. This needs deeper inspection.

doodlebro, 2024-02-17:
Thanks for confirming, my last successful build was Jan 21st at 1:21 UTC. Hope it helps.

doodlebro, 2024-02-25:
Adding a few notes as I found time to debug parts of this:

The main issue comes from forced encryption on the boot partition by grub, but I'm not sure when this started. Forcing encrypted /boot breaks remote access since the password must be entered at the grub menu.

Prior to Jan 21st, using the Armbian build flag "CRYPTROOT_ENABLE=yes" only encrypted the root partition, which allowed grub to boot into the initramfs so that I can remotely unlock the root partition. I would expect this to be preserved, or at least have a flag for the boot partition's inclusion.

I don't think we want "GRUB_ENABLE_CRYPTODISK=y" added to extensions/grub.sh for every build, or at all in my case. When I add this, images build and boot alright, so forced encryption of /boot by grub seems like the only thing to handle.

uzername123, 2025-02-02:
any updates on fix CRYPTROOT_ENABLE for uefi-x86 ?

sashasimkin, 2025-09-09:
Hi all! I managed to get it working in `v25.5.1` (branch `v.25.02`) with a few modifications:
1. Modifications to armbian build:
```
diff --git a/extensions/grub.sh b/extensions/grub.sh
index 563e70b4d..4c91038a5 100644
--- a/extensions/grub.sh
+++ b/extensions/grub.sh
@@ -172,7 +172,11 @@ pre_umount_final_image__install_grub() {
 	# Irony: let's use grub-probe to find out the UUID of the root partition, and then create a symlink to it.
 	# Another: on some systems (eg, not Docker) the thing might already exist due to udev actually working.
 	# shellcheck disable=SC2016 # some wierd escaping going on there.
+	# Root is needed to have UUID of unlocked /dev/mapper/armbian-root be discovered by grub-update, and put into grub.cfg instead of /dev/mapper/armbian-root
 	chroot_custom "$chroot_target" mkdir -pv '/dev/disk/by-uuid/"$(grub-probe --target=fs_uuid /)"' "||" true
+	# Use /boot so that it point to separate boot partition in case one exists and is required (lvm, cryptroot)
+	# If it doesn't exist - it'll be the same as mkdir for / above
+	chroot_custom "$chroot_target" mkdir -pv '/dev/disk/by-uuid/"$(grub-probe --target=fs_uuid /boot)"' "||" true

 	display_alert "Extension: ${EXTENSION}: Creating GRUB config..." "grub-mkconfig" ""
 	chroot_custom "$chroot_target" update-grub || {
@@ -276,6 +280,8 @@ configure_grub() {
 		GRUB_DISABLE_OS_PROBER=false                             # Have to be explicit about enabling os-prober
 		GRUB_FONT="/usr/share/grub/unicode.pf2"                  # Be explicit about the font to use so Ubuntu does not freak out and mess gfxterm
 		GRUB_GFXPAYLOAD=keep
+		GRUB_DISABLE_UUID=false  								 # Be explicit about wanting UUID
+		GRUB_DISABLE_LINUX_UUID=false  							 # Be explicit about wanting UUID
 	grubCfgFrag

 	if [[ "a${UEFI_GRUB_DISABLE_OS_PROBER}" != "a" ]]; then
diff --git a/lib/functions/image/initrd.sh b/lib/functions/image/initrd.sh
index 476ae8417..2aa6ab377 100644
--- a/lib/functions/image/initrd.sh
+++ b/lib/functions/image/initrd.sh
@@ -62,6 +62,7 @@ update_initramfs() {
 			[[ -d "${chroot_target}/etc/dropbear-initramfs/" ]] && initrd_files_to_hash+=("${chroot_target}/etc/dropbear-initramfs/")
 			[[ -d "${chroot_target}/etc/dropbear/initramfs/" ]] && initrd_files_to_hash+=("${chroot_target}/etc/dropbear/initramfs/")
 		fi
+		initrd_files_to_hash+=("${chroot_target}/etc/crypttab") # for updates to UUID
 	fi

 	# Find all the affected files; parallel md5sum sum them; invert hash and path, and remove chroot prefix.
```
2. In my custom extension I did following override (key here is setting `BOOTSIZE` to that boot partition is created):
```
function extension_prepare_config__yyy_override_prepare_grub_standard() {
    # Actual size is 30MB, this is already plenty, and default 256 is too much
    # We don't need 256 defaut MB for this
	declare -g UEFISIZE=128
	declare -g BOOTSIZE=128
}
```
With the following changes the partition table looks like this (below), which makes `/boot` open (and I saw it's also hardcoded across the framework) and contain linux & initramfs images.
```
[🐳|🔨]   New situation:
[🐳|🔨]   Disklabel type: gpt
[🐳|🔨]   Disk identifier: 20FC54EA-DD2F-42EE-B59A-265875A2583F
[🐳|🔨]
[🐳|🔨]   Device                                                          Start      End  Sectors  Size Type
[🐳|🔨]   /armbian/.tmp/rootfs-1247F982-0EDB-4716-BF00-8572B55D521D.raw1   8192   270335   262144  128M EFI System
[🐳|🔨]   /armbian/.tmp/rootfs-1247F982-0EDB-4716-BF00-8572B55D521D.raw2 270336   532479   262144  128M Linux extended boot
[🐳|🔨]   /armbian/.tmp/rootfs-1247F982-0EDB-4716-BF00-8572B55D521D.raw3 532480 12056575 11524096  5.5G Linux root (x86-64)
[🐳|🔨]
[🐳|🔨]   The partition table has been altered.
[🐳|🔨]   Syncing disks.
```
---
For the why - I'm not sure how it was working before or what changed. I went over armbian history and couldn't find any changes to BOOTSIZE=0 setting is `grub.sh` or its handing in `partitioning.sh`, so I assume that bootpart was always skipped in this scenario.
I can speculate how it was working before Jan 2024 - the /boot/efi must've had contained initramfs and kernel images, and then the change from grub-mkconfig to grub-update caused grub to not find the files anymore.
---
I'm not sure about sending the PR to armbian/build though, because of the `BOOTSIZE` override in `grub.sh` - I've checked and it's there from the very beginning, so for some reason armbian team doesn't want both uefi and boot together.

P.s.: I tried setting `UEFI_MOUNT_POINT` to `/boot` but that wold've required to change grun-install, and it still couldn't find the partition. Likely because of `/dev/disk/by-uuid/` missing by that time. But the approarch with separate boot is perfect for my use-case, so I'll leave it like that and to your judgement as another option.

sashasimkin, 2025-09-09:
Jup, or rather not setting BOOTSIZE to 0 if CRYPTROOT_ENABLE=yes .

Btw, could you explain why there is BOOTSIZE=0 if UEFI partition is enabled?

igorpecovnik, 2025-09-09:
> could you explain why there is BOOTSIZE=0 if UEFI partition is enabled?

https://github.com/armbian/build/blob/main/extensions/grub.sh#L23
I can try. When grub is used (UEFI images), we don't have special boot partition as EFI partition takes that role.

sashasimkin, 2025-09-09:
Yep, saw that, thank you!

I'm still confused how it could work before because EFI partition is mounted at `/boot/efi`, while the code has `/boot` hardcoded for initramfs and kernel images. Maybe in the past these files lived in `/boot/efi`.

Delcaran, 2025-09-28:
> Using branch=v24.08, I successfully compiled an arm64 distro, but encountered the same issue when compiling the x86 distro.
For me it's broken in branch v24.08 and v25.08, with this error:
`grub-install: error: attempt to install to encrypted disk without cryptodisk enabled. Set `GRUB_ENABLE_CRYPTODISK=y' in file `/etc/default/grub'.`

This is my cli:
`./compile.sh build CARD_DEVICE=/dev/sde BOARD=uefi-arm64 BRANCH=current RELEASE=noble BUILD_DESKTOP=yes BUILD_MINIMAL=no BSPFREEZE=yes DESKTOP_APPGROUPS_SELECTED='browsers desktop_tools chat editors internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=xfce CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="password"`

igorpecovnik, 2025-09-28:
Older branches are not maintained. We try to keep last one in shape ... as there were some fixes for cryptoroot, this might only work in main branch.

Delcaran, 2025-09-29:
> Older branches are not maintained - too many moving parts. We try to keep last one in shape ... try main + patch from this issue.

That's understandable, but an untouched branch that's been proven to built should build again...

Anyway, I checkout main, applied the patch (manually, because it's made for another branch), and now the error is a generic lack of space, I guess in the devices created by the builder.
I'm sorry I do not possess the knowledge nor the time to investigate further.

SteeManMI, 2025-09-29:
> That's understandable, but an untouched branch that's been proven to built should build again...

Not true in the Armbian case.  A label will pin the Armbian build code, but an actual build pulls in code from dozens of other repositories and those obviously don't fall under the Armbian label/tag and may change over time.  And then add the issue that the build is highly dependent on your build environment so any change to any number of base packages in your environment could cause problems.  So the reality is that Armbian builds generally aren't reproducible for all of these reasons using the same version of the armbian build code.

closed 2025-11-02 by igorpecovnik