armbian/build #6316: Error with partitions on /dev/loop device building image if builder is ubuntu-x64-jammy

iav, 2024-02-29:
### What happened?

Trying to build image on laptop x64platform, ubuntu:jammy process finishes with error:
```
`[🐳|🔨]   The partition table has been altered.
[🐳|🔨]   Syncing disks.
[🐳|🌿] Allocated loop device [ LOOP=/dev/loop2 ]
[🐳|🔨]   Error: Partition(s) 1 on /dev/loop2 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        run_host_command_logged() --> lib/functions/logging/runners.sh:193
             prepare_partitions() --> lib/functions/image/partitioning.sh:230
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:86
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50`
```

### How to reproduce?

`/compile.sh build BOARD=rockpi-4a BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=jammy`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/olufuqotex

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iav, 2024-02-29:
I see that only on x64, the same on my aarch64 odroid-N2 with same ubuntu jammy works as it should.

igorpecovnik, 2024-02-29:
> I see that only on x64, 

Try this on your machine:
`sudo apt install linux-image-generic-hwe-22.04`

> the same on my aarch64 odroid-N2 with same ubuntu jammy works as it should.

Armbian is another level ;)

iav, 2024-02-29:
> Try this on your machine:
sudo apt install linux-image-generic-hwe-22.04

I did, and now my laptop build image again.
It's a magic. 

Thank you!

closed 2024-02-29 by iav

Lebensgefahr, 2024-03-02:
> > I see that only on x64,
> 
> Try this on your machine: `sudo apt install linux-image-generic-hwe-22.04`
> 
> > the same on my aarch64 odroid-N2 with same ubuntu jammy works as it should.
> 
> Armbian is another level ;)

Did you read ticket I've created? Maybe month ago I build armbian successfully. But today I have the error the same as in this ticket. 
There is a suggestion there that not breaking logic just one little parameter --partscan to losetup. 
And the main thing it is described in TODO of that command. Maybe it will be added. It will not break anything.

igorpecovnik, 2024-03-02:
> Maybe month ago I build armbian successfully. 

Yes, this bug is known and even we spent weeks fighting with it, we still don't have a proper / perfect solution. This is not Armbian build framework problem, but its down to loop management. 

Upgrading kernel helps, but its not 100% cure.

We generate hundredth of images in CI every day and so far watchdog is the only 100% cure :(

Lebensgefahr, 2024-03-02:
> Yes, this bug is known and even we spent weeks fighting with it, we still don't have a proper / perfect solution. This is not Armbian build framework problem, but its down to loop management. [...]

But there is a solution

igorpecovnik, 2024-03-02:
> But there is a solution. Why don't use it?

Open a PR, lets test it in most brutal ways. If its confirmed working, then its a solution. I would be very grateful if this works ...

Lebensgefahr, 2024-03-02:
> > But there is a solution. Why don't use it?
> 
> Open a PR, lets test it in most brutal ways. If its confirmed working, then its a solution. I would be very grateful if this works ...

There is a command line history as proof in ticket you labeled as duplicated.

Lebensgefahr, 2024-03-02:
Ready. https://github.com/armbian/build/pull/6334