armbian/build #6319: jammy rootfs build failed

adeepn, 2024-03-01:
### What happened?

New build rootfs for jammy fails on package version:

```
[🐳|🔨]   The following packages have unmet dependencies:
[🐳|🔨]    libc-bin : Depends: libc6 (< 2.36) but 1:2.35-0ubuntu3.4+2widevine is to be installed
[🐳|🔨]    libc6-dev : Depends: libc6 (= 2.35-0ubuntu3.6) but 1:2.35-0ubuntu3.4+2widevine is to be installed
[🐳|🔨]                Depends: libc-dev-bin (= 2.35-0ubuntu3.6) but it is not going to be installed
[🐳|🔨]   E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

### How to reproduce?

```
./compile.sh artifact BETA=yes BOARD=jethubj100 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no RELEASE=jammy REVISION=24.5.0-trunk.99.jh.216.1 SHARE_LOG=yes UPLOAD_TO_OCI_ONLY=yes WHAT=rootfs
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/enezorocim

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2024-03-01:
glibc 1:2.35-0ubuntu3.4+2widevine is from [my ppa](https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia), is that ported to armbian's repo? @igorpecovnik

igorpecovnik, 2024-03-01:
i already removed it. Should be gone from index soon.

closed 2024-03-01 by igorpecovnik