armbian/build #6406: latest rolling release missing Wdk2023 images

mirostauder, 2024-03-18:
### What happened?

images for `Microsoft Windows Dev Kit` - AKA: `WDK2023` AKA: `Voltera' AKA: 'Blackrock`
are missing since build
https://github.com/armbian/os/releases/tag/24.5.0-trunk.127

### How to reproduce?

check builds after
https://github.com/armbian/os/releases/tag/24.5.0-trunk.127

the latest available are:
https://github.com/armbian/os/releases/download/24.5.0-trunk.127/Armbian_24.5.0-trunk.127_Wdk2023_noble_wdk2023_6.7.0-rc6_xfce_desktop.img.txt
and
https://github.com/armbian/os/releases/download/24.5.0-trunk.127/Armbian_24.5.0-trunk.127_Wdk2023_trixie_wdk2023_6.7.0-rc6_minimal.img.txt

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 13 Trixie

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-03-18:
If status changed to community maintained, then images are produced here:
https://github.com/armbian/community/releases/latest

mirostauder, 2024-03-19:
thanks for the info,
what are the reasons/conditions to move the board to community maintained?
how can we have the board officially supported?

igorpecovnik, 2024-03-19:
> what are the reasons/conditions to move the board to community maintained?

https://docs.armbian.com/User-Guide_Board-Support-Rules/

> how can we have the board officially supported?

If conditions set in that rules are met.

With [levels of support we got from general public](https://github.com/sponsors/armbian), we can't finance anything, so its down to luck if someone will step up. For us more "support" = bigger cost, troubles, anger, pain, frustration, ... which remain largely uncompensated.

Linux distributions usually claim "support" status, when hardware is supported by Linux kernel, some rely on support set by others, many (ab)use Armbian name to claim "supported". But that means close to nothing. That doesn't assure that you can actually boot / use the thing or that features works. If we don't have information that someone is actually maintaining, if we never tried to boot, this hardware and/if status is unknown. Telling you that its supported, would be wrong, but in about IMO 60-70% things (partially) works. Information about support is still not very well maintained, so this information is often not accurate. We do what we can - this project can only provide you best effort since we pay for around 99% of the costs you create. Partners mitigate this a bit, but this is it.

mirostauder, 2024-03-27:
Yes, closing ticket.
And big thanks for your work!

closed 2024-03-27 by mirostauder