armbian/build #6417: Build error: Unable to find free loop device

AaronNGray, 2024-03-21:
### What happened?

I built a CSC image which fails: BananaPi M3:-

https://paste.armbian.com/pejewuyoha

then all other builds fail after this even after a build system reset :-

https://paste.armbian.com/evururohol

### How to reproduce?

```
./compile.sh build BOARD=bananapim3 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes EXPERT=yes KERNEL_CONFIGURE=no RELEASE=bookworm
```
```
./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/evururohol

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AaronNGray, 2024-03-21:
deleting the `cache` directory seems to solve it.

AaronNGray, 2024-04-02:
Still getting this issue after failed builds :-

https://paste.armbian.com/yarohanuzi

AaronNGray, 2024-04-02:
I will install another version of Ubuntu on a new drive and test whether the problem is specific to this install or not.

This seems to solve the issue temporarily. 

How to clean up old Snap revisions and remove loop devices :-

https://ashrafur.medium.com/how-to-clean-up-old-snap-revisions-and-remove-loop-devices-93e38f1ad1f8

hqnicolas, 2024-04-03:
Same here, clear cache don't help

#
$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
#

$ uname -a
Linux 6.5.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Mar 12 10:22:43 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

igorpecovnik, 2024-04-03:
This loop devices problem is related to host and its kernel. 6.5.x kernel with Jammy should work (better). We have many build runners, some on stock Jammy, some on Armbian Jammy, thus with our kernel. Perhaps try this https://github.com/armbian/sdk ... 

/dev/loop devices makes us troubles too, here and there, automation breaks in random manner due to that and which is why we need to repeate few builds here and there. 

Also check if you have enough of those loop devices. I have added `max_loop=96` to my build machine kernel as otherwise I ran out of them. But due to building several images in parallel.

hqnicolas, 2024-04-03:
@igorpecovnik 
> It only works once after fresh install and reboot doesn't help??

reboot with clean armbian cache doesn't help

i Was trying to build: JP_TVBOX 3566 on V6.6 with jammy 22.04 desktop cinnamon

today I will use the armbian v24.0.2 and so far, so good (waiting to finish) 15:11 GMT

igorpecovnik, 2024-04-03:
> i Was trying to build: JP_TVBOX 3566 on V6.6 with jammy 22.04 desktop cinnamon

First try some officially supported target like generic x86 / arm64 image. tvboxes are untested ...

edumeneses, 2024-04-03:
Hello everyone, and thanks for this awesome project.

I tested with a Pop!_OS host (Ubuntu 22.04) and found that it cannot find a free loop when using the main branch. It works fine if you checkout to v24.0.2.

Also, I found that building using docker works on the main branch (on my machine) when I modified line 229 in the [partitioning.sh](lib/functions/image/partitioning.sh) file. I changed the old amount (10) to 30:

`[[ $FIND_LOOP_CYCLES -gt 30 ]] && exit_with_error "Unable to find free loop device"`

A `lsblk` call in my terminal showed more than 20 loop devices in use (blaming Snap for that). Could the assumption of a free loop under 10 be the reason for the error ?

hqnicolas, 2024-04-03:
> First try some officially supported target like generic x86 / arm64 image. tvboxes are untested ...

you have a point! but the V24.02 solve everything

AaronNGray, 2024-04-04:
I found a solution to my issue it appears *snaps* was causing an issue with eating loop devices on my machine :-

https://ashrafur.medium.com/how-to-clean-up-old-snap-revisions-and-remove-loop-devices-93e38f1ad1f8

loganmancuso, 2024-05-09:
> Hello everyone, and thanks for this awesome project.
>
> I tested with a Pop!_OS host (Ubuntu 22.04) and found that it cannot find a free loop when using the main branch. It works fine if you checkout [...]

this worked for me. Snap did not have any dangling loop devices on my laptop but allowing it to continue to look for an available loopback to use allowed my build to work thanks so much.

hqnicolas, 2024-05-10:
> (blaming Snap for that).

Me to!

Ekmansoft, 2024-05-19:
So, I get this problem too, so we shouldn't close this ticket until we fix it on master. Just counting the number of loop devices seem to be a bad way to check if armbian has created too many of those. Like previous users, I have 20+ from snap. I'll try and see if the partitioning.sh patch helps on my Ubuntu 23.10 system.

EvilOlaf, 2024-05-19:
> master

`master` will no longer receive updates. If the problem is successfully tackled in `main` this can be closed.

Ekmansoft, 2024-05-19:
Lol. I meant the main branch, whatever it's called :-)

kadrabak, 2024-05-26:
I had similar issues where deleting the cache folder did not always help.
Once I started using fixed image size (see FIXED_IMAGE_SIZE= parameter) my problem "Unable to find free loop device" disappeared. 
However, I like the btrfs filesystem and with this, an operating system needs a separate boot partition. 
After some kernel upgrades failed, I learned to set BOOTSIZE parameter like in the example below to get a bigger boot partition.

./compile.sh build BOARD=pine64 BOOTSIZE=300 BRANCH=edge BTRFS_COMPRESSION=zlib BUILD_DESKTOP=no BUILD_MINIMAL=no FIXED_IMAGE_SIZE=7000 INSTALL_HEADERS=yes KERNEL_CONFIGURE=no RELEASE=noble ROOTFS_TYPE=btrfs SEVENZIP=no

kadrabak, 2024-05-26:
./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm
**..failed**

./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm FIXED_IMAGE_SIZE=7000 
**..failed**

 ./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm BOOTSIZE=300 FIXED_IMAGE_SIZE=7000 ROOTFS_TYPE=ext4
**... success**

kadrabak, 2024-05-26:
here is a build test on my setup:

./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm
**..failed**

./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm FIXED_IMAGE_SIZE=7000
**..failed**

./compile.sh build BOARD=bananapim5 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm BOOTSIZE=300 FIXED_IMAGE_SIZE=7000 ROOTFS_TYPE=ext4
**... success**
note: The system image was built with everything in one partition on filesystem ext4 because Armbian build system knows u-boot is able to boot from ext4 so it will not create boot partition

system: Ubuntu Mantic in Virtualbox, 12880MB Ram, 16Cores, 6.6.29-060629-generic
note: everything is done without reboot

guoxiaoqiao, 2024-05-29:
> I found a solution to my issue it appears **snaps** was causing an issue with eating loop devices on my machine :-
> 
> https://ashrafur.medium.com/how-to-clean-up-old-snap-revisions-and-remove-loop-devices-93e38f1ad1f8

I use Ubuntu 22.04 and this worked for me, thank you very much.

tjhexf, 2024-06-04:
I can't build it in ubuntu 22.04, getting this same issue.

igorpecovnik, 2024-06-04:
On "ubuntu 22.04" this is not possible to reproduce. Without logs, there is nothing we can do as we are at "it works on our computer" levels ;) Make sure you are using clean stock Ubuntu install with all updates. This is what our CI runs. If you have some software installed, that can eat out loops ... curse of snapd.

Fedora on the other hand is not officially supported, but its probably lack of loop devices at all or lack of capacity - not sufficient number of them.

AaronNGray, 2024-06-04:
It helps if people actually read issues properly.

kadrabak, 2024-06-11:
I run in same issue and there is hint why this is happened.
Command to list all loop devices =>
losetup -a | sort
/dev/loop0: [0027]:2897018 (/var/lib/snapd/snaps/bare_5.snap)
/dev/loop10: [0027]:3577273 (/var/lib/snapd/snaps/gtk-common-themes_1534.snap)
/dev/loop1: [0027]:5998653 (/var/lib/snapd/snaps/core20_2264.snap)
/dev/loop11: [0027]:4618189 (/var/lib/snapd/snaps/gtk-common-themes_1535.snap)
/dev/loop12: [0027]:6007698 (/var/lib/snapd/snaps/snapd_21465.snap)
/dev/loop13: [0027]:6079887 (/var/lib/snapd/snaps/snapd_21759.snap)
/dev/loop2: [0027]:6007912 (/var/lib/snapd/snaps/core20_2318.snap)
/dev/loop3: [0027]:5604489 (/var/lib/snapd/snaps/core22_1122.snap)
/dev/loop4: [0027]:5999535 (/var/lib/snapd/snaps/core22_1380.snap)
/dev/loop5: [0027]:5604994 (/var/lib/snapd/snaps/firefox_4090.snap)
/dev/loop6: [0027]:5999688 (/var/lib/snapd/snaps/firefox_4209.snap)
/dev/loop7: [0027]:4834933 (/var/lib/snapd/snaps/gnome-3-38-2004_119.snap)
/dev/loop8: [0027]:5266160 (/var/lib/snapd/snaps/gnome-3-38-2004_143.snap)
/dev/loop9: [0027]:5604818 (/var/lib/snapd/snaps/gnome-42-2204_176.snap)

Command to list an empty one =>
losetup -f
/dev/loop14

One loop device is quite problem when armbian build script need at least 2, one for boot and second for root filesystem.

kadrabak, 2024-06-11:
[🌱] Enabling command-not-found after build-time  [ tinkerboard-2:noble-current ]
[🌱] Unmounting [ /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d ]
[🌿] Actual rootfs size [ 2752MiB ]
[🚸] Rootfs post-tweaks size is larger than estimated tmpfs size [ 2752MiB > 2300MiB ]
[🌱] Preparing image file for rootfs [ tinkerboard-2 noble ]
[🌱] Current rootfs size [ 2752 MiB ]
[🌱] Using user-defined image size [ 7000 MiB ]
[🌱] Creating blank image for rootfs [ truncate: 7000 MiB ]
[🌱] Creating partitions [ /boot: ext4 root: btrfs ]
[🔨]   Checking that no-one is using this disk right now ... OK
[🔨]   
[🔨]   Disk /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw: 6.84 GiB, 7340032000 bytes, 14336000 sectors
[🔨]   Units: sectors of 1 * 512 = 512 bytes
[🔨]   Sector size (logical/physical): 512 bytes / 512 bytes
[🔨]   I/O size (minimum/optimal): 512 bytes / 512 bytes
[🔨]   
[🔨]   >>> Script header accepted.
[🔨]   >>> Created a new DOS (MBR) disklabel with disk identifier 0xc7059d71.
[🔨]   /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw1: Created a new partition 1 of type 'Linux extended boot' and of size 300 MiB.
[🔨]   /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw2: Created a new partition 2 of type 'Linux' and of size 6.5 GiB.
[🔨]   /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw3: Done.
[🔨]   
[🔨]   New situation:
[🔨]   Disklabel type: dos
[🔨]   Disk identifier: 0xc7059d71
[🔨]   
[🔨]   Device                                                                    Boot  Start      End  Sectors  Size Id Type
[🔨]   /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw1       32768   647167   614400  300M ea Linux extended boot
[🔨]   /data/armbian/build/.tmp/rootfs-d8196a9c-8447-477f-9876-d98ca78c957d.raw2      647168 14335999 13688832  6.5G 83 Linux
[🔨]   
[🔨]   The partition table has been altered.
[🔨]   Syncing disks.
[💥] error! [ Unable to find free loop device  ]
[💥] Exiting with error 43 [ at /data/armbian/build/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
             prepare_partitions() --> lib/functions/image/partitioning.sh:229
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:86
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] Unmounting recursively [ SDCARD - be patient ]
[🌿] Unmounting recursively [ MOUNT - be patient ]
[🚸] Freeing loop [ trap_handler_cleanup_rootfs_and_image /dev/loop13 ]
[🌿] Freeing loop device [ /dev/loop13 ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-d8196a9c-8447-477f-9876-d98ca78c957d.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-d8196a9c-8447-477f-9876-d98ca78c957d.log.ans https://paste.armbian.com/log ]

kadrabak, 2024-06-11:
log says freeing  Freeing loop device [ /dev/loop13 ]

after finishing all this I run these commands:
 losetup -f
/dev/loop14
admi2@xubuntu:/data/armbian/build$ losetup -a
/dev/loop1: []: (/var/lib/snapd/snaps/core20_2264.snap)
/dev/loop8: []: (/var/lib/snapd/snaps/gnome-3-38-2004_119.snap)
/dev/loop6: []: (/var/lib/snapd/snaps/firefox_4209.snap)
/dev/loop13: []: (/var/lib/snapd/snaps/snapd_21759.snap)
/dev/loop4: []: (/var/lib/snapd/snaps/core22_1380.snap)
/dev/loop11: []: (/var/lib/snapd/snaps/gtk-common-themes_1535.snap)
/dev/loop2: []: (/var/lib/snapd/snaps/core20_2318.snap)
/dev/loop0: []: (/var/lib/snapd/snaps/bare_5.snap)
/dev/loop9: []: (/var/lib/snapd/snaps/gnome-42-2204_176.snap)
/dev/loop7: []: (/var/lib/snapd/snaps/gnome-3-38-2004_143.snap)
/dev/loop5: []: (/var/lib/snapd/snaps/firefox_4090.snap)
/dev/loop12: []: (/var/lib/snapd/snaps/snapd_21465.snap)
/dev/loop3: []: (/var/lib/snapd/snaps/core22_1122.snap)
/dev/loop10: []: (/var/lib/snapd/snaps/gtk-common-themes_1534.snap)

So for me device 13 was taken because when build script finished this loop device is taken with snapd. 
Also and there is free device 14
???
Cannot understand it

igorpecovnik, 2024-06-11:
> Cannot understand it

Evil of snap :( 

Many of us are using snapd free Armbian Ubuntu where this problem doesn't happen. For machines, where we build in parallel, I had to add max_loop to kernel command line:

GRUB_CMDLINE_LINUX_DEFAULT=" console=tty1 max_loop=96"

kadrabak, 2024-06-11:
Great, applied.

alex3d, 2024-06-11:
Pre-allocation of 96 loop devices greatly increases chances of success, but this race condition could be prevented altogether by atomic find&take (`losetup --show ...`). Here is PR https://github.com/armbian/build/pull/6729

AaronNGray, 2024-06-23:
Fixed here :- https://github.com/armbian/build/pull/6729#pullrequestreview-2133571543

closed 2024-06-23 by AaronNGray