armbian/build #6444: Applying patch/misc/rtw88/6.6/001-drivers-net-wireless-realtek-rtw88-upstream-wireless.patch fails

theCalcaholic, 2024-03-27:
### What happened?

As of today, all my Armbian builds fail in the step `applying patch/misc/rtw88/6.6/001-drivers-net-wireless-realtek-rtw88-upstream-wireless.patch`.
4 days ago the builds were working with the same version of armbian/build (`v24.2.1` / revision `2cd46aceb5571db4746fdac9a32a5755f360b26b`), so I suspect that something related to an upstream project has changed.

Here are the logs of a failing build: 
[8456499587-rock64-logs.zip](https://github.com/armbian/build/files/14781243/8456499587-rock64-logs.zip)

### How to reproduce?

Here's the script I used to run the build (for the given logs with arguments `rock64 Rock64`):

https://github.com/nextcloud/nextcloudpi/blob/master/build/build-SD-armbian.sh

But in this case, the userpatches should be irrelevant since the build aborts way before custom scripts are executed.
so this is the relevant portion of the script:

```sh
git clone --depth 1 --branch v24.2.1 https://github.com/armbian/build armbian
cat > "armbian/userpatches/config-ncp.conf" <<EOF
BOARD="rock64"
BRANCH=current
RELEASE=bullseye-security
KERNEL_CONFIGURE=prebuilt
BUILD_DESKTOP=no
BUILD_MINIMAL=yes
USE_CCACHE=yes
INCLUDE_HOME_DIR=yes
EOF
export INCLUDE_HOME_DIR=yes
armbian/compile.sh ncp
```

### Branch

other

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theCalcaholic, 2024-03-27:
EDIT: RELEASE is `bullseye`, not `bullseye-security`

igorpecovnik, 2024-03-27:
Try to use branch v24.02 as there upstream sources should be frozen. Upstream sources are changing daily so its a lot of maintainainace. Tags doesn't have frozen sources and you will need to deal with this on your own. Alternatively is using main branch, which is fixed, v24.02 should probably also works.

theCalcaholic, 2024-03-27:
Thank you I'll give that a try

igorpecovnik, 2024-03-27:
Also check how we make images via extensions - we do it for HA, OpenHab, for demo https://github.com/armbian/os/tree/main/userpatches/extensions and we can also make and distribute them automatically:

[image]

theCalcaholic, 2024-03-28:
That looks very interesting... Right now, I have the issue that Armbian builds tend to break basically for every release, which requires hours of testing and troubleshooting from my side. That's why I was trying to work with fixed versions in the first place. Unfortunately, branch v24.0.2 doesn't build for me for other reasons (that I've yet to troubleshoot).

However, at the moment, I'm considering moving to a immutable distro anyways (since NextcloudPi is a great fit for a immutable system as it's not really meant to be interacted with by users outside of the admin interface). Therefore, I'm not sure if it would make sense to provide a userpatch extension.
But I'll definitely keep it in mind in case I change my mind!

igorpecovnik, 2024-03-28:
> Right now, I have the issue that Armbian builds tend to break basically for every release,

This has to work well, but perhaps not the way you are doing right now. If we help you prepare this?

In past 2 years we reworked build framework and our 24.02 is the first semi "immutable" on source level. I think we forgot to freeze one or two sources, but otherwise its there. Troubles are in some legacy functions ... But on the other hand, it is significantly easier for you to build application images for various arhitectures with single extension https://github.com/armbian/os/blob/main/userpatches/extensions/ha.sh and single YAML file:
https://github.com/armbian/os/blob/main/userpatches/targets-release-apps.yaml
and leave the nasty work to us.

If you have NCPI as a docker instance, its almost the same as HA.

theCalcaholic, 2024-03-28:
```sh
git clone --depth 1 --branch ${BRANCH} https://github.com/armbian/build armbian

# NCP userpatch
mkdir -p armbian/userpatches armbian/userpatches/overlay
cp build/armbian/armbian.sh armbian/userpatches/customize-image.sh
rsync -Aax --delete --exclude-from .gitignore --exclude *.img --exclude *.bz2 . armbian/userpatches/overlay/

cat > "armbian/userpatches/config-ncp.conf" <<EOF
BOARD="rock64"
BRANCH=current
RELEASE=bullseye
KERNEL_CONFIGURE=prebuilt
BUILD_DESKTOP=no
BUILD_MINIMAL=yes
USE_CCACHE=yes
INCLUDE_HOME_DIR=yes
EOF
export INCLUDE_HOME_DIR=yes
armbian/compile.sh ncp
```

This is basically how I've been building images ever since moving to armbian next (with BRANCH being tags or `main` - I even tried checking out individual commits to find working versions since `main` tended to break so often). The script that gets actually executed (copied to `armbian/userpatches/customize-image.sh` is [this one](https://github.com/nextcloud/nextcloudpi/blob/master/build/armbian/armbian.sh))

igorpecovnik, 2024-03-28:
> NCP does not come as a docker container. It is, as of today, a quite complex installation script that installs many services into the debian host system.

I understand the complexity ...

I tried ... problem seems to be related to Bullseye packages / Qemu. When Debian make a new stable release (Bookworm is out for some time), old stable starts to detiriorate. For us, it is mission impossible to deal with increasing number of problems that are not sorted out upstream anymore. I tried to build bookworm, but your script says "unsupported". How far away you are from bookworm, that will work and can you build natively, without qemu. That might work for bullseye too.

theCalcaholic, 2024-03-28:
@igorpecovnik Funnily enough, building with the main branch works, but with 24.02 it fails in group `prepare_partitions` with:

```
Error: Partition(s) 1 on /dev/loop3 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
```

I'm currently working on bookworm support and it should hopefully land in 1-2 weeks.

Anyways, I'll use the main branch for the upcoming release and hope it won't break too soon :D
Thank you for your help and sorry for my frustration previously.

igorpecovnik, 2024-03-28:
> Anyways, I'll use the main branch for the upcoming release and hope it won't break too soon :D

IMO you should be fine. Also our next "stable" branch will be more stable in the future. An improved version of sources freezing is [in the PR](https://github.com/armbian/build/pull/6318). 

You can achieve additional hardening by applying BSPFREEZE="yes" where kernel packages won't upgrade on apt upgrade. It will apply `apt-mark hold` on them.

> I'm currently working on bookworm support and it should hopefully land in 1-2 weeks.

That would help here too. All the best in adjusting.

> and sorry for my frustration previously.

Oh, its software development. We have plenty of that around ;)

theCalcaholic, 2024-04-23:
FYI: The main branch was working for the last release and the next release will be based on bookworm, so this shouldn't be an issue anymore for NCP.

Thank you for the support!

I'll leave the issue open for @hqnicolas, feel free to close it if it is solved for them as well or not relevant anymore